KeithSloan/GDML #7: improvement in setMaterial

lambdam94, 2020-04-24:
Hello Keith

in def setMaterial(obj, m) in GDMLObject,
 could you please replace "if len(MaterialsList)>0" by "    if (MaterialsList != [])  and (MaterialsList != None):"
Thanks!

In https://gitlab.cern.ch/VecGeom/VecGeom/tree/master/persistency/gdml/gdmls, there are some pb in the gdml test files without material.

Regards
Damien

KeithSloan, 2020-04-24:
Many Thanks - Please can you let me know one of the examples without materials.

Thanks for the suggested code, wondering if obj.materials has to be set even if not in file.

So thinking of this coding
``
def setMaterial(obj, m) :
    #print('setMaterial')
    if MaterialsList != None :
        obj.material = MaterialsList
        obj.material = 0
        if len(MaterialsList) > 0 :
            if not ( m == 0 or m == None ) :
                obj.material = MaterialsList.index(m)
    else :
        obj.Material = 0

lambdam94, 2020-04-24:
Hello Keith

https://gitlab.cern.ch/VecGeom/VecGeom/-/blob/master/persistency/gdml/gdmls/oneSphere.gdml
has not material
It is only a test file

Regards
Dam

lambdam94, 2020-04-24:
In this case, MaterialsList == [], you should probably test "[]" and "none" cases,
Regards
Dam

lambdam94, 2020-04-25:
Ok with the last upgrade

closed 2020-04-25 by lambdam94