KeithSloan/GDML #10: Solids to initially concentrate on

KeithSloan, 2020-04-27:
Parsing : ../SampleFiles/CERN/lhcbvelo.gdml

List of Solids in : ../SampleFiles/CERN/lhcbvelo.gdml
=====================================================
box
tube
union
trap
subtraction
cone
trd
polycone

Parsing : ../SampleFiles/CERN/Alice/alice.gdml

List of Solids in : ../SampleFiles/CERN/Alice/alice.gdml
========================================================
polyhedra
xtru
polycone
cone
tube
subtraction
box
union
trap
intersection
trd
para
arb8
torus
eltube
cutTube

lambdam94, 2020-04-27:
oneBox.gdml
-> ok under Freecad 0.18 and Freecad 0.19.AppImage
[image]

lambdam94, 2020-04-27:
oneTube.gdml
-> ok under Freecad 0.18 and Freecad 0.19.AppImage
[image]

closed 2022-03-07 by KeithSloan