KeithSloan/GDML #11: Can't add any objects, just the world volume gets exported.

quantatanu, 2020-05-12:
Hi, 

I am very new to CAD/GDML/Geometry in general. I was able to create and export the WorldVolume but no other objects get saved:

`GDMLElTube Object - added
Running the Python command 'ElTubeCommand' failed:
Traceback (most recent call last):
  File "/home/me/.FreeCAD/Mod/GDML/freecad/gdml/GDMLCommands.py", line 170, in Activated
    GDMLElTube(a,10,20,30,"mm",0)
  File "/home/me/.FreeCAD/Mod/GDML/freecad/gdml/GDMLObjects.py", line 613, in __init__
    obj.ViewObject.ShapeColor = colourMaterial(material)
  File "/home/me/.FreeCAD/Mod/GDML/freecad/gdml/GDMLObjects.py", line 174, in colourMaterial
    coeffRGB = MaterialsList.index(m)`

I was opening the saved GDML using a ROOT macro that opens other complex gdml files that I obtained from experts, I can see detailed structures. But in my case, despite adding objects like spheres, cones, etc inside the box, I just see an opaque box. I will really appreciate any help.

Thanks
____________________________________________
Ubuntu 18.04
ROOT 6-20
python3
freecad installed through apt-get from stable ppa

KeithSloan, 2020-05-12:
Hi unfortunately my computer had some tea and I have to wait for the end of lockdown to see if it is repairable, so currently high and dry. I have emailed Dam Lamb to ask if he can help. He is user lambdam94.

KeithSloan, 2020-05-12:
“But in my case by adding objects I just see opaque box”

Is that box like a bounding box in a world volume?

If so then if you activate the GDML workbench one of the icons is cycle.
Select the volume of the box and then click on the cycle icon, will(should) cycle though all objects in the volume , solid to wireframe to no display then back to solid.

KeithSloan, 2020-05-12:
Looking at the trace back, think the problem maybe that creation of solid through the GUI, I.e using icons in the the workbench was before material support and colour was added, the problem may well be because there is no material allocated by default.

quantatanu, 2020-05-12:
Hi, 

thank you so much for your prompt response.

I am adding a gif of what is going on with my freecad: 

[image]

The material of the object shows up, by default steel or air, but on the "appearances" I can choose other materials. But ultimately doesn't get drawn. 

"the problem may well be because there is no material allocated by default."

How can I add them in that case?

Thanks

KeithSloan, 2020-05-12:
Think we need lambdam94 to look at this, especially as the colour code is his. You might try dropping back to a commit prior to colour being added. Sorry I cannot help more as I don’t have access to a unix machine at present

lambdam94, 2020-05-13:
Hello,

I think there are 2 pbs :
1- “But in my case by adding objects I just see opaque box”
the opaque world box can be changed in "Display Mode" of the "GDMLBox_WorldBox"
  replace "Shaded" by "Wireframe"
We will fix it in a previous version.

2- Color pb:
In ~/FreeCAD/Mod/GDML/freecad/gdml/GDMLObjects.py
on line172 please add: 
       elif m not in MaterialsList: return  (0.5,0.5,0.5) 
(be carefull of the 8 spaces)

Does it fix your pbs?
 If it is ok, we will fix it for a next release.

Regards,
Dam

KeithSloan, 2020-05-13:
Have a new machine on order, hope to fix shortly after it gets delivered.

quantatanu, 2020-05-13:
As I am like an infant in this whole business, your freecad+gdml is my hope.. thanks to both of you for everything.

So...

> the opaque world box can be changed in "Display Mode" of the "GDMLBox_WorldBox"
> replace "Shaded" by "Wireframe"
> We will fix it in a previous version.

That's the first thing I did, the opacity is not a problem in the freecad display, on freecad "wireframe" makes the box transparent but when I open the exported GDML using ROOT, it's completely opaque.

And:

[image]

Now I can add objects to the world, this is how it looks:

[image]

You can see that it's a "Wireframe"

But, this is how it looks on ROOT... 

[image]

KeithSloan, 2020-05-14:
That is correct all boxes/objects in ROOT/Geant4 will display as opaque/solid. It is just that is not very convenient for viewing things that we set the default display of the World GDML object, box or sphere or what ever to wireframe. For checking things I use a program called gdmlview ( need to build from source, think I got it from Jefferson repro) it has a facility to switch from solid to wireframe and does it for the whole model. Remember that you are dealing with Monte Carlo modelling of particles and things like protons and other sub particles don’t respect boundaries/walls they still go through them

KeithSloan, 2020-05-14:
Can I check your scenario.

Am I correct that you are taking an existing file importing that and then adding a few solids and then exporting and that is when you initially hit the problems.

If yes then I think we still have an issue to think about. Adding solids via the GUI assumes that the default gdml is loaded. It contains the stainless steal material definition.

When you loaded your sample file it will have bought in its material definitions which probably don’t contain the stainless steel. You add things via GUI and with Dam’s fix hopefully the model will export, but the exported file may well have an object that references a material that does not exist in the materials section of the file and therefore ROOT/Geant4 will not be happy about this. I think we therefore have some thinking to do.
 again

lambdam94, 2020-05-14:
In your GDMLSphere name, could you add the 'GDMLSphere__' prefix?
For example 'GDMLSphere__mySphere'
It seems working for my simple case under freecad and root.

KeithSloan, 2020-05-14:
Just as a point of reference, the reason for names in FreeCAD starting with GDML is so that the user can tell that they are GDML solids implemented as FreeCAD objects, rather than FreeCAD box, sphere, etc

lambdam94, 2020-05-14:
With the "GDML" prefix, it seems working well:

[image]

quantatanu, 2020-05-14:
@KeithSloan  no I am creating an entirely new file.
@lambdam94 it's by default.. names already have that prefix.

By the way, I am facing a new problem, ``list index out of range``, what does that mean? I haven't seen this before! My object is entirely inside the world box, also the material is by default "SSteel" and the world is made of "G4_Air" (there are no other choices though! Unless I go to "appearances", but does that change material though?)

Please check out the attached gif. 

[image]

quantatanu, 2020-05-14:
By the way, [here](https://github.com/KeithSloan/FreeCAD_Python_GDML/blob/master/README.md), the instructions are different, it's an independent installation, while I installed through **freecad addon manager**. Should I remove the GDML addon and try the git method? Maybe you have a bug-free newer version on git now?

quantatanu, 2020-05-14:
Or, do you have a list that of material/objects that I can add to some file, that might fix this `list index out of range` issue?

KeithSloan, 2020-05-14:
> By the way, [here](https://github.com/KeithSloan/FreeCAD_Python_GDML/blob/master/README.md), the instructions are different, it's an independent installation, while I installed through **freecad [...]

That is the old version of the workbench it is a different repository. It has even more bugs. You are better off with this repro.

KeithSloan, 2020-05-14:
> @KeithSloan no I am creating an entirely new file.
> @lambdam94 it's by default.. names already have that prefix.
>
> By the way, I am facing a new problem, `list index out of range`, what does that [...]

This is a bug in that export has changed to remove the leading GDMLname, so that importing and exporting matched. The code to add an item via the GUI does not allow for this and hence the bug, sorry about this. New machine is on its way. Hope to fix shortly after it arrives.

KeithSloan, 2020-05-14:
> Or, do you have a list that of material/objects that I can add to some file, that might fix this `list index out of range` issue?

It’s a bug the name created by the GUI should have more info as export now splits off the leading GDMLname

quantatanu, 2020-05-14:
And what does ``Cannot compute Inventor representation for the shape of GDMLSphere`` mean?

lambdam94, 2020-05-14:
By default the names haven't the full prefix with '_'. '_' should be added to the prefix.
Replace "GDMLTube" by "GDMLTube_mytube01"
Replace "GDMLBox" by "GDMLBox_mybox01"
etc.
Today, only the name part after "_" is used in the exported gdml file.
[image]

quantatanu, 2020-05-14:
I don't know what happened, I just did the following:

1. Original label was ``GDMLBox_WorldBox`` I removed it and renamed it back the same.
2. Did the same for the part objects, it was ``GDMLSphere``, this time I changed it to ``GDMLSphere_mySphere``

Either both **1** and **2**, or one of them solved my problems. This is what I see on ROOT now:

[image]

quantatanu, 2020-05-14:
> By default the names haven't the full prefix with the "underscore". "underscore" should be added to the prefix.
> Replace "GDMLTube" by "GDMLTube_mytube01"
> Replace "GDMLBox" by "GDMLBox_mybox01"
> etc.
> Today, only the name part after "_" is used in the exported gdml file.
> [image]

Ah, I saw your reply only now! But somehow did what you suggested.

quantatanu, 2020-05-14:
Let me keep this issue open for a few more days...so that nothing bad pops up.. and maybe in the meanwhile, you might provide a newer and better version.

quantatanu, 2020-05-14:
Also, how can we add more materials, currently:

[image]

KeithSloan, 2020-05-14:
> Let me keep this issue open for a few more days...so that nothing bad pops up.. and maybe in the meanwhile, you might provide a newer and better version.

Can only update when my new machine arrives. It is on it way

KeithSloan, 2020-05-14:
> And what does `Cannot compute Inventor representation for the shape of GDMLSphere` mean?

Means something went wrong with the Code that draws the Sphere

KeithSloan, 2020-05-14:
New machine should arrive tomorrow some time. In the mean time I have created a new branch quanta and edited via the web, so untested. Tries to fix name and material not in list. Have to find where World_Box is created in the code. Extra materials etc can be added by editing the Default.gdml file in the Resources directory. Will try and add the facility to import xml file that gets parsed and added to current environment, but that will wait for new machine to arrive. Thank you for your patience and testing

quantatanu, 2020-05-15:
Thanks a ton for all the help! I will keep checking for updates...

quantatanu, 2020-05-18:
Hi,

> 2- Color pb:
> In ~/FreeCAD/Mod/GDML/freecad/gdml/GDMLObjects.py
> on line172 please add:
> elif m not in MaterialsList: return (0.5,0.5,0.5)
> (be carefull of the good number of spaces before "elif": 7)

I just removed and added back GDML through **Freecad tools->Addons Manager**, and had the problem of 0 not in list and remembered, I had add that line suggested by @lambdam94 , and it fixed the problem. So, this bug is still there or the GDML available through **Addons Manager** is the old one, and I have to manually git clone the new version?

KeithSloan, 2020-05-18:
> Hi,
>
>
> I just removed and added back GDML through **Freecad tools->Addons Manager**, and had the problem of 0 not in list and remembered, I had add that line suggested by @lambdam94 , and it fixed [...]

The only update I did was via the web to a branch called Quanta - Which had errors.

My new machine has now arrived and I am now able to test changes.

Have pushed updates to master and hopefully things will be better.
Still have to fix the worldbox being opaque/Solid, suggest you select and click on cycle icon before adding Part/Volume. 

Let me know if there are still problems.

Thanks for your patience, just a bit of bad luck that my machine had died just as you found the bugs.

KeithSloan, 2020-05-22:
Okay - latest master should fix the worldbox being shaded. If you could confirm all now fixed so that I can close. Thanks

quantatanu, 2020-05-22:
Yes, it's perfect now, thanks!
Btw, about the material, when I add an object, in the property section, I see just air and steel like before, but in the appearance section I am able to choose various materials, as I was also able to do this before, I am guessing the choice of materials through the appearance option should work too?

[image]

KeithSloan, 2020-05-23:
Thanks for the heads up on Material. Your scene capture shows the property for selecting a FreeCAD material ( As would be used by FEM etc ) which at the current time is different from GDML material. 

I will have to put thinking hat on as to the best way to handle what at the moment are two different  definitions of material.

I am currently working on materials, there is now a new branch materials. It is still under active development, but with it you should be able to select the Materials group and export to an XML file. The next step is to be able to import such an XML file. The idea being that you could export the materials from an existing file like one of the CERN examples and use in a new file that you are creating.

KeithSloan, 2020-05-24:
The latest version of branch materials allows one to select materials and export an XML file of just material definitions. The XML file can now be imported into an existing file.

closed 2020-05-26 by KeithSloan

quantatanu, 2020-06-06:
Thanks a lot. I was unavailable for the past few days and couldn't check your response or apply the solution. I will try it out soon.

KeithSloan, 2022-02-28:
For ways to add more materials see issue #73.

lambdam94, 2022-03-01:
Hello keith
Sorry for the delay

I think users want to manage a local material database. I.e. one more
database.
We need to Add and delete materials
Delete is already included in Freecad GI.
You need to manage the 'add material'  based on the G4 single atom for the
first step ( it Will complete most ofthe material question).
Add material needs:
density
List of g4 atom and ratio (number or mass)

Regards
Dam