KeithSloan/GDML #13: Unable to import complex external GDML files

quantatanu, 2020-05-26:
Hi Keith,

I was trying to import this geometry:

[image]

that was created using  [GeGeDe](https://github.com/DUNE/gegede), it has layers of active material and the file is ~32MB, it takes forever and never imports.

[image]

I was trying to add such an active layer to another such complex geometry by importing it to Freecad+GDML. 

Is this a bug or just it's early for Freecad+GDML and we can expect such features of importing complex GDMLs in the near future?

Thanks

KeithSloan, 2020-05-27:
I also have similar experience trying to open the CERN sample file Alice which is only 12.6Mb
which is why I added the function to scan rather than import. Scan just looks at high level
volumes and then you can expand them individually.

Have tried starting FreeCAD with output to a temp file i.e. freecad > /tmp/trace and
in a separate window run a tail on /tmp/trace. Get the impression that things slow down
as it imports more volumes.

Think we need to do some profiling as per [https://forum.freecadweb.org/viewtopic.php?f=10&t=44785](url)

You could try in Utils/gdml2step.py from a command window, leave it running
overnight. There is also gdml2FC.py but not sure that this works currently.

Maybe things would be faster using the Geant4 parser rather than lxml.

I did start a Geant4 version of the workbench [https://github.com/KeithSloan/FreeCAD_Geant4](url) Cannot remember what state it is in, but there a couple of batch type scripts
Would be interested to know if scanGDML.py works on your file and how long it takes?

The workbench does require you to install Geant4 and build the asssocited Python  library

Another thing we might try is to construct a script that removes all the GDMLShared.trace calls from all the modules and see if that helps.

quantatanu, 2020-05-28:
For `gdml2step.py`, here's the log:

```
python3 gdml2step.py complex_input.gdml.gdml out.gdml.gdml 
FreeCAD 0.18.4, Libs: 0.18.4R
Importing : complex_input.gdml.gdml
FreeCAD Gui not present.
Trace set to : True
Print Verbose : True
Import GDML file : complex_input.gdml.gdml
ImportGDML Version 0.3
running with lxml.etree 
Call set Define
Process Constants
Position Not processed & Displayed
Expressions Not processed & Displayed
Rotations Not processed & Displayed
Process quantity (not taken into account in this version)
{'name': 'Default', 'version': '0'}
Materials List :
['SiO2', 'FeO', 'Al2O3', 'MgO', 'CO2', 'CaO', 'Na2O', 'P2O5', 'TiO2', 'Fe2O3', 'Rock', 'Dirt', 'Air', 'Air35C', 'Bakelite', 'Honeycomb', 'C3H6', 'RadiatorBlend', 'RadiatorBlendNewConf', 'RadiatorBlend2', 'stGas_Xe', 'stGas_Xe19', 'Water', 'ArgonTarget', 'Aluminum', 'Copper', 'MagnetCoil', 'CarbonFiber', 'stGas_Ar', 'stGas_Ar19', 'Kapton', 'Tungsten', 'Gold', 'Iron', 'Graphite', 'Calcium', 'Steel', 'polycarbonate', 'polyurethane', 'sttFrameMix', 'fibrous_glass', 'CH2FCF3', 'C4H10', 'SF6', 'rpcGas', 'epoxy_resin', 'Epoxy', 'Glass', 'FR4', 'PVT', 'Scintillator', 'Oil', 'Lead', 'LAr', 'GAr', 'KLOEEcal', 'PVF', 'Kevlar', 'NomexHoneycomb', 'Mylar', 'KevlarPrepreg', 'Methane', 'Ethane', 'CF4', 'CH2F2', 'DME', 'Nitrogen', 'HP_Ar', 'HP_ArCO2', 'HP_ArCH4', 'HP_ArCF4', 'NoGas']
world : volWorld
ParseVolume : volWorld
expandVolume : volWorld
Positions : px 0 py 0 pz 0
solidref : World
box
materialref : NoGas
createSolid box
px : 0
CreateBox : 
{'lunit': 'cm', 'name': 'World', 'x': '60000.0', 'y': '60000.0', 'z': '60000.0'}
vval : 60000.0
chkval : 60000.0
return value : 60000.0
vval : 60000.0
chkval : 60000.0
return value : 60000.0
vval : 60000.0
chkval : 60000.0
return value : 60000.0
GDMLBox init
Traceback (most recent call last):
  File "gdml2step.py", line 22, in <module>
    FreeCAD.loadFile(iname)
OSError: 'NoneType' object has no attribute 'ShapeColor'

```

For the `gdml2FC.py`, the log:

```
python3 gdml2FC.py complex_input.gdml out2.gdml 
FreeCAD 0.18.4, Libs: 0.18.4R
Importing : complex_input.gdml
FreeCAD Gui not present.
Trace set to : True
Print Verbose : True
Import GDML file : complex_input.gdml
ImportGDML Version 0.3
running with lxml.etree 
Call set Define
Process Constants
Position Not processed & Displayed
Expressions Not processed & Displayed
Rotations Not processed & Displayed
Process quantity (not taken into account in this version)
{'name': 'Default', 'version': '0'}
Materials List :
['SiO2', 'FeO', 'Al2O3', 'MgO', 'CO2', 'CaO', 'Na2O', 'P2O5', 'TiO2', 'Fe2O3', 'Rock', 'Dirt', 'Air', 'Air35C', 'Bakelite', 'Honeycomb', 'C3H6', 'RadiatorBlend', 'RadiatorBlendNewConf', 'RadiatorBlend2', 'stGas_Xe', 'stGas_Xe19', 'Water', 'ArgonTarget', 'Aluminum', 'Copper', 'MagnetCoil', 'CarbonFiber', 'stGas_Ar', 'stGas_Ar19', 'Kapton', 'Tungsten', 'Gold', 'Iron', 'Graphite', 'Calcium', 'Steel', 'polycarbonate', 'polyurethane', 'sttFrameMix', 'fibrous_glass', 'CH2FCF3', 'C4H10', 'SF6', 'rpcGas', 'epoxy_resin', 'Epoxy', 'Glass', 'FR4', 'PVT', 'Scintillator', 'Oil', 'Lead', 'LAr', 'GAr', 'KLOEEcal', 'PVF', 'Kevlar', 'NomexHoneycomb', 'Mylar', 'KevlarPrepreg', 'Methane', 'Ethane', 'CF4', 'CH2F2', 'DME', 'Nitrogen', 'HP_Ar', 'HP_ArCO2', 'HP_ArCH4', 'HP_ArCF4', 'NoGas']
world : volWorld
ParseVolume : volWorld
expandVolume : volWorld
Positions : px 0 py 0 pz 0
solidref : World
box
materialref : NoGas
createSolid box
px : 0
CreateBox : 
{'lunit': 'cm', 'name': 'World', 'x': '60000.0', 'y': '60000.0', 'z': '60000.0'}
vval : 60000.0
chkval : 60000.0
return value : 60000.0
vval : 60000.0
chkval : 60000.0
return value : 60000.0
vval : 60000.0
chkval : 60000.0
return value : 60000.0
GDMLBox init
Traceback (most recent call last):
  File "gdml2FC.py", line 22, in <module>
    FreeCAD.loadFile(iname)
OSError: 'NoneType' object has no attribute 'ShapeColor'
```

Although I couldn't find `scanGDML.py`. But when I import the `gdml` file on the `GUI`, I see 2 options: `import` and `scan`, scan almost immediately works but everything is empty.. except a wireframe `EorldBox` nothing is visible.

KeithSloan, 2020-05-28:
Please could you make the file available on the web somewhere? Dropbox/google drive
Or do you have, or could install TeamViewer? So we could do a file transfer?
If you send me an email Keith[at]sloan-home.co.uk we can set things up.

scanGDML.py is in a different repository https://github.com/KeithSloan/FreeCAD_Geant4
and would need the installation of Geant4.

KeithSloan, 2020-05-28:
Are you sure NoGas is defined as one of your materials? 
The Materials List in the trace is truncated at 80 char so I cannot check from the info provided.

DraTeots, 2020-12-12:
I think I have the same issue

Here is the file. It was created by Geant4 (10.6.2)

[g4e_output.geo.gdml.zip](https://github.com/KeithSloan/GDML/files/5682572/g4e_output.geo.gdml.zip)

KeithSloan, 2020-12-19:
Sorry DraTeots only just noticed your comment, must reduce the repro's I am watching so that I don't miss ones for mine, wish GitHub treated ones that are mine differently from ones I wish to watch.

Anyway digress. Downloaded your file and a quick grep for volumes i.e. grep volumes g4e_output.geo.gdml gives
grep volume g4e* | wc
   17035   33844  869914
So it looks like your file has 17035 volumes

lhcbvelo.gdml only has 437 volumes and that loads okay
Alice.gdml has 49780 and I cannot successfully load that

What you can is rather than load the file, select the 'Scan Vol'  option. This loads quite quickly and you can then
click on World_logic, you then get a list of volumes that have not been expanded. You can select one of these
and then either scan to a full depths or just one level.

I am working on a batch python script to parse a gdml file and build a directory structure of volumes see https://github.com/KeithSloan/GDML_Command_Line_Utils and program buildDirStruct.py in branch getSolids
but have hit what I believe is a bug either in my code or in the lxml library.

There is also extractVol.py which you can run command line to extract a Volume and all its sub volumes.

KeithSloan, 2020-12-22:
> I think I have the same issue
> 
> Here is the file. It was created by Geant4 (10.6.2)
> 
> [g4e_output.geo.gdml.zip](https://github.com/KeithSloan/GDML/files/5682572/g4e_output.geo.gdml.zip)

Okay see issue 13