KeithSloan/GDML #18: improvement: expand volume without lxml

lambdam94, 2020-06-15:
Hello Keith,

normaly, we cannot expand volume without lxml.
But I tried, to replace line 542 in GDMLcommand.py by the following lines and it seems working well

Regards
Dam

               #import lxml.etree  as ET 
               try :
                     from lxml.etree as ET
               except ImportError:
                     try:
                         import xml.etree.ElementTree as ET
                     except ImportError:
                         print('pb xml lib not found')
                         sys.exit()

lambdam94, 2020-06-15:
Thanks Keith!
Regards
Dam

lambdam94, 2020-06-15:
Hello again Keith,

I give you a bad advice,
could you please replace 
# line 544
                 from lxml.etree as ET
by
# line 544
                 from lxml.etree import ET

Sorry and thanks!
Regards
Dam

KeithSloan, 2020-06-15:
Funny I nearly questioned that when I saw it

lambdam94, 2020-06-15:
Thanks!
Regards
Dam

closed 2020-06-15 by lambdam94