KeithSloan/GDML #22: position ref

lambdam94, 2020-07-07:
Hello Keith

in position definition
  <position name="P-positionz" unit="mm"  z="0.000075" />
and 
  <position name="P-positionz" unit="um"  z="0.075" />
and
  <position name="P-positionz" unit="nm"  z="75.0" />
should give the same thing.

But only, unit="mm" gives the good position.
It seems that unit is not taken into account.
 Freecad IHM shows only 2  numbers after comma, 
so "0.000075mm" appears "0.00mm" (even if 000075mm is writen in the file).

Could you take a look?
Thanks!

Regards
Dam

KeithSloan, 2020-07-07:
Not understanding - please could you clarify

in position definition

and

and

should give the same thing.

KeithSloan, 2020-07-07:
Is the issue any object with a non default position is not exported correctly unless units = mm?

KeithSloan, 2020-07-07:
Have tried changing the position of a GDMLBox via FreeCAD preferences and even if I enter 1M it changes to 1000mm. 
How are you changing position to get a non mm value.

lambdam94, 2020-07-08:
Hello Keith

sorry about the bad message, 
 the preview is a good option...

It will be better with some exemples.
It is a small box in a world with a delta z =5um.
 If we write a "delta z =5e-6 m", the delta z is considered equal to 0.

exempleDelta5um is ok
exempleDelta5e-6m is nok

[exempleDelta5um.gdml.txt](https://github.com/KeithSloan/GDML/files/4891299/exempleDelta5um.gdml.txt)
[exempleDelta5e-6m.gdml.txt](https://github.com/KeithSloan/GDML/files/4891296/exempleDelta5e-6m.gdml.txt)

In exempleDelta5e-6m.gdml, it seems that "m" is considered "mm".

Regards
Dam

KeithSloan, 2020-07-12:
Just come across this which may or may not be relevant - Anyway posting link here just in case [https://forum.freecadweb.org/viewtopic.php?f=10&t=48451](url)

lambdam94, 2020-07-14:
Hello Keith,

Your link seems to be a good solution in order to manage the different type of length.

I found the solution of this issue in def getMult(fp) in GDMLShared.
Here 'unit' in fp.attrib is not well read. 
hasattr(fp,'unit') and   unit = fp.unit don't give the same thing.

If you add these 2 lines (on lines147), it seems working
    elif 'unit' in fp.attrib : unit = fp.attrib['unit']
    elif 'lunit' in fp.attrib : unit = fp.attrib['lunit'] 

cf. a solution:
[getMult.txt](https://github.com/KeithSloan/GDML/files/4919404/getMult.txt)

Could you please add this to main code?

Regards
Dam

KeithSloan, 2020-07-14:
Have updated - hopefully now okay.

lambdam94, 2020-07-15:
Yes, It is ok. Thanks!

Regards
Dam

KeithSloan, 2020-07-15:
Tried importing lhcbvelo.gdml and these changes cause errors with Polycone

str object has no attribute attrib

GDMLShared.py line 147

KeithSloan, 2020-07-15:
Code changed to 

`
if hasattr(fp,'lunit') :
        trace('lunit : '+fp.lunit)
        unit = fp.lunit
    elif hasattr(fp,'unit') :
        trace('unit : '+fp.unit)
        unit = fp.unit
    elif hasattr(fp,'attrib') :
        if 'unit' in fp.attrib :
           unit = fp.attrib['unit']
        elif 'lunit' in fp.attrib :
           unit = fp.attrib['lunit']
    else :
        return 1
`

lambdam94, 2020-07-16:
With the hasattr(fp,'attrib') test, it seems like before!
Regards
Dam

KeithSloan, 2020-07-16:
"With the hasattr(fp,'attrib') test, it seems like before!"

Like before?

It behaves like before your code is added ? i.e. fixes nothing.

lambdam94, 2021-01-18:
Fix in the today version.

closed 2021-01-18 by lambdam94