KeithSloan/GDML #28: improvement: enumeration for lunit - length quantity

lambdam94, 2020-07-15:
Hello Keith,

Freecad can deal with 7 different length types ['nm','um', 'mm','cm', 'dm','m', 'km'] 
(cf. https://wiki.freecadweb.org/Quantity).
So we can propose a enumeration for lunit in each part definition.

global LengthQuantityList
LengthQuantityList =  ['nm','um', 'mm','cm', 'dm','m', 'km']
# cf definition https://wiki.freecadweb.org/Quantity

I tried to do this for GDMLBox in GDMLObject.py.

I add at the beginning of the file:

```
global LengthQuantityList
LengthQuantityList =  ['nm','um', 'mm','cm', 'dm','m', 'km']
# cf definition https://wiki.freecadweb.org/Quantity

def setLengthQuantity(obj, m) :
    #print('setMaterial')
    if LengthQuantityList != None :
        obj.lunit = LengthQuantityList
        obj.lunit = 0
        if len(LengthQuantityList) > 0 :
            if not ( m == 0 or m == None ) : 
                obj.lunit = LengthQuantityList.index(m)
    else :
        obj.lunit = 2

```

And line 355
I replace 
`      #obj.addProperty("App::PropertyString","lunit","GDMLBox","lunit").lunit=lunit`
by:
```
      obj.addProperty("App::PropertyEnumeration","lunit","GDMLBox","lunit")
      setLengthQuantity(obj, lunit)
```
We should do quite the same for each class - object definition:
GDMLArb8: line ~275
GDMLCone: line ~425
GDMLElCone: line ~516
etc.

Do you think it is a good idea to add this?
Thanks!
[GDMLObjects.py.txt](https://github.com/KeithSloan/GDML/files/4923944/GDMLObjects.py.txt)

Regards
Dam

KeithSloan, 2020-07-15:
Yes - Good contribution.

Will add

Problem is we really should check all shapes, just in case of errors from typo's

KeithSloan, 2020-07-15:
If you could eye ball the changes in branch enum_lunit before I merge with master. There are a lot of changes so want to avoid errors due to typo's. I will then merge and test.

lambdam94, 2020-07-15:
There is only 2 IndentationError on lines 2155 and 2156.
Otherwise, the 7 GDML objects seem ok.

KeithSloan, 2020-07-15:
Should now be good - have merged into master

lambdam94, 2020-07-16:
Great!
Thanks!
Dam

lambdam94, 2020-07-16:
Perfect!
Thanks!

closed 2020-07-16 by lambdam94