KeithSloan/GDML #34: manual copy install problem

shangjiaxuan, 2021-02-01:
The latest FreeCAD release (19_pre 23756) have the same problem as 18 that the addon manager is broken likely because of GitHub web code update (thus cannot retrieve a list of addons). I tried to manually move the files in freecad/gdml into the Mods folder, and renamed the `__init__.py` and `init_gui.py` to `Init.py` and `InitGui.py` (and also rename the freecad used in code to FreeCAD to match the definition of install). FreeCAD is able to locate and try to load the extension, but fails with error `name 'joinDir' is not defined`. I tried move the definition and such, but failed. How do I fix this problem?

shangjiaxuan, 2021-02-01:
I replaced the init_gui.py with the InitGui.py from the older repository, and fixed a typo where there's a missing comma in exportGDML.py (and replaced lxml with xml.etree.ElementTree, since I was having problem with python resolving etree in lxml, and removed the hint in XMLParser). Now it works fine.

KeithSloan, 2021-02-01:
Please could you let me know where the missing comma was in exportGDML.py

shangjiaxuan, 2021-02-01:
> Please could you let me know where the missing comma was in exportGDML.py

I'm sorry,  I mistook the older un maintained repo for this one. The comma problem was with that repo, along with other problems. I tried to workaround the addon-manager problem and the installed version worked smoothly.

closed 2021-02-01 by shangjiaxuan