KeithSloan/GDML #35: auxiliary info support?

shangjiaxuan, 2021-02-03:
GDML has the element type of auxiliary where user can use it to store some information on detector info, display color, etc. and use that info in Geant4 or other parts. Is there any plan on preserving or even allowing edit of that part? (It seems base Geant4 already can uses that to mark certain volumes as a sensitive detector of a certain name.)

KeithSloan, 2021-02-03:
I will look into it, Do you have a small sample file that shows typical use, that you could provide?

KeithSloan, 2021-02-03:
I have added branch SensDet.

If you import a file with <auxillary auxtype="SensDet" auxvalue=Value>
then the volume/Part will have the SensDet Property added.

You can then change the value via the Property editor.

I will have to think about how to add to a new Volume

shangjiaxuan, 2021-02-03:
The geant4 implementation uses the auxiliary in volume and in userinfo to define some exported data. Region cuts are exported in userinfo->auxiliary with type Region then a list of a few region parameters to export region cuts. The auxiliary in volume with auxtype=SensDet is what you just said. There's also an example of how to use auxiliary info to store some visualization data (display color) without using extensions. I think it's meant to be a generic tree structure that holds whatever the extended use is.
 
Sample files will be sent a while later (exported using base Geant4).

KeithSloan, 2021-02-03:
branch auxcolor for hex rgb values i.e. #RRGGBBAA

shangjiaxuan, 2021-02-04:
[Here](https://github.com/shangjiaxuan/SmallFileShares/tree/master/GDML) is my sample of exported gdml file where the parser is set to export extra data.
[Here](https://github.com/Geant4/geant4/blob/master/examples/extended/persistency/gdml/G04/auxiliary.gdml) is an example in Geant4 examples where SensDet and Color are used. [Here](https://github.com/Geant4/geant4/tree/master/examples/extended/persistency/gdml/G01) are some examples that use extended functionality. It seems the the 
auxiliary.gdml has region info in userinfo auxiliary, but the auxtype name has `region` instead of `Region`(which is hard-coded into the current version).