KeithSloan/GDML #36: BoxCommand failes

stephanlachnit, 2021-02-11:
When I create a new file (with GDML active), and then try to add a Box to worldVOL, this happens:
```
Running the Python command 'BoxCommand' failed:
Traceback (most recent call last):
  File "/home/stephan/.FreeCAD/Mod/GDML/freecad/gdml/GDMLCommands.py", line 159, in Activated
    GDMLBox(obj,10.0,10.0,10.0,"mm",material)

__init__() missing 1 required positional argument: 'colour'
```

The part will be created, but the "Box" in the part is greyed out and not shown. It also doesn't have the properties of the GDMLBox like from the world. I'm using freecad `0.19~pre1+git20210207.a3fb41502b+dfsg-1` on Debian Sid.

KeithSloan, 2021-02-11:
Thanks for reporting - Bug was introduced when adding support for setting colour from GDML import.
Sorry about that - Should now be fixed

stephanlachnit, 2021-02-11:
Jup works, thanks for the quick fix!

closed 2021-02-11 by stephanlachnit