KeithSloan/GDML #37: GDML objects an boolean operations

mat-Sc, 2021-03-10:
First of all I command you on the excellent workbench,

- I tried creating a boolean solid using two primitive gdml objects but got thrown the following exception :   
   <Exception> 'Group' is not part of the enumeration

- Even with the exception i did get a boolean solid that i put inside a new part (marked it for recompute) and put it inside wordVol, and did an export but the resulting gdml file doesn't have the boolean solid inside the solids block.

KeithSloan, 2021-03-10:
Any chance you could save the FreeCAD file and export GDML file and then upload both. Thanks in anticipation.

mat-Sc, 2021-03-13:
> Any chance you could save the FreeCAD file and export GDML file and then upload both. Thanks in anticipation.

Hi, thanks for the replay. 

I'm trying to model an HPGe detector, so far I did get the detector to export and open in Geant4 but as a meshed geometry, considering my limited computational resources doing sims with meshed geos take a considerable execution time. 

I think my issue is that my union solid  has a shapes property instead of the base and tool properties, like the union solid found inside the samples folder.

Your assistance is much appreciated. (I'm running FreeCAD 0.19 in both Ubuntu 16.04, and OpenSUSE LeapFrog 15.4)
 
[image]
[image]
[image]

KeithSloan, 2021-03-13:
Okay I see the problem here. To deal with copies of volumes I hide GDML objects Placements and you set the Placement in the Part ( Equivalent of Physical Volume ). With booleans the Placement of the Tool is with respect to the Base. I have to have a think about dealing with Booleans when user created.

KeithSloan, 2021-03-14:
Okay : Some progress

There is now a branch booleans. i.e. in your installed repro in Mod/GDML directory  run 
     git fetch booleans
     git checkout booleans
On my test file if I select the LV's in the correct order Tube then Box and click on the GDML BoolUnion ( Current icon with an X ) it will create a new volume with a Union.

Have not tried an export but at present you would have to delete the LV-Box and LV-Tube.
Also currently does not work if you select the LV's in the wrong order.

I will try and find some time to do more work on it, but may have to wait till next weekend.

File I am testing with https://www.dropbox.com/s/kfwn3xx0jut337o/BooleanTest.FCStd?dl=0

mat-Sc, 2021-03-15:
I think the problem I was having is that the union operation (selected my objects and clicking on the union icon) was giving me MultiFuse instead of my desired union operation. 

Testing a boolean cut worked just fine, so I did the following in the python console for fuse operation and my boolean solid was able to be exported.

App.activeDocument().addObject("Part::Fuse","Fuse")
App.activeDocument().Fuse.Base = App.activeDocument().GDMLBox_Box
App.activeDocument().Fuse.Tool = App.activeDocument().GDMLCone_Cone
Gui.activeDocument().GDMLBox_Box.Visibility=False
Gui.activeDocument().GDMLCone_Cone.Visibility=False

App.getDocument('Unnamed').getObject('Fuse').ViewObject.ShapeColor=getattr(App.getDocument('Unnamed').getObject('GDMLBox_Box').getLinkedObject(True).ViewObject,'ShapeColor',App.getDocument('Unnamed').getObject('Fuse').ViewObject.ShapeColor)

mat-Sc, 2021-03-15:
Okay, I'll be trying you suggestions in the meantime

KeithSloan, 2021-03-22:
Hopefully now fixed - Please let me know if not.

closed 2021-03-22 by KeithSloan