KeithSloan/GDML #38: Support for variables

AntonFetzer, 2021-03-26:
Does this plugin support user-defined variables?

I have a GDML file in which I use variables to assign the thicknesses in a stack of materials and then calculate the positions of the volumes based on these variables.

The file works in Geant4, but when I try to open it in FreeCad, I get the error message:
<class 'NameError'>: name 'PCBThick' is not defined

With 'PCBThick' being the variable in which I store the thickness of a PCB so that I can calculate the position of the components on it.

[block.txt](https://github.com/KeithSloan/GDML/files/6213645/block.txt)

KeithSloan, 2021-03-26:
Thanks for reporting the issue.

Have done a quick fix to read in variables as constants which should allow you to read and view your files.
Obviously will need more work for export.

Will try and address this Sunday evening

KeithSloan, 2021-03-26:
Okay have separated variables and constants so should now be able to import and export GDML files with variables.

Not sure if it would be possible to use in FreeCAD properties, will have to investigate.

Also need to add facility to add definitions

So will leave this issue Open

AntonFetzer, 2021-03-26:
@KeithSloan 
Wow, I did not expect to get such fast and effective support.
Your update works and I can now import my file.
Thank you very much!

closed 2021-06-22 by KeithSloan