KeithSloan/GDML #40: Problems to import GDML file with many user variables and calculations

PaulyChristian, 2021-03-29:
This is a great module, being able to import (and even modify!) GDML in Freecad is a fantastic perspective !

However ... In my case unfortunately it does not work yet.
I have attached a "parametric" GDML file, which is maybe a bit uncommon in using many variables and calculations.
The motivation behind was to be able to define the resulting geometry using few defines / used variables.

I updated to the recent module version today, which already gave some progress: before the update I got error messages on import that certain user variables are unknown. This seems to be fixed now, however, now i get a long error message in the console ending with: 
<class 'NameError'>: name 'false' is not defined

My hope is, that this is maybe a minor thing to fix in the GDM ?
Could you maybe have a quick look at the attached gdml file, and tell me if there is any hope to import such a file in near future, or if i have to look for another way to go ?
Thanks a lot !

[rich_mirror12_457.gdml.txt](https://github.com/KeithSloan/GDML/files/6220665/rich_mirror12_457.gdml.txt)

KeithSloan, 2021-03-29:
Its barfing on
<variable name="Back_cut_enabled" value="false"/>
and would no doubt also barf on the next variable
<variable name="Nose_enabled" value="true"/>
In setting the variables I am executing the following line of python
globals()[name] = eval(value)
and as value is the string 'false' python barfs.
A quick fix might be to change it to 'False' which python understands
I have to check, is 'false' gdml's way of defining what python would define as False?
Not sure how some of your other variable definitions might work out as again would have to have a direct mapping to python.

KeithSloan, 2021-03-29:
Okay I have changed the handling of variables to
try
   globals()[name] = eval(value)
except 
   globals()[name] = value

Which gets you passed the variables, but I then hit a bug with colour and am struggling to see the problem.
Going to ask Dan Lambert as he did a lot of work on handling colour.

Anyway the fix to variables is in branch fixVariables if you want to try i.e. maybe you have some other file you could try.

PaulyChristian, 2021-03-29:
Great, thanks for the quick help !
I just tried to change false->False and true->True, and then got stuck at the same problem you encountered regarding the colour. I will update to your latest version in a minute.

But it loads partially already ! So i am getting hope it might be possible to get it running :-) 
That would be soo useful !
Actually there is no colour statements at all in my gdml file...

KeithSloan, 2021-03-29:
"Actually there is no colour statements at all in my gdml file.."

It is to do with a default situation where no colour is given and seems to be to do with a boolean subtraction, it is a bit weird.
I hope Dan can shed some light

KeithSloan, 2021-03-29:
Okay found a problem, the call to create a Sphere has material.colour when it should have been material, colour.
Should be fixed if you get latest version

Now when I try importing your file seems to be looping :-(.

One way we can try and track this down is by doing a scan rather than importing the file.
i.e. on Open select the option '**Scan Vol**'

Then select a volume that starts Not_Expanded and click on either the scan minimum level or scan full depth icon.
Then by a process of elimination find which volume is causing the problem.

I notice that it gives errors because you are using **pi** in some calculations, so I need to add something on the lines of pi=math.pi or as a quick solution you might add a variable pi=3.142.....

If you could try and track down the cause of the loop, that would help as I have a deadline for a report approaching.

PaulyChristian, 2021-03-30:
Thanks for helping, and in particular for the quick response !
I will try to do as you suggest, and activate the volumes step by step. However, maybe waiting long enough is sufficient already... After about 10 minutes it says "End processing GDML file" and some geometry appears. I have to check how correct it is.

However, i do get the same error message as you regarding constant Pi. But i checked my GDML file, and i dont use pi anywhere in there. Just cos and sin functions.

`File "/common/home/pauly/.FreeCAD/Mod/GDML/freecad/gdml/GDMLObjects.py", line 1552, in createGeometry
    rmax/math.cos(pi-startthetaRad),rmax, spos, \
<class 'NameError'>: name 'pi' is not defined
10:24:07  End processing GDML file`

"pi-startthetaRad" is not in the gdml file.

KeithSloan, 2021-03-30:
Found the problem with pi it was in my code. Should now be fixed.

You may want to try realthunders LinkStage 3 branch, it under development but shows things with transparency
[image]
 so easier to see inside. Screen capture with just a few Volumes expanded.
[image]

PaulyChristian, 2021-03-30:
Excellent !
Import works now. Takes a while if i do the full import (~5 minutes), but once its done it seems to be all there. Thanks for the tip with the FreeCAD assembly3 branch, which enables the transparency option in the "appearance" dialogue. That helps.

Re-Export into gdml still seems to have some problems with the file, but i am still testing...

PaulyChristian, 2021-03-30:
Sorry to bother you again - but i think i discovered another problem, this time  when exporting.
I am loading my file from above, activate/select the master volume "rich_opt", and go to export->asGDML.
Doing that, i obtain a gdml file which is ~30% smaller than the orignial file. But if i try to open it, i do see again the master volume in the tree, but there is nothing in there, and loading only takes fraction of a sec.
Looking into the exported file, i do see a lot of volumes in there. But they somehow dont load.

Am i doing something wrong, or is this maybe still a bug ? Or due to my weird gdml file ?

I attach the exported file...
[rich_mirror12_457_exported.gdml.txt](https://github.com/KeithSloan/GDML/files/6232368/rich_mirror12_457_exported.gdml.txt)

KeithSloan, 2021-03-30:
I looked at and tried to import your exported file. The import is failing because it does not find a solid_ref & material in the volume rich_opt. Wondering if it is a bug in export in that is should be an Assembly rather than a Volume. An Assembly would not have a Solid & material. i.e. Failing to check if what is root volume for export has any GDML objects and if it contains just Part/Logical Volumes it should be set as an Assembly. I will have to do some checking

Note the export will never create an exact copy of an import as a number of entities will end up being treated slight differently.
 But should be the same as far as Geant4 is concerned. For example the export makes up Physical Volume names based on Logical volume rather than what was imported.

KeithSloan, 2021-03-31:
Hand edited version now with assembly [rich_mirror12_457_exported.gdml.txt](https://github.com/KeithSloan/GDML/files/6235004/rich_mirror12_457_exported.gdml.txt)
How does it look?

KeithSloan, 2021-03-31:
You might want to try branch rootAssem.

Code was getting untidy so needed some restructuring, often means new silly bugs get introduced. 
Want to do some more testing before merging.

But would be grateful if you tried,

PaulyChristian, 2021-03-31:
Hej Keith
I tried the new branch "rootAssem", and again tried to load my original file, export the root element, and reload the exported file.
This works now - kind of. But not quite as expected as you see in the attached pictures:
First picture is how the original load looks like, second picture after loading the exported file.
Reimport is much faster (70sec instead of 600sec, using Linux) then loading the original file, but in particular the mirror structure (calculated using sin / cos function sin the gdml) is broken after reload.
Any idea what that could be ?

[image]

[image]

KeithSloan, 2021-03-31:
Well I know of one large problem with export and rotation, GDML and FreeCAD have different systems of axis when defining rotation. I am lead to believe that there are 12 ways of getting Euclid angles from a rotation and FreeCAD was only implementing one. The methods exist is OCC but FreeCAD as not making access. Realthunder main ( Linkstage 3 ) developer has added support, but I don't think the PR made 0.19.1. I don't know why so much of the file has gone missing.
Unfortunately at the moment I only have time for quick fixes, I have to create two reports one by 9th April and by the 7th May and an exam 1st week in May all MSc

KeithSloan, 2021-03-31:
Just tried a load of an exported version of the file and it is giving up the first time it encounters a volume with no solid defined. So there is a bug in that either the volume should be an assembly for other volumes or there should be a solidref and there is not. I will look at adding code to print out more info when a no solidref occurs

PaulyChristian, 2021-03-31:
I dont quite understand : you are saying the original gdml file is faulty ? Could well be... I will have to read a bit about gdml to understand the problem. Though it works fine in GEANT and is regularly used... 
Maybe i will try to have a look with the ROOT GeometryViewer (GeometryManager ?)

Meanwhile : Please dont neglect your examn !
I know, usually its much more fun doing these things then writing reports...

Thanks a lot for all the quick help so far !

KeithSloan, 2021-04-01:
No the problem is in the export of the GDML, for some reason what is produced is not valid. My understanding is that a GDML volume should always contain a solid ( solidref) and a material and the file I am creating with the export is not.

KeithSloan, 2021-04-01:
Flying parts I am sure will be down to handling of rotation different FreeCAD and GDML need to try Realthunders addition for options on ToEuler.

Missing parts - Where there is a copy of a volume the importer correctly creates a FreeCAD Linked Part. There is a bug in that a Linked Part is not being correctly exported i.e. Not following the link to picking up the Solid and Material.

Back to studying

KeithSloan, 2021-04-12:
You might want to check [https://github.com/KeithSloan/GDML/pull/42](url)

Especially if the time of the file import comes down to seconds from Minutes. Might also make import of Alice.gdml feasible :-)

PaulyChristian, 2021-04-13:
Hej Keith
I checked your latest version - at least i hope i did:
I did "git pull" in the .Freecad/Mod/GDML directory, and according to  
status messages there is an update done to exportGDML.py and  
importGDML.py

Loading my model still takes around 10min. On GDML-export and  
subsequent GDML-reimport the model seems to be complete now - i dont  
see any parts missing. But something still goes wrong with the  
re-import of the main mirrors of my model, the orientation is wrong  
after the reimport of the exported file. And few other parts also have  
wrong orientation. See attached pictures, in the original the main  
mirrors are within the main vessel (which is not active in the picture).
[...]
I then updated the asm3 branch of freecad to
"FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210412-glibc2.12-x86_64.AppImage"

That indeed gives an immense speedup, importing the original file now  
takes 9s instead of 10 min!
Unfortunately, still the same problem with the orientations of few  
elements on reimport. And also when turning the model around (after  
correct import of the original), few elements behave "strange" on  
rotating and are shown in wrong perspectives.

So something still seems to be wrong with the Euler angle  
implementation. But ex- and re-import seem to work now.

Please, go on :-) Its getting better and better...

Thanks !
Christian

KeithSloan, 2021-04-13:
To me the import of an export looks okay, are you sure you are using branch link3?
For now you need to use the branch link3 to fix the export issue(and therefore subsequent import)
The reason I have not merged it is because it will only work with one of realthunders new releases.

If you are using link3 and still seeing problems please could you let me know the volume/part names pf the parts with issue and what issue you are seeing.

KeithSloan, 2021-04-13:
Following Lei's advice I find rendering much improved with
"Please make sure to test it using my branch. For your workbench, I recommend you to test with my Daily release with experimental renderer enabled (Preference -> Display -> Render cache -> Experiemental). I have just made a new release, make sure to download the image with Daily keyword."

closed 2021-06-22 by KeithSloan