KeithSloan/GDML #44: Add translations support

luzpaz, 2021-04-19:
https://github.com/FreeCAD/FreeCAD-addons/issues/187

luzpaz, 2024-09-10:
@KeithSloan would you be open to this happening for GDML ?

KeithSloan, 2024-09-10:
Yes would be happy for somebody to add translation support.
Yes we would be happy for somebody to improve the documentation.

KeithSloan, 2024-09-10:
Question re translation. We are finding that we are having to do more with pure Qt code for Dialogs.

For better translation support should one err more to using Qt's Designer, does that offer a better route to translation than straight qt code?

Unbelievably we found we had been using InList and OutList's to navigate the structure of a document and flabbergasted when we found the only way to do this correctly was via Qt rather than FreeCAD API see https://singerlinks.com/2023/11/how-to-iterate-the-freecad-document-tree-using-python/

Slowly building a list of improvements to FC api which would make our lives easier.

luzpaz, 2024-09-11:
Wow, this is a great question. I think it's something that the devs should be aware of. Maybe you could start a discussion in https://github.com/FreeCAD/FreeCAD-translations/issues ?