KeithSloan/GDML #47: Using GDML workbench to create GDML from STEP file using tesselation, GMSH ?

PaulyChristian, 2021-06-14:
Hej Keith
I am back testing / trying out your GDML workbench. This time, i am trying to convert a (fairly simple) step model into GDML by using tesselation. If i am not mistaken, this would be the easiest (so certainly not fastest for simulation ...) way to get the attached step model into a GEANT simulation ?

[shieldingbox.stp.zip](https://github.com/KeithSloan/GDML/files/6651311/shieldingbox.stp.zip)

So far i managed to tesselate the object using "GDML tesselate", and after creating a new part (yellow freecad symbol) i also managed to export the tesselation in .gdml (single file) or .GDML (directory). However, the tesselation seems to be quite fine, finer than necessary for such simple geometries (55000 facets).
Then I tried to use gmsh instead for tesselation ("GMSH tesselate"), hoping this might provide more options, but gmsh is not running from inside Freecad : I always get an error message "No module named 'gmsh'".

I tried to follow your instructions on the github page ("pip install --upgrade --target /usr/bin gmsh; python -m pip install -U pip") but no success so far. I am using linux (mint), not Windows.
I did install gmsh, and i can run "gmsh info" in the command line. I also tried to add the full path to gmsh in preferences/FEM/gmsh, but still no success.

What am i doing wrong ? And is there a way to influence how fine the tesselation is done ?

paulychris, 2021-06-15:
Let me add:
I am using latest Freecad 0.20 Link branch (2021.05.31) under linux mint 20.1 / debian 10, together with latest development version of GDML workbench (link3 branch)...

paulychris, 2021-06-17:
I managed to get a bit further, maybe it helps for others having similar problems:
In order to use GMSH for meshing i had to modify the file 
   ~/.FreeCAD/Mod/GDML/freecad/gdml/GmshUtils.py
and add 
    sys.path.append('/usr/lib/python3/dist-packages')
such that GDML module finds the file gmsh.py

I also had to switch to branch 1.5_Beta, which is the only one were i got the gmsh call working so far.

However - so far i did non manage to change the mesh parameters (coarseness). Parameters m_Remesh, m_curve Len, m_max Len and m_Point Len somehow dont seem to have any effect.

KeithSloan, 2021-06-17:
> I managed to get a bit further, maybe it helps for others having similar problems:
> In order to use GMSH for meshing i had to modify the file
> ~/.FreeCAD/Mod/GDML/freecad/gdml/GmshUtils.py
> and add [...]

This should not be necessary and is not a generic solution.
It looks like you have not installed the Gmsh python Module where FreeCAD can find it.
See the README **Required python libraries - lxml & gmsh**

As to using Gmsh

Try branch Gmsh - Still need to do quite a lot of tidying and don't know why when I select the World Volume and import the STEP file it does not end up in the world volume.

Anyway with this branch the Gmsh options appear in the Task Window before being actioned.
I Tessellated each Part in turn, closing the Task options before Tessellating the next Part.

You/We need to resolve the the Placement in the World Volume and when you are happy with the amount of Tessellation
delete the Parts but leaving the Tessellated Parts before exporting to GDML by selecting the World Volume.

There maybe problems with the GDML if using GEANT, there is a branch called Geant4 which is an attempt to make Geant4 happy, but it is untested.

[Archive.zip](https://github.com/KeithSloan/GDML/files/6669697/Archive.zip)

KeithSloan, 2021-06-18:
Some other workarounds until I manage to improve the workbench.

Goto GDML workbench
Open New ( To get basic GDML structure )
Import STEP file
Goto Mesh Workbench
For each of the Parts Mesh the Part
Then 'Decimate' the mesh ( The Decimate facility should simpolify the mesh
Goto GDML workbench
Use Mesh 2 Tess ( select then icon) to convert Mesh to Tessellated
Delete Meshs
Select World Volume and Export to GDML

If you use the Gmsh branch to Tessellate the Parts try
Convert the Tessellated Objects to Mesh - Tess 2 Mesh
Goto Mesh Workbench and Decimate the Meshes
Convert Decimated Meshs to Tessellated
Delete Meshs
Export to GDML

I don't know if your three parts are all the same material,
if yes you could try using the Part workbench to create a compound
then Mesh or Tessellate the compound

KeithSloan, 2021-06-18:
> This should not be necessary and is not a generic solution.
> It looks like you have not installed the Gmsh python Module where FreeCAD can find it.
> See the README **Required python libraries - lxml & [...]
FreeCAD does not access Python modules in the 'normal' location but uses a different location.
There is a valid reason for this but I don't remember what it is.

Your amendment is adding a standard location to GmshUtils, it would be better to install
the Mesh Python module where FreeCAD looks for such things - see README

paulychris, 2021-06-18:
Hej Keith
Thanks again for all the support.
 I managed now to install the latest GMSH using the python method (pip install --upgrade gmsh). Before i used (in parallel) the gmsh package from my distro, and there were obviously some conflicts.
 However, i still have to adapt the GmshUtils.py, since on my Linux machines the automatic installation path of Gmsh.py is
/usr/local/lib/python2.7/site-packages/gmsh-4.8.4-Linux64-sdk/lib on my debian10 machine, and
/usr/local/lib/python3.8/site-packages/gmsh-4.8.4-Linux64-sdk/lib on Linux Mint 20.

But then it works, in particular the gmsh branch. Very nice  !
However, i can only configure the gmsh tesselation finer. A more coarse tesselation than with the standard settings i never managed so far.
I always tesselate all 3 objects together (they will be same material), that is no problem.

I also wanted to try to use the mesh workbench to decimate the mesh as you suggested, but converting the tesselation to mesh somehow does not work: i always get error message "Tesselated2Mesh is not defined" in GDMLCommands.py, line 982 :-(

But all in all its pretty impressive. I also managed to export back to gdml (somehow its a bit picky here). Right now we are trying if the import in GEANT4 works. 
What exactly are the problems you mention here ? Just slight changes in the gdml file structure ?

KeithSloan, 2021-06-18:
Could I trouble you to in FreeCAD python console enter
````
 import sys
 print(sys.path)
````
and post the output here

Tessellation options
If I select your Part1 and use Gmsh Tessellate with the defaults that get prompted I get 112 Vertex 224 Facets
If I change minimum length to 400 I get 108 Vertex 216 Facets so slightly courser.
If I change to minimum length 500 I get the same as 400.

Todate: I have not found a way of getting Gmsh to produce a really coarser mesh, it seems to have a minimum.
 It maybe worth asking Gmsh?  The forum got retired some time ago but they answer questions if you raise an issue at [https://gitlab.onelab.info/gmsh/gmsh/-/issues](url)

Tesselated2Mesh is not defined - Not sure how that slipped in. The function is in GmshUtils and an import was missing.
Try latest version of Gmsh branch.

Note: The latest version closes the Task Window when you mouse leaves, would like to find a better solution but cannot
find one. The problem with not closing is, if the user goes off and selects a different Object it does not match the one the task window is working on. Would be interested in how you find it.

For potential issues with GEANT4 import see [https://geant4-forum.web.cern.ch/t/can-i-run-a-gdml-file-that-contains-gdml-file-names-of-other-detector-part/5312/18](url)

It seems Geant4 wants to check the GDML import against a GEANT4 specific xsd. I am led to believe that the xsd can be a local copy or a version on the web. The geant4 branch is an attempt to fix for web access. Would really like some feedback it it works. At the end of the day its a change to the initial gdml tag.

PaulyChristian, 2021-06-18:
Hey Keith
New Version works like a charm !
I can export now to mesh, can decimate in the mesh workbench, ... Can even nicely compare the tesselated shape (in display shaded mode) with the original step geometry...
Really nice !
Automatically removing the options window helps to avoid error messages compared to before.

What is xsd in Geant ?

Here is my sys path (No idea why it is so long)
>>> print(sys.path)
['/home/pauly/.FreeCAD/Mod/Assembly4', '/home/pauly/.FreeCAD/Mod/Defeaturing', '/home/pauly/.FreeCAD/Mod/GDML-Gmsh', '/home/pauly/.FreeCAD/Mod/GDML-Gmsh.zip', '/home/pauly/.FreeCAD/Mod/BOLTSFC', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/asm3', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/appimage_updater', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Web', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Tux', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Test', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/TechDraw', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Surface', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Start', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Spreadsheet', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Sketcher', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Show', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Robot', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/ReverseEngineering', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Raytracing', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Points', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Path', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/PartDesign', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Part', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/OpenSCAD', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/MeshPart', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Mesh', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Measure', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Material', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Inspection', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Import', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Image', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Idf', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Fem', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Drawing', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Draft', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/Arch', '/tmp/.mount_FreeCAXb4mzf/usr/Mod/AddonManager', '/tmp/.mount_FreeCAXb4mzf/usr/Mod', '/tmp/.mount_FreeCAXb4mzf/usr/lib', '/tmp/.mount_FreeCAXb4mzf/usr/lib64', '/tmp/.mount_FreeCAXb4mzf/usr/Ext', '/tmp/.mount_FreeCAXb4mzf/usr/bin', '/tmp/.mount_FreeCAXb4mzf/usr/lib/python39.zip', '/tmp/.mount_FreeCAXb4mzf/usr/lib/python3.9', '/tmp/.mount_FreeCAXb4mzf/usr/lib/python3.9/lib-dynload', '/tmp/.mount_FreeCAXb4mzf/usr/lib/python3.9/site-packages', '/tmp/.mount_FreeCAXb4mzf/usr/Macro']
>>>

KeithSloan, 2021-06-18:
> Hey Keith
> New Version works like a charm !
> I can export now to mesh, can decimate in the mesh workbench, ... Can even nicely compare the tesselated shape (in display shaded mode) with the original [...]

You are using a version of FreeCAD 0.20 right?

These paths look like they are set for Python3.9, was not aware that FreeCAD had upgraded from Python3.8
If you post your FreeCAD help info what version of Python is it reporting. (FreeCAD | about | copy to clip board | paste)

xsd is a XML thing, it defines the rules for the XML definitions in this case GDML. The/a parser will check that the GDML follows the defined rules.

The start of the GDML file is something like
````
<?xml version='1.0' encoding='ASCII'?>
<gdml xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNameSpaceSchemaLocation="http://service-spi.web.cern.ch/service-spi/app/releases/GDML/schema/gdml.xsd">
````
This tells any XML parser where to find the rules definition.

The lady in India who had the problem was getting valid XML/GDML reported as invalid as GEANT4 was not finding the right xsd file, it appears to be picky and I don't know why.

KeithSloan, 2021-06-26:
Hi Pauly

Have you had any success with exporting GDML for GEANT4?

Also as an aside have you tried Daily's clipping plane feature with your model?
[image]

PaulyChristian, 2021-06-27:
> ![Image 26-06-2021 at 23

Hej Keith
Unfortunately not yet, but i am still working on it.
I tried to open the gdml in tGeometryManager from ROOT, but got error message on first try, that type "Tesselated" is unknown.
But that might have been some old root install...
I am currently working on it, and report back once i know more.
Christian

KeithSloan, 2021-06-27:
> Hej Keith Unfortunately not yet, but i am still working on it. I tried to open the gdml in tGeometryManager from ROOT, but got error message on first try, that type "Tesselated" is unknown. But that [...]

Did you mean 'Tessellated'? If I am outputting 'tesselated' rather than 'tessellated' that is a bug.

KeithSloan, 2021-06-27:
Also 'tessellated' is NOT supported by ROOT only Geant4.

From the GDML guide
3.4.26 Tessellated solid
NB: supported only by Geant4

PaulyChristian, 2021-06-27:
i will check that tomorrow. But probably that would be to easy an explanation :-)

KeithSloan, 2021-08-10:
I had problems with Gmsh and now cannot installed the latest python library version of Gmsh :-(

The initial problem with the standard FreeCAD mesh'er producing large numbers of facets and vertex is due to the size of the imported STEP file, if you look at it compared to the WorldBox it is a lot larger, even if you set its size to 2000 x 2000 x 2000 your part looks larger, maybe scaling would help, but as it is the default mesh options produce a large mesh

Using the mesh workbench  (uses a different interfaces ( c++ to Gmsh )) I get using defaults 956 faces, but there is a problem see the bottom right part of mesh is attached file
[shieldingbox.FCStd.txt](https://github.com/KeithSloan/GDML/files/6964708/shieldingbox.FCStd.txt)
( Need to remove txt from file name )

Using Decimate on the GDML Tessellated Object with defaults reduced mesh size to 172 vertex 339 facets
[shieldingboxDecimated.FCStd.txt](https://github.com/KeithSloan/GDML/files/6964719/shieldingboxDecimated.FCStd.txt)