KeithSloan/GDML #55: Exporting GDML with tessellated volumes hangs FreeCAD

alowell-ssl, 2021-11-01:
Hi,

I followed your instructions to open a new GDML project.  I then create a part with a body that I form from extruding a sketch (using Part Design WB).  I am able to tessellate the body, and then drag the part into the World Volume part.  When I try to export to GDML, FreeCAD hangs.

I tried creating a geometry with the simple GDML volume tools and I was able to successfully export a GDML in that case.  So, it seems that the problem is when I try to include tessellated bodies.

I'm using:
FreeCAD-asm3-Daily-Conda-Py3-Qt5-20211015-glibc2.12-x86_64.AppImage

KeithSloan, 2021-11-01:
Thanks for reporting.

Please could you supply a problem file - see email address at bottom README.md, replacing [dot] with a dot and [at] with an @.

alowell-ssl, 2021-11-02:
Fixed, thanks very much :)

closed 2021-11-02 by alowell-ssl