KeithSloan/GDML #71: tessellating stl geometry

aminetec, 2021-12-14:
Greetings,

I imported the included stl geometry and tried the tessellate selected object option but nothing happened.

[compare80.stl.txt](https://github.com/KeithSloan/GDML/files/7715261/compare80.stl.txt)
[brain.FCStd.txt](https://github.com/KeithSloan/GDML/files/7715271/brain.FCStd.txt)

KeithSloan, 2021-12-15:
You have to activate the workbench and open a new file which creates the GDML structure. Import the STL, then use the Mesh 2 Tess icon

[image]

FreeCAD structure needs to look like. You can drag things around in Tree View
[image]

This is the Tessellated object in FreeCAD
[image]

The is the file exported as GDML ( You need to remove the txt -used as github does not allow gdml
[Compare80-worldVOL.gdml.txt](https://github.com/KeithSloan/GDML/files/7717465/Compare80-worldVOL.gdml.txt)

Screenshot of object loaded into Geant4
[image]

Thanks for the issue I will look at changing MeshTess so that there is less to do with creating the structure.

KeithSloan, 2021-12-15:
Number of questions
1) The file is titled brain, is this like a section of a brain and do you have a number that you need to load into Geant4?
2) What sort of material will you be specifying?
3) Would being able to specify a materials using Geometry  from Text help [https://geant4.web.cern.ch/sites/geant4.web.cern.ch/files/geant4/collaboration/working_groups/geom…](url)
4) Not tested in a while but a OBJ files can have Quad & Tri Faces the same as GDML Tessellated object, the workbench should offer the facility to go straight to a Tessellated Object if you try and import an OBJ file, would you like to see STL files handled the same. Could you have just as easily created the file as OBJ rather than STL?
Thanks

aminetec, 2021-12-15:
Thank your for your response, 
Yes I have 8 sections of a brain as separate stl files 
I'm using the material G4_A-150_TISSUE