KeithSloan/GDML #72: Boolean operations involving Torus

aminetec, 2021-12-21:
Greetings,

I'm trying to make a GDML geometry of an HPGe detector. My parts include a torus and when I try a union operation the resulting union part have two union solids instead of one.

Any help is appreciated. Thank for your time.

[hpge_sperated.FCStd.txt](https://github.com/KeithSloan/GDML/files/7756316/hpge_sperated.FCStd.txt)
[hpge_union.FCStd.txt](https://github.com/KeithSloan/GDML/files/7756318/hpge_union.FCStd.txt)

KeithSloan, 2021-12-21:
Please can you check the attached file, Is it nearer to what you are trying to achieve?

If yes, then to achieve a Union you need to select the logical Volumes (Parts in FC) and then use the GDML union (icon in toolbar)

With GDML Unions the second part is placed relative to the first. The Placement information is in the Logical Volumes, hence why you need to select them rather than the GDML objects and use the GDML Booleans rather than FreeCAD booleans.

Not sure how things could be made better, open to suggestions on how documentation on this could be improved.

[hpge_My_Union FCStd.txt](https://github.com/KeithSloan/GDML/files/7756537/hpge_My_Union.FCStd.txt)

KeithSloan, 2021-12-21:
I exported hpge_MyUnion and found some problems with materials. Checking your flle there materials seems to have been messed up some how. It had multiple sets for Geant4 and some Geant4 materials in the non Geant4 sections.

Tidied up the mate
[hpge_MyUnion_CleanFCStd.txt](https://github.com/KeithSloan/GDML/files/7756614/hpge_MyUnion_CleanFCStd.txt)
rials and loaded into Geant4

[hpge_MyUnion_Clean-world0x55d123cff820.gdml.txt](https://github.com/KeithSloan/GDML/files/7756616/hpge_MyUnion_Clean-world0x55d123cff820.gdml.txt)

[image]

aminetec, 2021-12-21:
Thank you for the speedy response.

I'm trying to achieve the following 
[image]
I did this with an earlier version of the toolkit but following the same step with selecting the logical volumes with placement it seem to not produce the same results this is the freecad file i did a while back 
[hpge_union.FCStd.txt](https://github.com/KeithSloan/GDML/files/7757291/hpge_union.FCStd.txt)

KeithSloan, 2021-12-21:
Okay there is a problem with a Boolean operation on another Boolean operation.
Need a clear head so will be tomorrow at least.

aminetec, 2021-12-21:
Thank you immensely

KeithSloan, 2021-12-21:
Try the latest version
1) Hopefully has fixed the problem with a boolean operation which involves a boolean
2) Add torus command/icon ( currently an X) SVG icon to follow.

Please let me know how you get on - Thanks

aminetec, 2021-12-22:
Thank you it worked perfectly

aminetec, 2021-12-22:
The double union geometry worked great in FreeCAD but the exported gdml file is not right, only containing one union operation.

[HPGe_union.FCStd.txt](https://github.com/KeithSloan/GDML/files/7762874/HPGe_union.FCStd.txt)
[HPGe_union.gdml.txt](https://github.com/KeithSloan/GDML/files/7762875/HPGe_union.gdml.txt)

KeithSloan, 2021-12-22:
Are you able to try a branch?

I have pushed a new branch called **newExport** which has restructured the code

 I don't want to merge as I still need to sort out multifuse and a number of things before tidying up and removing redundant code, which will have to wait till after the holidays.
 
 Your File has the materials still messed up a bit as you have multiple Geant4 groups in Materials. Do you know how this is happening? Are you importing an XML file with Materials? In which case I need to fix, but it would be helpful to know how your materials get into the state
[HPGe_union4_Clean.FCStd.txt](https://github.com/KeithSloan/GDML/files/7765041/HPGe_union4_Clean.FCStd.txt)
[HPGe_union4_Clean-worldVOLgdml.txt](https://github.com/KeithSloan/GDML/files/7765042/HPGe_union4_Clean-worldVOLgdml.txt)

aminetec, 2021-12-22:
Sorry I lack the technical knowledge to answer your questions, for the export I just select the world volume and name the file with the .gdml extension. when it comes to the materials I just edit them materials in the resulting file. 

But I'll try the new branch hopefully it'll fix the problem.

Again, thank you for your effort.

KeithSloan, 2021-12-23:
Has now been merged.

aminetec, 2022-01-15:
Hi,

Any news on the fix of the double boolean operation issue.

KeithSloan, 2022-01-15:
> Hi,
> 
> Any news on the fix of the double boolean operation issue.

Thought it was fixed and the new branch merged.

KeithSloan, 2022-01-15:
Sorry about this
Weird somewhere along the line I changed the code to insert booleans at the start of **solids** for some reason.
Using you test file seems this is wrong and just need to be inserted in the order as found.

Try latest code.

aminetec, 2022-01-16:
I tried the latest merge the problem is fixed thank you

closed 2024-03-10 by KeithSloan