KeithSloan/GDML #73: cant make a new material

lichuanlong, 2022-01-07:
we have G4_BORON_CARBIDE  and G4_POLYETHYLENE in freecad gdml.
how to make a new material with 0.05 * G4_BORON_CARBIDE + 0.95 *G4_POLYETHYLENE ？

KeithSloan, 2022-01-07:
Currently there is no facility to define new materials via the GUI.

You can create an XML file and import it 

The XML should look like
````
<materials>
      <material formula="New_Material" name="New_Material" >
            <D unit="g/cm3" value="1.127" />
             <fraction n="0.05" ref="G4_BORON_CARBIDE"/>
              <fraction n="0.95" ref="G4_POLYETHYLENE"/>
       </material>
</materials>
````

For New_Material choose an appropriate name.

You need to give a value for the density, not sure how one goes about working this out.
I am guessing 0.05 * density of G4_BORON_CARBIDE + 0.95 * density of G4_POLYETHYLENE

Save as NewMaterial.xml then import

It should then appear as a NewMaterial in the main Materials Group and the definition exported for any export GDML
file

Geant4. Materials are defined in Geant4Materials.xml which you should find in the Resources directory. i.e. ../..Mod/GDML/freecad/gdml/Resources. These get loaded into the Group Geant4 within Group Materials.
They are NOT exported as Geant4 should already know about them.

If it is a material that Geant4 knows about and missing from Geant4Materials.xml,  if you let me know the name and density I will add it to Geant4Materials.xml

KeithSloan, 2022-01-07:
If the material is already defined in another existing GDML file, then you should be able to 

- load this file with the material defined.
- Select Materials 
- Then Export, use xml as the file extension
- Create a new file with GDML as the current workbench
- Import the xml file should add the Material definition to the Materials Group

MITgnome, 2023-08-26:
Does the import of materials still work in the current release. I tried the instructions above and the new materials did not appear under the model. The log just showed:

08:27:58  Insert filename : /Users/rickfoster/Desktop/dataBaseMaterialMIT.xml docname : Unnamed
08:27:58  process XML : /Users/rickfoster/Desktop/dataBaseMaterialMIT.xml
08:27:58  Parse : /Users/rickfoster/Desktop/dataBaseMaterialMIT.xml
08:27:58  running with lxml.etree 
08:27:58  Process Materials DocSet
08:27:58  define xml None
08:27:58  Process - Opticals: matrix_spreadsheet
08:27:58  define xml None

The following is the XML file I imported:

[dataBaseMaterialMIT.txt](https://github.com/KeithSloan/GDML/files/12445335/dataBaseMaterialMIT.txt)

KeithSloan, 2023-08-26:
Apologies if the instructions are not clear.

The file needs to have an extension of xml and must include 
```
<xml>
```
 at the start of the file and
```
</xml>
```
 at the end.

[dataBaseMaterialMIT.xml.txt](https://github.com/KeithSloan/GDML/files/12445594/dataBaseMaterialMIT.xml.txt)

MITgnome, 2023-08-26:
Yes, silly mistake on my part. I cleaned up the file a bit and used the G4 elements already defined by you. Things seemed to work. One thing that would be useful is if I could add the materials to a new folder as opposed to having them all at the top level of the Materials folder, is that possible?
[image]

KeithSloan, 2023-08-26:
"One thing that would be useful is if I could add the materials to a new folder as opposed to having them all at the top level of the Materials folder, is that possible?"

So you would like the new Elements added to a new group under Materials based on the say the imported file name e.g dataBaseMaterialMIT.

That is certainly doable and a good idea if importing several sets.
Will also require updates to various material selection facilities/code.

I fly back from Seattle in the US to the UK on Monday and will look into when I get back.

Thanks for the suggestion.

MITgnome, 2023-08-30:
It seems if I import the new materials using a xml file, those new materials do not seem to be available to GDML solids that have **_already been placed_** in the workspace. They do seem to be available for GDML solids that I place **_after_** I imported the XML Materilas file. Is this what you would expect?

KeithSloan, 2023-08-30:
Yes. When I a GDML solid is created the Material Property is made up of an enumeration list of the known materials at the time. You could try a recompute of the object but I suspect that would not help without a change of the current coding.
Worth trying using the Set Material Icon / Command i.e. select object and then the icon but as the object is already created with the enumeration list of known materials at the time, this may just give an error when trying to set a new value.

I would have to look into the possibility of changing a enumerated property of an object without recreating it, sorry about   that.

KeithSloan, 2023-08-30:
Okay the doc says
"App::PropertyEnumeration
An enumeration property. The allowed items are defined by setting the property to a list. After that, it can contain items of the given list. The list of allowed items can be changed by setting the property to a list again."

I need to look into a code change. I assume this is more urgent than keep all the MIT materials together as per the Geant4 ones.

KeithSloan, 2023-08-30:
Okay I just tested.

I start GDML workbench
create a GDML object
Then I import a material XML file like your MIT one.
Select the object who's material I wish to change.
Click on Set Material Icon
You should fine the new imported Materials under the "Normal" category.

I am open yo a better term than Normal for the Non Geant4 groups

MITgnome, 2023-08-31:
Speaking as an end-user of Geant4, as opposed to a developer, I can say that Freecad with GDML has been progressing in the right direction over the last couple of years. Construction and maintenance  of the geometry models has always been difficult for us. Most of the solutions for geometry generation/editing seem to have had a rather start/stop flavor to them. There seems to be effort by the Fastrad folks to provide an integrated solution, but it is not an open source solution that more casual users, such as myself, are looking for.

One way you might get some increased visibility for FreeCad/GDML is to show it off at the upcoming Space Users Workshop in December. The Geant4 collaboration folks that conduct the workshop have been around for a long time and I suspect have the type of influence you are looking for.

https://indico.esa.int/event/477/

You are correct that being able to import materials lists into a geometry model that has already been created is more important than sorting those materials into folders. You can imagine wanting to be able to add or modify materials, import them into the model and update the parts with either a new material or a parametric change to an existing material.

I will try the set material icon approach you described and let you know how it worked.

Thanks
Rick

MITgnome, 2023-08-31:
Quick Question using the Set Material Icon.
Is the Material set at the **_Part Object_** (such as Si3N4_coating) or say the **_GDML Object_** (such as GDMLBox_Box002)
[image]

KeithSloan, 2023-08-31:
The GDML material property is in the GDML Objects, where you see a Material Property ( i.e. A Capital M) on a Part is a FreeCAD material and these are ignore as far as use of the workbench. I might be wrong but it was introduced after the GDML workbench was started. They are used for FEM and Arch workbenches.

This I agree can be confusing. Unfortunately the FreeCAD materials facility is not flexible enough for us to use.

MITgnome, 2023-08-31:
OK, so then that didn't seem to work. I selected the GDMLBox_Box002 object. Used the Set Material icon and selected NORMAL:MIT_SILICON_NITRIDE, expecting it to update the previous placeholder material G4_SILICON_DIOXIDE. The console had the message Set Material MIT_SILICON_NITRIDE, but it does not seem to actually update the material, but leaves it as G4_SILICON_DIOXIDE.
[image]

KeithSloan, 2023-08-31:
Might be a bug in terms of the property window not being updated, could you try selecting a different object and then going back to GDMLBox_Box002

MITgnome, 2023-08-31:
That did not work either, it stayed as G4_SILICON_DIOXIDE. I also ran recompute on the whole project.

MITgnome, 2023-08-31:
I just looked at the exported gdml file and it still shows G4_SILICON_DIOXIDE
[image]

KeithSloan, 2023-08-31:
Seems to be working for me.

[code]
OS: macOS 12.4
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods:

ImportNURBS 1.1.0
ImportExport_3DM 0.1.0
DynamicData 2.46.0
Pyramids-and-Polyhedrons
occi-freecad-plugin 0.1.0
GDML 2.0.0
Design456 0.0.1
Alternate_OpenSCAD 1.0.0
toSketch 1.0.1
Curves 0.6.13
[/code]

Can you provide your about FreeCAD info.

MITgnome, 2023-08-31:
[code]
OS: macOS 12.6.8
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * GDML 2.0.0
[/code]

KeithSloan, 2023-08-31:
"I just looked at the exported gdml file and it still shows G4_SILICON_DIOXIDE
image"

If the Material does not show a change in FreeCAD it will not change on export.

Did you click on the "Set Material Box" after selecting SILICON_NITRIDE?

MITgnome, 2023-08-31:
Yes, I hit the Set Material button and seemed to get a positive confirmation in the Report View
[image]

MITgnome, 2023-08-31:
I went thru the process again, selecting the object, clicking Set Material Icon, etc and here is the full log in the report window. Not sure what the QCompleter message is all about.
[image]

KeithSloan, 2023-08-31:
Okay, the only difference I can see is I am OS: macOS 12.4 and you are on OS: macOS 12.6.8.

I will upgrade and see if I can recreate the problem.

The message is coming from Qt and we seem to have the same version of that.

MITgnome, 2023-08-31:
I just ran it on MacOS Ventura 13.5 and had the same result.

KeithSloan, 2023-08-31:
I upgraded my Mac and am now seeing the "Setting QCompleter on non-editable QCombox is not allowed"  message
and seeing the same problem. I think the update has exposed a bug in that an option should be set to allow a QCombox to be changed.

MITgnome, 2023-08-31:
Bummer when upgrades break things that worked before.

MITgnome, 2023-08-31:
My MacOS 12.6.8 is already my **_old_** workstation. I have a mountain of older machines piled in the lab, but that approach is too scary to contemplate.

MITgnome, 2023-08-31:
I guess for now, I will just do geometry modifications in Freecad and if I have to tweak the materials I can hand edit the exported gdml file.

KeithSloan, 2023-08-31:
Okay have fixed the "Setting QCompleter on non-editable QComBox is not allowed" message, just had to change the order on some QCombobox options

To fix the none update with added material definitions just had to set the combobox to the latest list before setting the value.

Thanks for your patience and testing, please try the latest version.

MITgnome, 2023-09-01:
That seems to have done the trick. I tested changing a few materials and then exported out to a gdml file successfully.
Thank you!

KeithSloan, 2023-09-02:
Well, thank you for - reporting the problem and assisting with problem determination.