KeithSloan/GDML #74: Running the Python command 'TessellateCommand' failed

gipert, 2022-01-12:
Hi, the standard tessellate action seems to be broken for me:
```
14:45:25  Running the Python command 'TessellateCommand' failed:
Traceback (most recent call last):
  File "/home/gipert/.FreeCAD/Mod/GDML/freecad/gdml/GDMLCommands.py", line 1139, in Activated
    mesh = MeshPart.meshFromShape(Shape=shape,Fineness=2,\

Wrong arguments
```
Any idea? I installed freeCAD on Arch Linux via package manager.

KeithSloan, 2022-01-12:
Strange is working here.
Line 1139 and the continution should look like
````
        mesh = MeshPart.meshFromShape(Shape=shape,Fineness=2,\
                      SecondOrder=0,Optimize=1,AllowQuad=0)
````
Its invoking the standard FreeCAD mesh facility and I don't see why Arch Linux should be any different to other OS's

Do you have sample file with the Object you are trying to Mesh.
Do you get the same error if you try meshing a Cube

KeithSloan, 2022-01-12:
If you click on FreeCAD in the title bar | About FreeCAD | Copy to Clipboard | And Paste what versions are reported?

gipert, 2022-01-13:
Yes, it fails even with a simple cube. Can it be that the function interface has been changed?

[image]

KeithSloan, 2022-01-13:
24291 is FreeCAD 0.19.2 and I have no problem with the same version on MacOS.

There is an Arch version of FreeCAD 0.19.3 https://archlinux.org/packages/community/x86_64/freecad/

[image]

If click on 'Copy to clipboard' and then paste the result it will tell us the level of software that FreeCAD is using.

What happens if you try and create a mesh? The workbench includes the Mesh Workbench facility, Hovering over the icon shows 'Tessellate Shape'. As a work around you should then be able to create a GDML_Tessellated object with 'Mesh 2 Tess' icon.

Have asked in FreeCAD forum if anybody has any ideas [https://forum.freecadweb.org/viewtopic.php?f=4&t=65281](url)

I think there is also version of FreeCAD 0.20 for arch [https://aur.archlinux.org/packages/freecad-git/](url)
might be worth a try.

gipert, 2022-01-13:
That 0.19.3 is a lie I think: https://github.com/archlinux/svntogit-community/blob/packages/freecad/trunk/PKGBUILD#L30. I just updated my system but the freecad version stayed the same:
```
OS: Arch Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: makepkg
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.10.1
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```
> What happens if you try and create a mesh? The workbench includes the Mesh Workbench facility, Hovering over the icon shows 'Tessellate Shape'. As a work around you should then be able to create a GDML_Tessellated object with 'Mesh 2 Tess' icon

Indeed, this works!

I will try the unstable version, I need to compile it...

adrianinsaval, 2022-01-13:
> That 0.19.3 is a lie I think: https://github.com/archlinux/svntogit-community/blob/packages/freecad/trunk/PKGBUILD#L30. I just updated my system but the freecad version stayed the same:

This should probably reported as a bug to the arch maintainers

adrianinsaval, 2022-01-13:
Also take into account that running FreeCAD with python 3.10 is kinda broken right now: https://forum.freecadweb.org/viewtopic.php?f=8&t=65263&p=561194 so you should consider using the appimage or the flatpak version. Not sure if this is the cause here.

luzpaz, 2022-01-13:
> This should probably reported as a bug to the arch maintainers
Created https://bugs.archlinux.org/task/73349

KeithSloan, 2022-01-13:
> > This should probably reported as a bug to the arch maintainers
> 
> Created https://bugs.archlinux.org/task/73349

Not sure it is Arch bug. Downloaded FreeCAD 0.19.3 for Mac OS and it also states
````
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git)
Build type: Release
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: C/Default (C)
````
Think it might be a case of whoever did the appimage build forgot to change the version.

adrianinsaval, 2022-01-13:
No, look at the commit count, your's says 24366 and gipert's says 24291, the later was 0.19.2's commit count and following the link in the PKGBUILD it is the commit tagged for 0.19.2, not 0.19.3

KeithSloan, 2022-01-23:
Comment in FreeCAD forum [https://forum.freecadweb.org/viewtopic.php?p=562253#p562253](url)

FreeCAD uses pyside2 and this is not compatible with Python 3.10

gipert, 2022-01-23:
Thanks @KeithSloan. I'll close this since it's a FreeCAD issue.

wwmayer, 2022-01-25:
The function meshFromShape is defined here: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/MeshPart/App/AppMeshPartPy.cpp#L476

Now you use the keywords Fineness, SecondOrder, Optimize and AllowQuad but they are only supported if smesh is built with Netgen support.

KeithSloan, 2022-01-26:
Okay I changed the code to
````
               try :        # Only supported if smesh built with netgen
                   mesh = MeshPart.meshFromShape(Shape=shape,Fineness=2,\
                          SecondOrder=0,Optimize=1,AllowQuad=0)
               except :
                   mesh = MeshPart.meshFromShape(Shape=shape)
````
But don't have an easy way to test maybe @gipert could test and report.

gipert, 2022-01-26:
Hi, it fails with this now:
```
11:00:54  Running the Python command 'TessellateCommand' failed:
Traceback (most recent call last):
  File "/home/gipert/.FreeCAD/Mod/GDML/freecad/gdml/GDMLCommands.py", line 1143, in Activated
    mesh = MeshPart.meshFromShape(Shape=shape)

Wrong arguments
```

KeithSloan, 2022-01-26:
Well looking at the code in the link that Werner @wmayer provided, you could try changing line 1143 in GDMLCommands.py to
````
mesh = MeshPart.meshFromShape(Shape=shape,MaxLength=<values>)

or 

mesh=MeshPart.meshFromShape(Shape=shape,MaxArea=<values>)

or

mesh=MeshPart.meshFromShape(Shape=shape,LocalLength=<values>)

or 

mesh=MeshPart.meshFromShape(Shape=shape,Deflection=<values>)

or 

mesh=MeshPart.meshFromShape(Shape=shape,MinLength=<values>,MaxLength=<value>)

````
or I am not reading the code correctly

KeithSloan, 2022-01-26:
I would favour changing to 
````
mesh = MeshPart.meshFromShape(Shape=shape,MaxArea=shape.Area)
````

gipert, 2022-01-26:
It works by patching as you suggested. You can commit the fix!

KeithSloan, 2022-01-26:
Thanks for reporting and testing.

closed 2022-01-26 by KeithSloan

wwmayer, 2022-01-26:
With https://github.com/FreeCAD/FreeCAD/commit/2cb478fd66dc76d91695dacff15fd764e80c8c62 a RuntimeError is raised now if you use `MeshPart.meshFromShape(Shape=shape,Fineness=2, ...)` but smesh was not built with netgen support.

So, now you can write:
```
               try :        # Only supported if smesh built with netgen
                   mesh = MeshPart.meshFromShape(Shape=shape,Fineness=2,\
                          SecondOrder=0,Optimize=1,AllowQuad=0)
               except RuntimeError:
                   mesh = MeshPart.meshFromShape(Shape=shape)
```

wwmayer, 2022-01-26:
There is still one aspect you have to look at:
The C define HAVE_NETGEN is set if the internal smesh is used and the build option BUILD_FEM_NETGEN is enabled.
However, when an external smesh version is used I don't know if HAVE_NETGEN is still set. So, what could happen is that an external smesh was built with netgen support but the build process of FreeCAD won't realize it and still claims that smesh wasn't built with netgen support.