KeithSloan/GDML #77: Moving README wiki documentation to the wiki itself

luzpaz, 2022-01-15:
Hey @KeithSloan 
So in https://github.com/KeithSloan/GDML#usage there is a section you intend to move to the wiki. Where in the wiki does it belong. I'll be happy to migrate it for you.

KeithSloan, 2022-01-15:
Create a new GDML Design
Should be moved and merged with Wiki Model Creation

Wiki - importing STEP and creating GDML Tessellated Objects
needs spliting into importing STEP ( Should cover tessellation or refer to)
GDML Tessellated Objects should be separate

GDML import -> wiki import GDML file
scan info -> wiki Scan GDML File

Viewing Volumes -> wiki Viewing A GDML

 wiki Materials section
 needs new info adding
 move Export/Import of Materials as an XML file.

wiki section on Export
   * GDML Objects Exporter use 
   * export of GDML Objects
   * export to STEP
   * export FreeCAD objects

wiki section for GEMC
   GEMC info to this section
   
 wiki section Compound & FEM - Finite Element Analysis
 Compound & FEM - Finite Element Analysis -> wikithis section
 
 wiki Standalone Utils
  Stand alone Utils -> wiki this section 
  
  Will do for starters - review when done

luzpaz, 2022-01-15:
I added markdown checkboxes to your above response to be able to record progress