KeithSloan/GDML #81: Strange behavior of the tessellated solids.

porosev, 2022-03-02:
Hello all,

I try to import the STEP file and convert it to GDML format to use it with Geant4.

But I see the strange behavior of FreeCAD 0.19.3 when I try to create tessellated solids. After conversion all parts are shifted on a small distance vertically and additionally, some parts are rotated 90 grads during the generation of "GDML Tessellate selected object". Additionally, when I try to use "GMSH Tesselate" it reports  
"Could not find module 'D:\Python\Python310\Scripts\gmsh' (or one of its dependencies). Try using the full path with constructor syntax.".... How to define the path to "gmsh" or check its existence? Why does it try to find the GLOBAL python module instead of FreeCAD one? Is it possible to use *.MSH files generated by the standalone "GMSH" application during generation of tesselate solids?

It checked with the latest FreeCAD build 0.20 and the result is the same...

porosev, 2022-03-02:
[](url
[TC_Fasteners.zip](https://github.com/KeithSloan/GDML/files/8169237/TC_Fasteners.zip)
)

I just add some simple example that clearly demonstrates the problem.

KeithSloan, 2022-03-02:
So, how to install "'gmsh'" for the GDML addon in FreeCAD???

As per the README

gmsh python library

The workbench uses the gmsh python library and must be installed in a location that FreeCAD sees

To check path FreeCAD uses from a command line/window.

freecad -c
import sys
print(sys.path)
In a command window / line

pip install --upgrade --target <Full path to directory> gmsh

Plus you need to install Gmsh shared library

On my mac I am actually using the latest snap shot of Gmsh

See https://gmsh.info

So for the Gmsh python library 
pip install -i https://gmsh.info/python-packages --force-reinstall --no-cache-dir gmsh-dev' -t<FreeCAD known path>

and in /Applications/FreeCAD_19.2.app/Contents/Resources/lib/ I have copied
/Applications/FreeCAD_19.2.app/Contents/Resources/lib/gmsh.jl
/Applications/FreeCAD_19.2.app/Contents/Resources/lib/gmsh.py
/Applications/FreeCAD_19.2.app/Contents/Resources/lib/libgmsh.4.9.4.dylib
/Applications/FreeCAD_19.2.app/Contents/Resources/lib/libgmsh.4.9.dylib

KeithSloan, 2022-03-02:
But I see the strange behavior of FreeCAD 0.19.3 when I try to create tessellated solids. After conversion all parts are shifted on a small distance vertically and additionally, some parts are rotated 90 grads during the generation of "GDML Tessellate selected object".

I am seeing the same and looking into it.

KeithSloan, 2022-03-02:
Why does it try to find the GLOBAL python module instead of FreeCAD one? 
FreeCAD has two standard interfaces to Gmsh - one supplied by the Mesh workbench, but FC only supports Triangular Mesh's where as GDML supports Quad meshes. Note the FC interface can produce Quad facets but then converts to Triangle. The other Gmsh interface is via the FEM workbench and again has limitations.

Is it possible to use *.MSH files generated by the standalone "GMSH" application during generation of tesselate solids?
Not currently but would be a good enhancement

KeithSloan, 2022-03-02:
As a work around until I can fix the shifts and rotations, that you mesh the object with
[image]

And then convert to GDML Tessellated with
[image]

KeithSloan, 2022-03-02:
Try latest version.

Not totally happy about the fix, when meshing using default FreeCAD meshing then Mesh points are created as if Shape has identity Placement have adjusted for this but not sure if Geant4 will be happy with this. 

Please try a few exports and load_gdml's with Geant4 and report how it goes.

porosev, 2022-03-04:
Dear Keith,
 
Many thanks for you advice! 
 
Last night, I checked both methods: the suggested way "create mesh -> Mesh2Tesselated" and your updated version of "GDML Tesselate selected object".
 
Now they both produce GEANT4 readable files!!!
 
There is only one difference.  When I use GDML created with "Mesh2Tesselated" I see warnings that there are few overlaps in my geometry between some screw and nuts. I think its OK, because I can increase/change mesh accuracy in a configuration dialog box at the start of operation. So, I hope that I can eliminate these warnings.
 
For "GDML Tesselate selected object" I have the same problems, but I have not any possibility to define desired accuracy ;(((
Additionally, while reading a GDML file created with "GDML Tesselate selected object" , GEANT4 reports the warning "there are holes in the surface" for some solids.
 
I think the suggested way "Mesh 2 Tesselated" is the best way in my case. Additionally it produces more smooth/good-looking surfaces. 
  
So, You did incredibly good FreeCAD extention!  
If you add possibility to add new material with some density as composition of atoms or weight fractions in GUI it will be great!
 
Anyway, many thanks!

porosev, 2022-03-04:
Concerning "gmsh"... It looks that FreeCAD have some installation/configuration problems under Windows. It looks like FreeCAD Python doesn't know own location and it try to use any modules witch it can find on computer intead of FreeCAD ones ;((.

KeithSloan, 2022-03-04:
"If you add possibility to add new material with some density as composition of atoms or weight fractions in GUI it will be great!"

Have created a new issue #83. Also please see #73 for non GUI method.

closed 2024-05-21 by KeithSloan