KeithSloan/GDML #95: Confusing definition of materials and ? about sensitive detector.

porosev, 2022-07-01:
Dear Keith...

I have a stupid question. 

When I define a part, I have property MATERIAL as for part as for its solid.
[image]
Is it possible to hide the first property field, because the last one only is saved in the GDML file only or propagate it down over tree?

Could you recommend any easy way to define sensitive detectors (names) in FreeCAD?

KeithSloan, 2022-07-01:
I think a Part having a material property is probably a bug,  I will investigate, thanks for reporting.

KeithSloan, 2022-07-05:
The Material you are seeing on a Part is a FreeCAD material allocation as might be used in FEM, Arch workbench, I don't think I can disable it. It is not being created by the GDML workbench as far as I can tell.

Unfortunately there is not a good mapping between what are GDML Materials and FreeCAD Materials [https://wiki.freecadweb.org/Material](url), so GDML Materials are treated totally separately.

I will look at defining sensitive detectors.

KeithSloan, 2022-07-05:
Okay I have added code to hide App::Part Material properties where an App::Part is created by the GDML workbench.

Not sure I have got them all and if a user created a Part using the Icon in the GDML workbench or other workbench there is not a lot I can do.

Change pushed to Main branch.

KeithSloan, 2022-07-05:
Have added code for setting SensDet to a new branch **SensDet**

If you could check it out

The Addon Manager in the newly released FreeCAD 0.20 allows you to switch branches, so it would again be good if
you could test that.

Looking for suggestions on icon design - currently an **X**

porosev, 2022-07-07:
Dear Keith... 

Many thanks for your advices. Unfortunately the text field with GitHub path is not editable  and I could not change branch name here ;(( 

Anyway, today I installed the new GDML Workbench 2.0.0 without any problem.
Below I have compiled a sequence of magic spells to get a working version FreeCAD/GDML/GMSH for Windows:

1. Install "only for me" to "D:\FreeCAD 0.20" to prevent any file/folder access problems.

2. Upgrade/Install additional python modules:

D:\FreeCAD 0.20\bin\python -m pip install --upgrade pip
D:\FreeCAD 0.20\bin\python -m pip install --target="D:\FreeCAD 0.20\bin\Lib\site-packages" --upgrade lxml
D:\FreeCAD 0.20\bin\python -m pip install --target="D:\FreeCAD 0.20\bin\Lib\site-packages" --upgrade gmsh
D:\FreeCAD 0.20\bin\python -m pip install -i https://gmsh.info/python-packages --force-reinstall --no-cache-dir --upgrade --target="D:\FreeCAD 0.20\bin\Lib\site-packages" gmsh-dev

3. Download the Software Development Kit (SDK) for Windows of GMSH from https://gmsh.info/bin/Windows/gmsh-4.10.5-Windows64-sdk.zip , unpack it and copy:

..\bin to "D:\FreeCAD 0.20\bin\Lib\site-packages\bin"

..\include to "D:\FreeCAD 0.20\bin\Lib\site-packages\include"

..\lib to "D:\FreeCAD 0.20\bin\Lib\site-packages"

to add the missed "gmsh.exe", "gmsh-4.10.dll" and update "gmsh.py".

4. install GDML workbench.

porosev, 2022-07-07:
Dear Keith...

I have solved my problems with GitHub branches of GDML workbench.  As I understood, the FreeCAD for Windows has some problems with GitPython module. In spite of the fact that it was installed, it not operated correctly. Therefore I have updated it with the command:

D:\FreeCAD 0.20\bin\python -m pip install --target="D:\FreeCAD 0.20\bin\Lib\site-packages" --force-reinstall --no-cache-dir --upgrade GitPython

After that I switched on the flag SHOW BRANCHES: 

[image]

and after that I can switch GDML branhes.

[image]

I hope I will be useful for somebody else...

porosev, 2022-07-08:
Dear Keith...

I am sorry, but with the new branch Main I could not define my  custom materials for volumes. Something goes wrong. With previous branch Master it still works. 

[image]

It is funny but the material group name "Normal" was added by FreeCAD, not me.
My "MyMaterials.xml" file:

<xml>
  <materials>
      <material formula="Quartz" name="Quartz" >
        <D unit="g/cm3" value="2.64" />
        <composite n="1" ref="G4_Si"/>
        <composite n="2" ref="G4_O"/>
      </material>
      <material formula="Epoxy" name="Epoxy" >
        <D unit="g/cm3" value="1.3" />
        <composite n="15" ref="G4_C"/>
        <composite n="44" ref="G4_H"/>
        <composite n="7"  ref="G4_O"/>
      </material>
      <material formula="FR4" name="FR4" >
        <D unit="g/cm3" value="1.025" />
        <fraction n="0.613" ref="Quartz"/>
        <fraction n="0.147" ref="Epoxy"/>
        <fraction n="0.080" ref="G4_Br"/>
        <fraction n="0.160" ref="G4_O"/>
      </material>
      <material formula="GAGG" name="GAGG" >
        <D unit="g/cm3" value="6.63" />
        <composite n="3"  ref="G4_Gd"/>
        <composite n="2"  ref="G4_Al"/>
        <composite n="3"  ref="G4_Ga"/>
        <composite n="12" ref="G4_O"/>
      </material>
  </materials>
</xml>

Or I did something wrong?

porosev, 2022-07-08:
[image]

KeithSloan, 2022-07-08:
Please could you post a copy of the xml file, if necessary add a file extension of txt and copy/drag to a post in this thread - Thanks

porosev, 2022-07-09:
Dear Keith...

This is my material fiel.... May be I do something wrong.

[MyMaterials.txt](https://github.com/KeithSloan/GDML/files/9075851/MyMaterials.txt)

KeithSloan, 2022-07-09:
Not sure what is going on. 

If I import your material file, for any GDML Object created after this I can change the Material property via the Properties window but the new materials are displayed after a very long list of Geant4 Materials and Elements :-(

If I try the set Materials icon, then I only get offered Geant4 Materials. :-(

Seems like branch **Main** has lost some updates at least, what branch are you using?

porosev, 2022-07-09:
Dear Keith,

In the branch - Main, I could not use my materials. In the branch - master it still works. Anyway, in the properties list box - material - I don't see my materials. I just use the toolbox button - Set Material.

KeithSloan, 2022-07-09:
Try the latest version of Main.

I want to do some work on this when I get back in just over a fortnight but hopefully it is an improvement ASIS.

The group **Normal** is all the Non Geant4 defined Materials, if you can suggest a better name, that would be good.

porosev, 2022-07-10:
Dear Keith,

I checked the Latest version of the Main branch - the result is the same ;((. For Comparison I  attached print-screens with Main and the master branch which still works.

[image]
[image]

Additionally in both cases the material list-box (highlighted with arrow) not includes new added materials.

Just few ideas about the material group name:  "User", "User defined", "Custom" etc.

KeithSloan, 2022-07-10:
Is the FreeCAD file one you had previously saved? i.e The GDMLBox you are trying to change the material of was created by a previous version of the Workbench. Not sure there is much I can do if it is, the enumeration is set by the code at the time of creation.

Not convinced you are using the latest version, when I try my Report View looks like
[image]

Your screen shot shows the number 320 after Normal, that is old code.

porosev, 2022-07-11:
Dear Keith...

You are right.  I created new world, added new materials, saved project, closed, and re-opened it. After that all works fine! 
If I try to open my old project I could not set material.

It's funny, but I have found an additional confusing feature... After start of FreeCAD and activation of the GDML workbench,
if I select "File->New"(arrow 1) a get empty project, if I use "Create new" (arrow 2) icon I get normal GDML project with Geant4 materials, default world volume etc...

 
[image]

KeithSloan, 2022-07-11:
Do you still have the old file? If yes please attach to this thread, I will look at adding code to recreate the material if there is a failure but ideally needs a test case.

porosev, 2022-07-11:
[Test.zip](https://github.com/KeithSloan/GDML/files/9081820/Test.zip)

KeithSloan, 2022-07-11:
Okay try latest version of Main should now handle invalid enumeration from old versions of Workbench.

Now going away, will look at your other issues on my return.

Thanks again for taking the time and trouble to report issues.