KeithSloan/GDML #97: The new problem with GMSH

porosev, 2022-07-21:
Dear Keith...

I am evaluating the limits of your GDML tool with GMSH. 

So, the initial conditions... I manually configurated path to GDML in FEM addon and restored check-box state.
[image]

It can correctly find GMSH when I press menu button:

[image]

and it shows GMSH info

[image]

After that I switched to GDML workbench and try "create mesh from shape"...
With "Standard, Mefisto, Netgen" it works fine but with GMSH it reports the error.

[image]

Additionally, when I try to use the command "Gmsh Tesselate" and press the  "Mesh" it reports error in Python code

[image]

In my private opinion, the mesh produced by stand-along GMSH application looks better than generated with Mefisto, Netgen etc,  therefore I try to compare it with other mesh generators in GDML workbench during a Geant4 geometry preparation.

KeithSloan, 2022-07-21:
There are a number of interfaces to Gmsh

1. FEM workbench
2. Mesh Workbench
3. GDML Workbench

The GDML workbench uses the Gmsh Python Library which is a different interface than 1) and 2) and these are outside of my control.

The last screen shot is clearly a bug, have tried a quick fix but need to look at this more carefully and I am now away till the 25th, sorry about this.

KeithSloan, 2022-07-27:
I have made some changes for Gmsh and pushed to Main.

I am having a problem with my Geant4 installation.

Please could you try the attached gdml file, it is a test of a simple Gmsh operation
[Gmeshed_Box.FCStd.txt](https://github.com/KeithSloan/GDML/files/9203822/Gmeshed_Box.FCStd.txt)
[Gmeshed_Box-worldVOL.gdml.txt](https://github.com/KeithSloan/GDML/files/9203823/Gmeshed_Box-worldVOL.gdml.txt)

KeithSloan, 2022-07-27:
By the way my limited understanding of MonteCarlo simulation and Meshed Objects is that you want the minimum mesh that accurately describes a shape, whereas in things like FEM analysis you want for accuracy as fine a mesh as possible ( subject to run times).

Now I got Christophe Geuzaine one of the professors that developed Gmsh to add a function in Gmsh to create a minimum mesh from a CAD/STEP file. I have yet to add it to the GDML workbench but I think the function may now be in the released version of Gmsh rather than just a development snapshot.

see [https://gitlab.onelab.info/gmsh/gmsh/-/issues/1726#note_16184](url)

If we can get the GDML workbench working well with Gmsh then I will try adding this as an option.

porosev, 2022-07-28:
Dear Keith,

I am out of office until next Monday. So, I will try it ASAP.

Sincerely yours,
Vyacheslav Porosev

KeithSloan, 2022-07-28:
Still a problem. When the workbench Tessellates an Object it does not delete the original object and just hides it, see screen shot

Options seem to be

1) Ignore any extra GDML Objects in a Part where one is a GDMLTessellated or GDMLGmshTessellated
2) Ignore hidden objects on export
3) Delete object being Tessellated ( Don't like this option, as if a user does not like the look with the current Tessellated parameters, there is no original to try again )

What do you recon?

[image]

porosev, 2022-07-29:
Dear Keith

As for me, ignore hidden objects during the export could be optimal way. It
means an user can keep different/full project configuration in FreeCAd and
export only required parts for tests/evaluation.

Sincerely yours
Vyacheslav Porosev

porosev, 2022-07-31:
Dear Keith,

I use ..\extended\persistency\gdml\G04 example for GDML evaluation.
During the geometry reading it report the following issues:

G4GDML: Reading 'Gmeshed_Box-worldVOL.gdml'...
G4GDML: VALIDATION ERROR! ID value 'GDMLBox' is not unique at line: 286
G4GDML: VALIDATION ERROR! ID attribute 'G4_AIR' is referenced but was never declared at line: 865
G4GDML: VALIDATION ERROR! ID attribute 'G4_A-150_TISSUE' is referenced but was never declared at line: 865
G4GDML: Reading definitions...
G4GDML: Reading materials...
G4GDML: Reading solids...
G4GDML: Reading structure...
G4GDML: Reading setup...
G4GDML: Reading 'Gmeshed_Box-worldVOL.gdml' done!
Stripping off GDML names of materials, solids and volumes ...

**************************************************************
 Geant4 version Name: geant4-11-00-patch-02 [MT]   (25-May-2022)
  << in Multi-threaded mode >> 
                       Copyright : Geant4 Collaboration
                      References : NIM A 506 (2003), 250-303
                                 : IEEE-TNS 53 (2006), 270-278
                                 : NIM A 835 (2016), 186-225
                             WWW : http://geant4.org/
**************************************************************

<<< Geant4 Physics List simulation engine: FTFP_BERT

Found 2 volume(s) with auxiliary information.

Volume V-GDMLBox has the following list of auxiliary information: 

--> Type: Color Value: #80808000
Volume GDMLTessellate_GDMLBox has the following list of auxiliary information: 

--> Type: Color Value: #80808000

Volume V-GDMLBox has the following list of auxiliary information: 

Volume GDMLTessellate_GDMLBox has the following list of auxiliary information: 

 hInelastic FTFP_BERT : threshold between BERT and FTFP is over the interval 
 for pions :   3 to 6 GeV
 for kaons :   3 to 6 GeV
 for proton :  3 to 6 GeV
 for neutron : 3 to 6 GeV

### Adding tracking cuts for neutron  TimeCut(ns)= 10000  KinEnergyCut(MeV)= 0
=======================================================================
=====                 Electromagnetic Physics Parameters      ========
....

and so on...  

After that I generally use VRML output to see final geometry and the world box looks empty.

[image]

porosev, 2022-07-31:
I am sorry, I was wrong..  It looks that GDML reader set to World box visualization attribute - INVISIBLE as default. 
I do it in my code sometimes.
 // logicWorld->SetVisAttributes(G4VisAttributes::GetInvisible());)

Therefore, it is a real GDMLBox.

I removed your tessellated solid, regenerated the mesh with "Gmsh Tesselate". After that I removed the original box and made export of the World. 
[image]
[image]
During VRML visualization I switched on option "Show only frame"
[image]
So, It look's correct with one exception for confusing material definition order..

G4GDML: Reading 'Gmeshed_Box-worldVOL2.gdml'...
G4GDML: VALIDATION ERROR! ID attribute 'G4_AIR' is referenced but was never declared at line: 685
G4GDML: VALIDATION ERROR! ID attribute 'G4_A-150_TISSUE' is referenced but was never declared at line: 685
G4GDML: Reading definitions...
G4GDML: Reading materials...
G4GDML: Reading solids...
G4GDML: Reading structure...
G4GDML: Reading setup...
G4GDML: Reading 'Gmeshed_Box-worldVOL2.gdml' done!
Stripping off GDML names of materials, solids and volumes ...

**************************************************************
 Geant4 version Name: geant4-11-00-patch-02 [MT]   (25-May-2022)
  << in Multi-threaded mode >> 
                       Copyright : Geant4 Collaboration
                      References : NIM A 506 (2003), 250-303
                                 : IEEE-TNS 53 (2006), 270-278
                                 : NIM A 835 (2016), 186-225
                             WWW : http://geant4.org/
**************************************************************

<<< Geant4 Physics List simulation engine: FTFP_BERT

Found 1 volume(s) with auxiliary information.

Volume LV_Box has the following list of auxiliary information: 

--> Type: Color Value: #80808000

Volume LV_Box has the following list of auxiliary information: 

 hInelastic FTFP_BERT : threshold between BERT and FTFP is over the interval 
...
There is one more feature ;)) When I press the list-box selector to select mesh characteristics the main dialog window disappear.
I can change a selection with mouse wheel only when the mouse cursor over it.
[image]

KeithSloan, 2022-08-01:
Been working on another problem see https://github.com/KeithSloan/GDML/discussions/93
which we have now made good progress on.
I am not going to be able to progress this before going away, should be back 6th.

With the current code it is best to delete the object being Tessellated before exporting, have a plan to fix this.
Materials is down to order and needs fixing, don't think it is an actual problem for simulation.

The dialog for Tessellate with Gmsh has code to close the dialog when the mouse leaves the window this maybe causing the one more feature problem

porosev, 2022-08-02:
Dear Keith,

You make great and very important job! 

Just for information. I did GDML file reading like in the example: Geant4-11.0.2\examples\advanced\CaTS in DetectorConstruction after initialisation of the RunManager and G4GDML: VALIDATION ERROR! with materials disappear!

Some comments about  #93.. 
Evaluating the CaTS example \Geant4-11.0.2\examples\advanced\CaTS\gdml\simpleLArTPC.gdml I have found that they keep all optical properties  inside of GDML file.  Anyway it has some drawback....  in some situations it requires many hand-work, because you should define wavelength dependent reflectivity, refraction, emission spectra of scintillators etc...

In my simulations (with using standard Box, Tubes etc.. solids to define a setup geometry) I try to use the following approach:

I define detector as:
...
class DetectorConstruction : public G4VUserDetectorConstruction, public DetectorSurfaces
...

this additional service class "DetectorSurfaces" is used for surfaces definitions from hand-typed data and reading from files:
/////////////////////////////////// Surfaces_h /////////////////////////////////////////////////
#ifndef Surfaces_h
#define Surfaces_h 1

#include <globals.hh>

class G4OpticalSurface;

class DetectorSurfaces 
{
	protected:
		// standard surfaces

		G4OpticalSurface *polishedAir;              // polished crystal surface, not wrapped
		G4OpticalSurface* polishedWhitePainted;     // polished crystal surface painted white
		G4OpticalSurface *PMTCathodeMaterial;       // surface between window and cathode

	public:
		DetectorSurfaces();
};

#endif
/////////////////////////////////// Surfaces_cc /////////////////////////////////////////////////
...
DetectorSurfaces::DetectorSurfaces()
{
	polishedAir           = NULL;              
	polishedWhitePainted  = NULL;     
	PMTCathodeMaterial    = NULL;       

	//Get ENVIROMENT VARIABLE TO MATERIAL DATA
	char* path = getenv("MY_G4DATA");
	if (!path)
	{
		G4Exception("DetectorConstruction::defineSurfaces()", "", FatalException, "MY_G4DATA environment variable not set.");
		return;
	}

	////////////////////////////////////////////////////////////////////
	//
	polishedAir = new G4OpticalSurface("polishedAir", unified);
	polishedAir->SetType(dielectric_dielectric);
	polishedAir->SetModel(unified);
	polishedAir->SetFinish(ground); // necessary even for polished surfaces to enable UNIFIED code
	polishedAir->SetSigmaAlpha(1.3 * degree); // Janecek2010

	//////////////////////////////////////////////////////////////////
	//
	// COMMON PROPERTIES
	G4double ener[2] = { .1*eV, 10.*eV };
	G4double paintspecularlobe[2] = { 1.0, 1.0 };
	G4double paintspecularspike[2] = { 0.0, 0.0 };
	G4double paintbackscatter[2] = { 0.0, 0.0 };

	//////////////////////////////////////////////////////////////////
	// EJ-510
	// https://eljentechnology.com/products/accessories/ej-510-ej-520
	// 
	G4double paintrindex[2] = { 1.62, 1.62 };

	polishedWhitePainted = new G4OpticalSurface("polishedWhitePainted");
	polishedWhitePainted->SetType(dielectric_dielectric);
	polishedWhitePainted->SetModel(unified);
	polishedWhitePainted->SetFinish(polishedbackpainted);
	polishedWhitePainted->SetSigmaAlpha(1.3 * degree);

	G4MaterialPropertiesTable* polishedWhitePaintedProperty = new G4MaterialPropertiesTable();
	polishedWhitePaintedProperty->AddProperty("RINDEX", ener, paintrindex, 2);
	polishedWhitePaintedProperty->AddProperty("SPECULARLOBECONSTANT", ener, paintspecularlobe, 2);
	polishedWhitePaintedProperty->AddProperty("SPECULARSPIKECONSTANT", ener, paintspecularspike, 2);
	polishedWhitePaintedProperty->AddProperty("BACKSCATTERCONSTANT", ener, paintbackscatter, 2);

	std::vector<double> dEJE;
	std::vector<double> dEJR;
	OptService::ReadParameters(path, "EJ-510_Reflectance.txt", dEJE, dEJR, true, CLHEP::eV);
	polishedWhitePaintedProperty->AddProperty("REFLECTIVITY", dEJE, dEJR);
	polishedWhitePainted->SetMaterialPropertiesTable(polishedWhitePaintedProperty);

	//////////////////////////////////////////////////////////////////
	// R10233
	PMTCathodeMaterial = new G4OpticalSurface("CathodeMaterial", unified);
	PMTCathodeMaterial->SetType(dielectric_metal);
	PMTCathodeMaterial->SetModel(unified);
	PMTCathodeMaterial->SetFinish(polished);
	PMTCathodeMaterial->SetSigmaAlpha(0.);
	G4MaterialPropertiesTable *PMTCathodeMaterialProperty = new G4MaterialPropertiesTable();

	std::vector<G4double> Cathode_energy;
	std::vector<G4double> Cathode_data;
	OptService::ReadParameters(path, "R10233_TrueReflectanceWithoutGlass.txt", Cathode_energy, Cathode_data, true, CLHEP::eV);
	PMTCathodeMaterialProperty->AddProperty("REFLECTIVITY", Cathode_energy, Cathode_data);

	std::vector<double> vEngEff;
	std::vector<double> vEff;
	OptService::ReadParameters(path, "R10233_QE.txt", vEngEff, vEff, true, CLHEP::eV);
	PMTCathodeMaterialProperty->AddProperty("EFFICIENCY", vEngEff, vEff);

	PMTCathodeMaterial->SetMaterialPropertiesTable(PMTCathodeMaterialProperty);
}

///////////////////////////////////////////////////////////////////////////////////////////////////

after that I can easily define any surface in my code (during detector construction) between any physical volumes:

G4VPhysicalVolume * DetectorConstruction::constructDetector()
{
   ....
  //-------------------------------------------------
  // setup surfaces
  //-------------------------------------------------
  new G4LogicalBorderSurface("scint2AirFaceBorderSurface", physiScint, physiWorld, polishedWhitePainted);

  new G4LogicalBorderSurface("grease2AirFaceBorderSurface", physiGrease, physiWorld, polishedAir);

  new G4LogicalBorderSurface("glass2CathodeBorderSurface", physiPMT, physiCathode, PMTCathodeMaterial);

  return physiWorld; 
}

So, I plan to test how it works with GDML because GDMLParser lets to get pointer to physical volume using its name...

class G4GDMLParser
{
....
GetPhysVolume(const G4String& name)
..
}

So, theoretically it should be possible....  for some projects ;))

KeithSloan, 2022-08-08:
porosev - Please could you move/copy you last comment about optical properties into #93, I would be interested in any comments from Munther, Ami or anybody else - Thanks

KeithSloan, 2022-08-10:
Please could I trouble you to try branch **TessProp**

With this branch, when objects are Tessellated a link property is created to the Tessellated Object. The property is checked for on export so if there is a valid Tessellated Object the source object is not exported.

I am still getting messages about Materials which Geant4 predefines not being defined but i think that is
due to validation rather than a real error.

[image]

[Gmeshed_Box.FCStd.txt](https://github.com/KeithSloan/GDML/files/9299021/Gmeshed_Box.FCStd.txt)
[Gmeshed_Box-worldVOL.gdml.txt](https://github.com/KeithSloan/GDML/files/9299022/Gmeshed_Box-worldVOL.gdml.txt)

porosev, 2022-08-11:
Dear Keith,

The Main branch works fine. The TessProp-branch generate 2 times bigger GDML file which contains dublicated description of the shape. Nevertheless, GEANT4 doesn't report any error :))

P.S. Some comments. In my private opinion, the command GDML Tesselate selected object is good for initial trials mainly. For import of more complex STEP shapes than the cube it produce terrible results (too many faces = too long simulation time). For example, I try import parts like this:

[image]

The tessellated object (strange irregular mesh) looks like this (GDML file - 1.2M): 

[image]

When I use two step approach: "Create mesh from shape" and "Mesh 2 Tess", I get more reasonable results:

with "Standard" (GDML file - 397K file)

[image]

Mefisto: (GDML file - 578K file)

[image]

Netgen + UserDefined settings : (GDML file - 1.3M file)

[image]

So, I expect that "Standard" mesh will be 3-4 times faster in simulations due to significantly smaller number of faces that GEANT4 should take into account.

Anyway, there are other problems with "holes", "over sections", "duplicated vertex" etc.. that arise in many cases and some parts require personal attention.

P.S.S... The stars changed they positions and GEANT4 start report VALIDATION ERROR about materials again. I think it is not real error too. It should be just WARNING during GDML parsing.

KeithSloan, 2022-08-12:
Thanks for all you cooperation.

"The TessProp-branch generate 2 times bigger GDML file which contains dublicated description of the shape. Nevertheless, GEANT4 doesn't report any error :))"

Are you able to supply the FC file as I don't think I see the same problem with the test files I supplied GMeshed_Box.
Loading the FC file and looking at the GDML_Tessellated object I see it has 272 vertex which agrees with the GDML file

[image]

Also 540 faces which agrees with GDML file

[image]

Also any chance of uploading the STEP file as I would like to use it for some more comparisons

porosev, 2022-08-12:
[TestShield.FCStd.txt](https://github.com/KeithSloan/GDML/files/9314367/TestShield.FCStd.txt)

KeithSloan, 2022-11-22:
@porosev You might want to try a new branch Gmsh-Min. A while back I got Christophe Geuzaine to add a facility to Gmsh to tessellate a Brep object and then combined adjacent Faces that are coplaner, the idea to create a minimum Mesh that
has both Triangular and Quad Faces. So if you use in on say a cube you get just 6 quad faces. Munther is on holiday this week, Thanksgiving in the US, he is going to check it out next week.

Needs recent version of Gmsh and Shared lib to FreeCAD installation.

porosev, 2022-11-24:
Dear Keith,

I tested Gmsh-Min. with my old "TestShield.FCStd.txt" geometry posted on Aug 12... and "GMSH Min Tessellate" command dies  slightly in different way than the "GMSH Tessellate" one ;))

GMSH:
[image]

GMSH-MIN
[image]

So, for me there is only one robust way:

Mesh_FromPartShape->Mesh_Evaluation->Mesh2TessCommand

KeithSloan, 2022-11-24:
Thanks for testing. When I try with

GMSH

It takes a long time then barfs with a very similar error, different surface number.
I will export the shape to Bref format and try native Gmsh to see if the same happens.

For me GMSH-Min it works, although the Mesh is not what I was expecting

 [TestShield_Gmsh_Min.FCStd.txt](https://github.com/KeithSloan/GDML/files/10086506/TestShield_Gmsh_Min.FCStd.txt)

[image]

KeithSloan, 2022-11-24:
Sorry My Bad.

Some where I am not clearing out the previous Gmsh operation/result, still have to track that down.

So then I got the same failure as you.

Have updated code to make more robust so that when Gmsh gives me 4 vertex that FC complains as not planar
I report the fact and create as two triangles.

See latest version

[image]

Least the Mesh looks nearer to what I was expecting

[TestShield-Gmsh-Min.FCStd.txt](https://github.com/KeithSloan/GDML/files/10086736/TestShield-Gmsh-Min.FCStd.txt)

porosev, 2022-11-25:
Dear Keith,

Yes, GMSH-min works now with previous object... but I tested it with another pieсe of my setup and it generates crazy-mesh :))

[image]

[TestShield2.FCStd.txt](https://github.com/KeithSloan/GDML/files/10088287/TestShield2.FCStd.txt)

KeithSloan, 2022-11-25:
Sorry must some more bugs in the way I am processing the Mesh that Gmsh creates, as if I perform the creation with a Gmsh script it looks okay.
[image]

porosev, 2022-12-09:
Dear Keith,
 
I tried to import GMSH-Min geometry to GEANT and have found a new feature with my favorite part :))

[image]

It looks like some vertices have the same coordinates and instead of G4QuadrangularFacets it is necessary generate TriangularFacets.

Additionally, if it possible, please generate an GMSH-ed object name based on transformed object label. Now they are transformed to  Feature000, . . ..Feature001 etc. and after import to GEANT they became undistinguished. 

[image]

Below my geometry with troubles during an import to G4.
[TestPart1.FCStd.txt](https://github.com/KeithSloan/GDML/files/10194894/TestPart1.FCStd.txt)

"May the Force be with you!"

KeithSloan, 2022-12-09:
Will look into the first two points.

By transformed object label, are you saying you would like in this case the Tessellate Part name to be
GDMLTessellate_Part_CH_SideShieldCu ?

porosev, 2022-12-09:
Dear Keith,

I will agree for any human readable name. As for me the suggested version will be ok. Because during debug it is painful to understood what's going wrong looking on strange names... 

[image]

Additionally, the latest version not export "Sens Det"  attribute to GDML file... only "Colors" ;((

KeithSloan, 2022-12-09:
branch testExport - Okay I think I have fix the Feature and add a check to see if the Facet Area is less than zero.
Not sure whether that is because Gmsh is creating them or an error in processing the Gmsh.

Still have a number of errors on Quads

*** G4Exception : GeomSolids1002
      issued by : G4TessellatedSolid::AddFacet()
Attempt to add facet not properly defined.
*** This is just a warning message. ***
-------- WWWW ------- G4Exception-END -------- WWWW -------

*********************************************************************
FACET TYPE       = G4QuadrangularFacet
ABSOLUTE VECTORS = 
P[0]      = (0.991066,-28.0227,29.1695)
P[1]      = (0.495533,-28.2614,28.8391)
P[2]      = (-3e-15,-28.5,28.5086)
P[3]      = (1.65,-28.5,29.5)
*********************************************************************

and
*** ExceptionHandler is not defined ***
*** G4Exception : GeomSolids1001
      issued by : G4QuadrangularFacet::G4QuadrangularFacet()
Facet has three or more collinear vertices.
P0 = (-1.65,28.5,29.5)
P1 = (-3e-15,28.5,28.5086)
P2 = (-0.495533,28.7386,28.8391)
P3 = (-0.991066,28.9773,29.1695)
Height in P0-P1-P2 = 0.240279
Height in P1-P2-P3 = 5.32708e-15
Height in P2-P3-P4 = 0.342227
Height in P4-P0-P1 = 0.480559
*** This is just a warning message. ***

porosev, 2022-12-12:
Dear Keith,

Could you restore the ability to export the "SensDet" attribute to a GDML file. The latest version of the program can define it, but does not export it to a file.  I wanted to try to do a simulation with GMSH-Min, but without the ability to define sensitive detectors it requires a lot of rework.

KeithSloan, 2022-12-12:
Hi porosev

Not sure what has changed with exportGDML

The code still has a test for the attribute, where vol would be the volume ( App::Part) in FreeCAD

```
   if hasattr(vol, "SensDet"):
        # SensDet could be enumeration of text value None
        if vol.SensDet != "None":
            print("Volume : " + volName)
            print("SensDet : " + vol.SensDet)
            ET.SubElement(
                xmlVol,
                "auxiliary",
                {"auxtype": "SensDet", "auxvalue": vol.SensDet},
            )
```
Please could you supply a test file where it defined and no longer exported, so I can investigate.
You could email (see Author at bottom of README ) if you don't want it shared on the internet

KeithSloan, 2022-12-12:
On a quick test I am not seeing a problem on export of SensDet, please supply a test file which exhibits the problem
Thanks
[testA.FCStd.txt](https://github.com/KeithSloan/GDML/files/10205379/testA.FCStd.txt)
[testA-worldVOL.gdml.txt](https://github.com/KeithSloan/GDML/files/10205380/testA-worldVOL.gdml.txt)

And just to clarify, are we talking about the Main branch?

KeithSloan, 2022-12-12:
I am attaching a couple of python scripts that perform the same Gmsh calls as the workbench.

We are investigating why the workbench loops rather than reports a problem

So, trying to mesh TestShield_Gmsh_MinBrep.brep with testGmsh.py gives an error. This is just calls to gmsh.py, no post processing, no FreeCAD.

```
Error   : Impossible to mesh periodic surface 45
Traceback (most recent call last):
 File "/home/hindi/geant/KeithSloan/NewGmsh/testGmsh.py", line 80, in <module>
   gmsh.model.mesh.generate(2)
 File "/home/hindi/.local/lib/python3.10/site-packages/gmsh.py", line 1973, in generate
[TestShield-Gmsh-Min.FCStd.txt](https://github.com/KeithSloan/GDML/files/10205484/TestShield-Gmsh-Min.FCStd.txt)

   raise Exception(logger.getLastError())
Exception: Impossible to mesh periodic surface 45
```
[testGmsh.py.txt](https://github.com/KeithSloan/GDML/files/10205464/testGmsh.py.txt)
[testMinGmsh.py.txt](https://github.com/KeithSloan/GDML/files/10205465/testMinGmsh.py.txt)

Selected FC object exported as Brep

[TestShield_Gmsh_MinBrep.brep.txt](https://github.com/KeithSloan/GDML/files/10205492/TestShield_Gmsh_MinBrep.brep.txt)
[TestShield-Gmsh-Min copy.FCStd.txt](https://github.com/KeithSloan/GDML/files/10205493/TestShield-Gmsh-Min.copy.FCStd.txt)

porosev, 2022-12-12:
Dear Keith,

below my geometry with "SensDet"... and output without it...

[TestPart2.FCStd.txt](https://github.com/KeithSloan/GDML/files/10206020/TestPart2.FCStd.txt)
[TestPart2.gdml.txt](https://github.com/KeithSloan/GDML/files/10206037/TestPart2.gdml.txt)

KeithSloan, 2022-12-14:
Okay have made a temp(?) fix.

Ideally the App::Parts should not be exported as Assemblies but this is not a quick fix with the current code and
we plan to do code review in the new year.

In the mean time I have made a fix so that the SensDet properties are exported.

We are still waiting for info about Gmsh to debug problem with Gmsh and Gmsh-Min

KeithSloan, 2022-12-19:
> I am attaching a couple of python scripts that perform the same Gmsh calls as the workbench.
>
> We are investigating why the workbench loops rather than reports a problem
>
> So, trying to mesh [...]

This has now been fixed by adding the option CharacteristicLengthMin of 1e-3

KeithSloan, 2023-03-07:
Closing as should now be fixed see last two examples at [https://github.com/KeithSloan/GDML/wiki/Gmsh---Min-Tessellate](url)

closed 2023-03-07 by KeithSloan