KeithSloan/GDML #103: Export assembly instead of multiunion

ufan, 2022-11-26:
Hi,

I tried to create a model with patterned layout of boxes. I used array function. The output file contains a multiunion solid for
the patterned boxes instead of assembly. It seems using mulitunion has performance gain. Problem is that ROOT's TGeo does not support this solid (the gdml parser doesnot recognize it). 

So, is it to export the pattern as an assembly instead of mulitunion?

KeithSloan, 2022-11-26:
A similar issue has come up in the support of G4Opt and a few macros were created see Macro directory in Repro
[https://github.com/KeithSloan/GDML/tree/Main/Macros](url)

  If my memory is correct 

  * ExpandArray and ExpandArrayCN3 Expand the Array to an Assembly that can then be exported 

    Sorry cannot remember the difference between ExpandArray and ExpandArrayCN3

  * Munther also created a Macro ExpandArray-MMH-5 which I think might expand direct to a gdml file.

I will ask Munther and Paulo to comment

mhindi2, 2022-11-26:
Try ExpandArray-MMH-5.FCMacro (on Keith's repo). It should expand into an assembly in the FreeCAD document and then you can export that as usual. You select the array that needs expansion and then you execute the Macro after the selection. Note that the original array will be removed, so perhaps you should save the modified document into a different file if you want to preserve the original (with array) structure. Also note that Geant generates its own names for members of an assembly, so you need to use those names if you need to retrieve the volume by its name. Check the Geant documentation.

ufan, 2022-12-03:
Hi,  thank you very much!

Sorry for the late reply. I tried all three macros. ExpandArray-MMH-5.FCMacro has some issue, the other two works.
The issue of MMH-5 macro is that it will delete the sheet which contains the parameters of my model. So
after conversion, freecad complains that it can't find these parameters.

I use latest 0.20.1 freecad.

KeithSloan, 2022-12-03:
Thanks for the feedback. I think ExpandArray was my original attempt. ExpandArrayCN3 was to create with CopyNumber properties. Ami and Munther who are in different time domain than me and  ExpandArray-MMH-5 was to address some issue.

ufan, 2022-12-03:
> Thanks for the feedback. I think ExpandArray was my original attempt. ExpandArrayCN3 was to create with CopyNumber properties.

Yes, and it also refers to the same lv volume instead of creating the separate with same dimension for all nodes in the assembly.

mhindi2, 2022-12-03:
I am not sure what you mean "freecad complains that it can't find these 
parameters". Yes, ExpandArray-MMH-5.FCMacro is meant to remove the 
original array after conversion. That is it's intended use. You should 
save the file under a different name if you want to preserve the 
original array.

Munther

ufan, 2022-12-04:
I'll try to give a detailed picture of what I observed.
I created a sheet named 'param', which is located at the top directory of the model file, in parallel with the world volume
part. I defined some aliases in this sheet to drive my model. When defining a gdml object, I use aliases from this sheet to
define the geometry dimensions, such as '<<param>>.box_x' '<<param>>.box_y'.
By changing the parameters in this sheet, the model changes as expected.

After I use your macro to convert an array object in the model. This 'param' sheet is lost.
The end effect is that, all the objects in the model complains it can't find the parameters.

Here is my freecad model for this test:
https://github.com/ufan/megat_geometry/blob/main/megat.FCStd

mhindi2, 2022-12-04:
> define the geometry dimensions, such as '<>.box_x' '<>.box_y'.

Ah, I see. Since the original model is deleted, parameters that refer to 
it will be missing. If I remember correctly, the other macros CN3 do not 
delete, but mark the original array so it will not be exported. However, 
I believe the other macros fail under certain circumstances, such is 
when there are assemblies of assemblies, with possible non-zero 
placements for both the containing volume and the array object. My 
version (...MMH-5...) handles those situations correctly. Perhaps we 
should come up with a new version that does not remove the original 
object. I'll take a look when I get a chance, but probably not soon. In 
the mean time, since the other versions work for you, use those!

Munther

ufan, 2022-12-14:
OK, thanks for your help. It's clear now, thus closing it.

KeithSloan, 2022-12-14:
Thanks for closing but I am going to leave as open and marked as Solved, until such time as information is added to the wiki,
in case others are trying to use Array's with ROOT rather than Geant4.

closed 2024-03-10 by KeithSloan