KeithSloan/GDML #107: Materials are not defined in the GDML file after the export

ozankilic7, 2023-03-22:
Hi Keith,

I export my simple model in GDML format to use in Geant4.
But Geant4 gives an error related to the materials.

```
error:  systemId: ...gdml line: 55 column: 8
ID attribute 'G4_A-150_TISSUE' was referenced but never declared
```

It looks like the materials are referenced in my GDML file but not defined. Aren't the materials defined automatically in GDML file when exporting? 

[image]

[image]

GDML file:
[gdml_file.txt](https://github.com/KeithSloan/GDML/files/11043904/gdml_file.txt)

KeithSloan, 2023-03-22:
Attached is a screen shot and listing when I use your file on my system.

And full listing and screen shot

The warnings
````
G4GDML: Reading '/Users/keithsloan/Downloads/gdml_file.gdml'...
G4GDML: VALIDATION ERROR! ID attribute 'G4_A-150_TISSUE' is referenced but was never declared at line: 55
G4GDML: VALIDATION ERROR! ID attribute 'G4_AIR' is referenced but was never declared at line: 55
G4GDML: Reading definitions...
G4GDML: Reading materials...
G4GDML: Reading solids...
G4GDML: Reading structure...
````
Warnings are normal for Geant4 materials and can be ignored, do you get an error or just a Validation Warning?

[image]

[List.txt](https://github.com/KeithSloan/GDML/files/11044854/List.txt)

mhindi2, 2023-03-23:
I believe the VALIDATIOP ERROR messages issued by the GDML reader about materials can be safely ignored, and are an indication that the materials were not defined in the GDML file. But they need not be define in the GDML, as long as the name is a valid name of a material in Geant's NIST data base. Obviuosly, if you use a material that is NOT in the NIST data base, then the material has to be defined in the GDML file.

As validation that the correct material is being used, despite the VALIDATION ERROR message, I provide the following tests:
I made a run with a test file that uses the materials G4_AIR for the Wold and G4_Water, for the trap solid.
The initial output shows the same VALIDATION ERROR you refer to:

G4GDML: Reading 'test_mat.gdml'...
G4GDML: VALIDATION ERROR! ID attribute 'G4_WATER' is referenced but was never declared at line: 88
G4GDML: VALIDATION ERROR! ID attribute 'G4_AIR' is referenced but was never declared at line: 88

However, in my initInter.mac file I have a command that I use for debugging:

/vis/ASCIITree/verbose 4
/vis/drawTree

This produces, among other output, the following lines:

  "World_PV":0 / "World" / "WorldBox"(G4Box), 8 L  , 1.20479 mg/cm3 (G4_AIR)
    "trap_PV":0 / "trap" / "trap1"(G4Trap), 4.224 cL , 1 g/cm3  (G4_WATER)
Calculating mass(es)...
Overall volume of "World_PV":0, is 8 L   and the daughter-included mass to unlimited depth is 51.8274 g 

So clearly, geant knows that ther materials assigned the World and the trap are G4_AIR and G4_WATER, respectively.
A screen shot with a 100 electrons run from inside the trap demosntrates that the particles are stopping in the water, as exected:
[image]

When I change both materials to G4_Galactic, again I get the VALIDATION ERROR:

G4GDML: VALIDATION ERROR! ID attribute 'G4_Galactic' is referenced but was never declared at line: 88

However, the output shows the solids having the correct materials:

  "World_PV":0 / "World" / "WorldBox"(G4Box), 8 L  , 1e-22 mg/cm3 (G4_Galactic)
    "trap_PV":0 / "trap" / "trap1"(G4Trap), 4.224 cL , 1e-22 mg/cm3 (G4_Galactic)

and a run with 100 electrons shows them not iteracting:
[image]

Note that if the material was truly undefined, then geant4 should not run. When I enter G4_Air as a material, the app I was running aborts with a message:

-------- EEEE ------- G4Exception-START -------- EEEE -------

*** ExceptionHandler is not defined ***
*** G4Exception : InvalidRead
      issued by : G4GDMLReadMaterials::GetMaterial()
Referenced material 'G4_Air' was not found!
*** Fatal Exception ***
-------- EEEE ------- G4Exception-END -------- EEEE -------

So, the upshot is that if the complainet is only from the GDML reader, they can safely be ignored.

lambdam94, 2023-03-23:
Hello all!
I am a little late, but I have the same answer,
@ozankilic7  your file work well under the classic Geant4 so you can do your simulation under this tool or under your gdml file, you should define a material compliant with spenvis (which shouldn't be fully compliant with the last version of the gdml definition).
Do you have a lot of different material or only a few?

KeithSloan, 2023-03-23:
There are G4 Materials defined in an XML file in the Materials Directory. If you import the xml file(s), preferable before you create you design, the G4 Materials defined in file will be added the same was as non Geant4 materials and should be included in the export.

See dataBaseNistAndSingleElements.xml and dataBaseSingleElements.xml

KeithSloan, 2023-03-24:
@ozankilic7 

When you open a new file with the GDML workbench active, it imports the file freecad/gdml/Resource/Default.gdml

You could edit this file and merge in the definitions in Geant4Materials.xml between 
<materials>
</materials>

And save as Default.gdml

I have done this and created Default-plus-G4Mat.gdml so you could try just replacing Default with this file, sorry not time to test and busy for next few days.

ozankilic7, 2023-03-24:
@lambdam94 Thanks for making me realize the I can actually define the materials in SPENVIS! I have only 4 materials to define, so it is not a big deal. On the Geant4 analysis side, things look alright.

I am now trying to define the materials in GDML for the sake of completeness. First I tried to import materials from Geant4Materials.xml, however it only defines the elements. For the materials it gives and error and does not define them in the GDML file.

[image]

@KeithSloan Changing Default.gdml file is also giving the same error for materials when I activate GDML workbench.

[image]

mhindi2, 2023-03-24:
@ozankilic7 I see where the bug is in the code. I will communicate with @KeithSloan to tell him where the bug is. I assume he will deal with it when he has time.

In the process of looking for the bug, I discovered that Geant4Materials.xml has a few errors in it (I think introduced by me when I updated the file). I will give the corrected file to @KeithSloan, but those corrections will not fix the above bug, so will have to wait for Keith to fix it.

In case you want to copy by hand the materials you use in your gdml file from the list of materials, here is the corrected file. I repeat, this will not fix the above error, so you still cannot import it on to the GDML Workbench.

[Geant4Materials.xml.txt](https://github.com/KeithSloan/GDML/files/11064098/Geant4Materials.xml.txt)

In the mean time I will try to see of I can provide a branch that does the required exports in the workbench, but no promises!

KeithSloan, 2023-03-24:
Import of Materials XML file hopefully now fix.

mhindi2, 2023-03-24:
I suspect if one imports ALL the materials in Geant4Materials.xml, then ALL of them will also be exported everytime a file is exported. I managed to create a version of the exporter that only exports the materials used in the created voumes, but we need some more work to avoid duplicates on re-import.
@ozankilic7, I suggest creating a version of the xml file that only includes the materials you need and just importing that. Perhaps you were already doing this.

KeithSloan, 2023-03-25:
Munther has added a  GDML workbench preference for exporting Geant4 materials used within the FreeCAD document
[image]

closed 2023-03-25 by KeithSloan