KeithSloan/GDML #111: Error during import

nicolamori, 2023-05-18:
Trying to import a GDML file I get this error in the report view:
```
10:32:05  solidref : worldbox
10:32:05  ERROR - Solid  : worldbox Not defined
```
and no element is then shown in the FreeCAD view. This is the relevant section of the GDML file:
```
  <define>
    <position name="center" x="0.153" y="-0.065" z="-0.17" unit="mm"/>
  </define>
  <solids>
    <box name="worldbox" x="101.788" y="101.788" z="101.788" lunit="mm"/>
  </solids>
  <structure>
    <volume name="V-bigAluBlock-0">
      <materialref ref="ALUMINUM"/>
      <solidref ref="T-bigAluBlock-0"/>
    </volume>
    <volume name="World">
      <materialref ref="VACUUM"/>
      <solidref ref="worldbox"/>
      <physvol name="P-bigAluBlock-0">
        <volumeref ref="V-bigAluBlock-0"/>
        <positionref ref="center"/>
      </physvol>
    </volume>
  </structure>
  <setup name="Default" version="1.0">
    <world ref="World"/>
  </setup>
```
The file is imported in Geant4 with no problem, so I guess it's an issue with the FreeCAD GDML workbench.
Ia am using FreeCAD 0.20.2 and just installed the workbench.

KeithSloan, 2023-05-18:
Not sure what is going on with import as the solid worldbox is defined in your gdml snip-it.

I added gdml tag to your snipt-it and that does not work because the Materials are not defined by default.
Changed to Geant4 materials and then had to add the solid T-bigAluBlock-0 which then imported okay,
no complaint about worldbox Not Defined see ( you need to change from a txt file to a gdml file) 
[testWorldBox3.gdml.txt](https://github.com/KeithSloan/GDML/files/11510011/testWorldBox3.gdml.txt)

Are you able to supply the problem file? keith[at]sloan-home.co.uk

KeithSloan, 2023-05-19:
The problem is that there are TWO solids sections in the file.

One with a the solid tessellated T-bigAluBlock-0 and another with the box worldbox.

```
<solids>
    <tessellated name="T-bigAluBlock-0">
      <triangular vertex1="18" vertex2="20" vertex3="19" type="ABSOLUTE"/>
      <triangular vertex1="17" vertex2="21" vertex3="20" type="ABSOLUTE"/>
      ....
</tessellated>
  </solids>
  <define>
    <position name="center" x="0.153" y="-0.065" z="-0.17" unit="mm"/>
  </define>
  <solids>
    <box name="worldbox" x="101.788" y="101.788" z="101.788" lunit="mm"/>
  </solids>

Did you use the workbench to create the GDML file?
If yes then I think we have a bug in export.

Plus as it loads in Geant4 okay, we should, also handle okay.

nicolamori, 2023-05-19:
No, I obtained the GDML converting a STEP file using the MRADSIM tool. But since Geant4 opens it without complaining then I thought it was a valid GDML and thus it could be imported in the workbench.

KeithSloan, 2023-05-19:
Okay I have pushed a new branch called worldSolid, which deals with multiple solid sections for this situation.

There are number of other situations where solid information is retrieved that will/may need updating for multiple solid sections and wish to discuss with Munther before embarking on some code restructuring.

Are you able to test the new branch( or do I need to push to Main branch)

nicolamori, 2023-05-19:
I'd need a bit of guidance about how to do that. I'm a new FreeCAD user and I don't know how to install the development version of the workbench.

KeithSloan, 2023-05-21:
Sorry have been away for a couple of days.

Okay  - have merged the update into the Main repro.

Will do some restructuring so that other times the solids section is accessed it should still work with multiple solid sections but that may take a while.

Would appreciate  any further testing and or reporting of problems.

nicolamori, 2023-05-22:
Hi Keith,

I uploaded the workbench and now it can import the gdml file correctly, thank you very much! But I tried it also on different file obtained in the same way (conversion from step to gdml using MRADSIM, correctly imported by Geant4) and I get this error:
```
10:38:22  Process Volume : V-1888-15
10:38:22  solidref : T-1888-15
10:38:22  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 96, in open
    processGDML(doc, True, filename, True, False)
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 3093, in processGDML
    parseVolume(doc, volDict, part, world, phylvl, 3)
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 2062, in parseVolume
    expandVolume(doc, volDict, parent, name, phylvl, displayMode)
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 2426, in expandVolume
    processVol(doc, vol, volDict, parent, phylvl, displayMode)
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 2353, in processVol
    part = parsePhysVol(
           ^^^^^^^^^^^^^
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 1996, in parsePhysVol
    expandVolume(doc, volDict, part, volRef, phylvl, displayMode)
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 2426, in expandVolume
    processVol(doc, vol, volDict, parent, phylvl, displayMode)
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 2305, in processVol
    createSolid(
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 1882, in createSolid
    return createTessellated(
           ^^^^^^^^^^^^^^^^^^
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/importGDML.py", line 1542, in createTessellated
    v1 = GDMLShared.getDefinedPosition(v1name)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mori/.local/share/FreeCAD/Mod/GDML/./freecad/gdml/GDMLShared.py", line 486, in getDefinedPosition
    pos = positions[name]
          ~~~~~~~~~^^^^^^
<class 'KeyError'>: ('102',)
```
If you are interested in troubleshooting also this then I can send you this gdml.

KeithSloan, 2023-05-22:
Yes please send file.

KeithSloan, 2023-05-22:
Okay similar issue to previous file but this time the gdml file has

materials
defines
solids
defines
solids
...
etc

Don't suppose MRADSIM tool has an option to create as flat xml structure, without repeats?
i.e. materials, defines, solids, structure

nicolamori, 2023-05-22:
The MRADSIM version I'm using is quite old, and I don't have access to a newer one. I don't see any option to tune the output file structure, so I guess it's not available at least in my version.

To be clear, I don't have an urgent need to open that file with the GDML workbench; it works in Geant4 so I'm using it as is. It's just that during my first test with the workbench I stumbled upon this bug and thought it was a good idea to report it. But it's not disrupting my workflow at the moment, so it's ok if you decide to not work on it.

KeithSloan, 2023-08-19:
Hi I have an idea for addressing this, please could you resend me the file as I am afraid I cannot now locate it.

Thanks

nicolamori, 2023-08-21:
Hi Keith, I lost the track of this issue and I'm not 100% sure which was the last file I sent you, but if I correctly reconstructed the issue history from my email archive then it can be downloaded from here:

[https://basket.fi.infn.it:443/f/f6c0e53643/?dl=1](https://basket.fi.infn.it/f/f6c0e53643/?dl=1)

Thank you.