KeithSloan/GDML #114: >180GB RAM usage when using `G4BooleanSolids`

SimonGLM, 2023-08-17:
First of all thank you for this project, you saved me a lot of time during sim's.

I am currently trying to simulate a kind of collimator, i.e. a block with many hollow channels.
In my `G4VUserDetectorConstruction`, I iteratively substract channels from the collimator body like this:
```c++
G4VSolid *collimator= new G4Box( ... );
G4VSolid *channel= new G4Box( ... );
for (auto pos : channelLocations)
    collimator = new G4SubtractionSolid( ... , collimator, channel, pos);
```
In total, about 1300 channels have to be substracted from the solid.

In Geant4 tracking complexity scales proportionally to the number of volumes contributing to a `G4BooleanSolid`.
Due to this fact, the tracking is very slow.

Then I came across your Workbench.
I took a simplified version of the described geometry and loaded it into FreeCAD, which took ~20min.
The geometry was then meshed with `gmsh` and then exported into a new `.gdml` file.
Importing this geometry into Geant4, was a breeze and the simulation time was reduced to 1/30th of what it was!
So far, so good.
Let's try with the full geometry:
I tried to import that into FreeCAD over night, which crashed at some point in the night (it also complained about max recursion depth, but I simply increased that to 1e6).
So then I spend a day packaging FreeCAD and the GDML workbench into a docker container to run on our server.
But as the title says, the OOM-Killer killed the container with ~180GB of RAM usage.

I cannot imagine, that this much memory is actually needed to load the geometry into FreeCAD, but I lack the experience to see which part of the code is at fault.

If someone could give me some pointers where I could start my investigations, I would very much welcome your input.

Cheers

mhindi2, 2023-08-17:
Hello Simon,

This is Munther Hindi, one of the contributors to the gdml workbench. Both Keith Sloan and I are currently on vacation and will likely be busy for the next couple 9f weeks (and me for the next month). However, we are definitely interested in solving your problem and are confident we can find solutions and/or workarounds. If it is possible to either post on the GitHub, or send via email (to the address posted at the bottom of the GitHub page), a scaled down version of your design, or, if possible, a hand sketch of the desired geometry and we will try to figure something out as soon as we have a chance.

Thanks,

Munther Hindi

KeithSloan, 2023-08-18:
To help me understand the situation. a little better, a couple of questions

1) Could you take the model with the subtractions import into FreeCAD and then perform the subtraction by creating them as an Array and subtracting that from the main body? Arrays are exported as MultiUnions which are more efficient than a set of single Objects.

If the objects don't fit with an array, for another user we did add code to allow for import of a GDML object using a FC point cloud but the user in the end went for a FC Macro to place the GDML objects which is probably easier and certainly no more difficult that writing a program to create the point cloud (obviously you would be subtracting). You mention recursion so again a FC macro maybe the answer.

What is the purpose of using Gmsh on the imported model? given the model is defined, generally GDML solids are faster than GDML Tessellations.

KeithSloan, 2023-08-18:
One point I forgot mention re your recursion is that you can have Arrays or Arrays see https://github.com/KeithSloan/GDML/wiki#arrays-of-objects.

You might try and quiz Ami Hashemi https://www.linkedin.com/in/amirreza-hashemi-41594044/ as he now has quite a bit of experience of creating things with Array's or Arrays. He has also used python to create GDML Objects in the workbench.