KeithSloan/GDML #116: Missing required postional arguments in expandVolume

ast0815, 2023-10-25:
Hello,

I am trying to import a large gdml file with the "Scan Vol" feature. The first import of the file seems to work alright, but if I then try to expand any volumes, I get this error:

```
12:52:26  NOT_Expanded_
12:52:26  Expand Function
12:52:26  Running the Python command 'ExpandMaxCommand' failed:
Traceback (most recent call last):
  File "/home/lukas/snap/freecad/common/Mod/GDML/./freecad/gdml/GDMLCommands.py", line 3119, in Activated
    expandFunction(obj, -1)
  File "/home/lukas/snap/freecad/common/Mod/GDML/./freecad/gdml/GDMLCommands.py", line 3064, in expandFunction
    expandVolume(obj, volRef, eNum, 3)

expandVolume() missing 2 required positional arguments: 'phylvl' and 'displayMode'
```

Looks like it is related to this line in the code:

https://github.com/KeithSloan/GDML/blob/aabb33e7c46151eec0b816ea7f54ec862ffdb6b1/freecad/gdml/GDMLCommands.py#L3064

There are 4 arguments there, while the function definition indeed requires 6:

https://github.com/KeithSloan/GDML/blob/aabb33e7c46151eec0b816ea7f54ec862ffdb6b1/freecad/gdml/importGDML.py#L2418

Is there a work-around for this or a quick way to fix this?

KeithSloan, 2023-10-25:
Thanks for reporting I am away from my main computer till Friday.

You could try changing line 3064 in expandFunction to
`
expandVolume(FreeCAD.ActiveDocument, {}, obj, volRef, eNum, 3) 
`
Other changes required the need to keep a dictionary of Volumes, namely optical surfaces,  so passing {} for volDict should stop it barfing but will not help if surfaces are involved.

Also expandVolume now required parent and name and somehow the selected object obj and volRef does not look right.

I will try and look at on Friday

I don't know if it will help but if you look at repro https://github.com/KeithSloan/CERN-Alice-MultiFile-gdml

There is a utility in newUtils called buildDirStruct.py which will create a directory structure where each volume/assembly is in its own directory. It runs python command line i.e. not under FreeCAD and for CERN's ALICE experiment takes just over 12 hours on my Mac. Once run you should be able to look at lower level volumes assemblies.

I am in the process of developing a macro to create a step file in each directory starting at the lower levels and adding an option to display the step file rather than the Non_Expanded used by Scan

KeithSloan, 2023-10-25:
Okay I have made the change to the Main branch.

Will have to do some checking of things that use the volDict on Friday

ast0815, 2023-10-27:
That `buildDirStruct.py` script seems to be working great, thanks!

closed 2024-03-10 by KeithSloan