KeithSloan/GDML #118: Exclude objects by material for STEP export?

ast0815, 2023-10-27:
My main objective for handling gdml files is to create thumbnails of the geometry in a batch process (as part of a continuous integration check of the geometry). For this, the "Air" volumes are super annoying because they cover up all the "actual" geometry inside.

Is it possible to exclude geometries by material when exporting the STEP file?

I have tried to modify the imported geometry by looping over `App.ActiveDocument.Objects`, but it looks like not all geometry elements are in there for some reason.

KeithSloan, 2023-10-27:
Are the "Air" volumes World Volumes? or within other Volumes?

If you look at this GDML file

[image]

i.e the Root Volume  rootDCoil has a solid WorldBox which is G4_Air

If you select the sub volume DCoil you can export that as a STEP file.

If you look at the OutList for the rootDCoil object you get
`
>> print(obj.OutList)
[<App::Origin object>, <Part::PartFeature>, <Part object>]
>>> 

The first item in the list is the Origin Object, the second is the GDMLBox object and you can check its material,
if G4Air then you don't export obj but if you export obj.OutList[2] that should be be DCoil

Will not help with sub volumes with G4_Air.
` 

"Is it possible to exclude geometries by material when exporting the STEP file?"

FreeCAD uses the OCC facility to export STEP files, it will create a Step file from a Selected Objects Shape and SubShapes. If you drop a Volume with Material set to Air then that would cause a problem with any SubShapes

Have you looked at buildDirStruct.py in https://github.com/KeithSloan/CERN-Alice-MultiFile-gdml ? And the FreeCAD macro ProcessDirStuct in the same repro. 

Obviously not very efficient to create a directory for each volume but you could create a modified version of the macro ProcessDirStruct to output step files for sub volume/assemblies where the root volume had a solid with Material set to AIr

It is doing the step export under FreeCAD. I think you can do with with python outside of FreeCAD but it was easier for me to do as a Macro. If doing export of Step from python outside FreeCAD this should be useful https://wiki.freecad.org/Embedding_FreeCAD

KeithSloan, 2023-11-04:
Okay have made an update to importGDML, if the GDML Material is G4_AIR I set the ViewOption.Transparency to 98.

I am told that export of STEP with option AP242 should support Transparency but does not appear to be working on my Max. Your milage may vary.

KeithSloan, 2023-11-04:
It appears to be working on Linux see last post of https://forum.freecad.org/viewtopic.php?t=82445

ast0815, 2023-11-23:
Would it be possible to modify or set the material name that gets set to invisible? In my case, the material is just "Air" and not "G4_AIR". At least I assume that is the issue that causes nothing to be changed on my side after updating the addon.

ast0815, 2023-11-23:
I could have a go at implementing this, but I just saw, that the `setDisplayMode` function is only called  if `FreeCAD.GuiUp`. Does that mean that this will not help for batch conversion of gdml to step with `gdml2step.py`?

KeithSloan, 2023-11-23:
As it involves the ViewObject as it stands it does require running under Gui.

You should be able to use a FreeCAD macro to process as per ProcessDirStruct_Brep.FCMacro
Need to change to creating Step file rather than Brep

I have modified it to check for AIR as well ad G4_AIR

KeithSloan, 2023-11-23:
Have add Macro ProcessDirStruct_Step.FCMacro

You probably want to change it not use the root part as this will contain a Part(Volume) with the WorldBox if you know you are just dealing with GDML files where the top object is a volume rather than an Assemly

KeithSloan, 2023-11-24:
Sorry there is a mistake in the Macro ProcessDirStruct_Step.FCMacro I posted.

The call to freecad.gdml.importGDML.open(path, processType = 2, prompt=False)

The processType should be 3 for step

1 means just parse/intepret gdml
2 means if there is a Brep filr for any volume use that, otherwise parse/interpret gdml
3 means if there is a Step file for any volume use that.   otherwise parse/interpret gdml

closed 2024-03-10 by KeithSloan