KeithSloan/GDML #127: `opticalsurface` import treats `finish` attribute differently from Geant4

shangjiaxuan, 2024-03-07:
`finish` attribute specifies the optical surface's polish, and is saved in geant4 as numeric number instead of text of the enumeration (c/cpp enumerations are saved as integers and only the integer value is known at runtime).

The workbench can only import numeric values in 0-9 range as `extended | z`. This means enumeration like 36, which is `PolishedESR_LUT` fails to import. Export from workbench works fine, since Geant4 recognizes text representation.

shangjiaxuan, 2024-03-07:
Similarly, workbench does not accest textual `type` attribute such as `dielectric_LUTDAVIS`

shangjiaxuan, 2024-03-07:
The `model` parameter is not read, and always default to `glisur`

KeithSloan, 2024-03-07:
Thanks for reporting

To clarify, you are trying to import a GDML file where polish is given as a numeric values, rather than the text of a enumeration value?

PolishedESR_LUT, dielectric_LUTDAVIS, please could you clarify which version of Geant4 you are using as I think we need to update the workbench for these values, the workbench currently just has the values as in Geant 4.10.

"model parameter not being read" - Does the xml element 'opticalsurface' have a name attribute?
If not that would explain why things are defaulting to glisur.

Please could you supply a section of xml with the opticalsurface element for testing

shangjiaxuan, 2024-03-08:
I'm using geant4 with this output version string: `geant4-11-01-patch-01 [MT]   (10-February-2023)`
The enumeration in header file is:
```
enum G4OpticalSurfaceFinish
{
  polished,              // smooth perfectly polished surface
  polishedfrontpainted,  // smooth top-layer (front) paint
  polishedbackpainted,   // same is 'polished' but with a back-paint

  ground,              // rough surface
  groundfrontpainted,  // rough top-layer (front) paint
  groundbackpainted,   // same as 'ground' but with a back-paint

  // for LBNL LUT model
  polishedlumirrorair,   // mechanically polished surface, with lumirror
  polishedlumirrorglue,  // mechanically polished surface, with lumirror &
                         // meltmount
  polishedair,           // mechanically polished surface
  polishedteflonair,     // mechanically polished surface, with teflon
  polishedtioair,        // mechanically polished surface, with tio paint
  polishedtyvekair,      // mechanically polished surface, with tyvek
  polishedvm2000air,     // mechanically polished surface, with esr film
  polishedvm2000glue,    // mechanically polished surface, with esr film &
                         // meltmount

  etchedlumirrorair,   // chemically etched surface, with lumirror
  etchedlumirrorglue,  // chemically etched surface, with lumirror & meltmount
  etchedair,           // chemically etched surface
  etchedteflonair,     // chemically etched surface, with teflon
  etchedtioair,        // chemically etched surface, with tio paint
  etchedtyvekair,      // chemically etched surface, with tyvek
  etchedvm2000air,     // chemically etched surface, with esr film
  etchedvm2000glue,    // chemically etched surface, with esr film & meltmount

  groundlumirrorair,   // rough-cut surface, with lumirror
  groundlumirrorglue,  // rough-cut surface, with lumirror & meltmount
  groundair,           // rough-cut surface
  groundteflonair,     // rough-cut surface, with teflon
  groundtioair,        // rough-cut surface, with tio paint
  groundtyvekair,      // rough-cut surface, with tyvek
  groundvm2000air,     // rough-cut surface, with esr film
  groundvm2000glue,    // rough-cut surface, with esr film & meltmount

  // for DAVIS model
  Rough_LUT,              // rough surface
  RoughTeflon_LUT,        // rough surface wrapped in Teflon tape
  RoughESR_LUT,           // rough surface wrapped with ESR
  RoughESRGrease_LUT,     // rough surface wrapped with ESR
                          // and coupled with optical grease
  Polished_LUT,           // polished surface
  PolishedTeflon_LUT,     // polished surface wrapped in Teflon tape
  PolishedESR_LUT,        // polished surface wrapped with ESR
  PolishedESRGrease_LUT,  // polished surface wrapped with ESR
                          // and coupled with optical grease
  Detector_LUT            // polished surface with optical grease
};

enum G4OpticalSurfaceModel
{
  glisur,   // original GEANT3 model
  unified,  // UNIFIED model
  LUT,      // Look-Up-Table model (LBNL model)
  DAVIS,    // DAVIS model
  dichroic  // dichroic filter
};
```
```
enum G4SurfaceType
{
  dielectric_metal,       // dielectric-metal interface
  dielectric_dielectric,  // dielectric-dielectric interface
  dielectric_LUT,         // dielectric-Look-Up-Table interface
  dielectric_LUTDAVIS,    // dielectric-Look-Up-Table DAVIS interface
  dielectric_dichroic,    // dichroic filter interface
  firsov,                 // for Firsov Process
  x_ray,                  // for x-ray mirror process
  coated                 // coated_dielectric-dielectric interface
};
```
Here is a sample that I was using (I've edited the numbers to enumeration text to make it importable to workbench, and also added a few lines in `GDMLObjects.py`).
[OpNovice_edit.zip](https://github.com/KeithSloan/GDML/files/14532448/OpNovice_edit.zip)

Edit in `GDMLObjects.py`
```
        obj.model = [
            "glisur",  # original GEANT3 model \
            "unified",  # UNIFIED model  \
            "LUT",  # Look-Up-Table model (LBNL model) \
            "DAVIS",  # DAVIS model \
            "dichroic",  # dichroic filter \
        ]
+++     obj.model = model
        obj.addProperty(
            "App::PropertyEnumeration", "finish", "GDMLoptical" "finish"
        )
```

KeithSloan, 2024-03-08:
Okay I have added the missing SurfaceTypes and fixed the problem that the model passed to GDMLopticalsurface object was not set.

Please could you check the latest version - Thanks.

shangjiaxuan, 2024-03-08:
Doesn't seem like converting numbers to python enumerations. The sample is already edited to use textual instead of numeric representation, here is a sample of one that has numeric values.
[OpNovice_edit.zip](https://github.com/KeithSloan/GDML/files/14535441/OpNovice_edit.zip)

KeithSloan, 2024-03-08:
Okay mixing numbers and enumerations presents a small challenge.

Currently finish can be any of the enumerations or 0-9., which gets set a extended | n
same with type.

If I change finish to work the same way as 0-9 but for any numerical value, does that work?

shangjiaxuan, 2024-03-08:
I think it should work, since they should be either numeric or textual. Strings that start with numbers should be all-numbers in Geant4 logic.

KeithSloan, 2024-03-08:
Are you okay with testing a branch?

I have created a branch OptSurfaceFix, which you should be able to access as origin/OptSurfaceFix

it is not quite there in that I have coded onChanged functions for the values and corresponding valueNum
but for some reason if I change one the code to change its partner is not reflecting in the property window.

ie. so If you want to change finish to a numeric values you have to change finish to "Numeric" and finishNum to the number you require.

KeithSloan, 2024-03-08:
After some testing merging optSurfaceFix with Main Branch.

closed 2024-05-21 by KeithSloan