KeithSloan/GDML #128: Incorrect Mother-Daughter Assembly Volume Relationship

eleonar, 2024-04-29:
When exporting a linear array of objects, the GDML exporter creates a volume and places it inside the assembly. When creating an assembly in GDML the tag is used in a similar way to a logical volume and the mother volume does the placement. 
```
....
<volume name="assay_cont">
    <materialref ref="G4_AIR"/>
    <solidref ref="assay_container_box"/>
    <physvol name="PV_assay">
        <volumeref ref="assay"/>
        <positionref ref="center"/>
        <rotationref ref="identity"/>
    </physvol>
</volume>
<assembly name="assay">
    ....
</assembly>
....
```

Attached is screen shot of A default behavior of creating an array, a screenshot of the desired result and a gdml. The commented-out items are the what the code produces and the uncommented is what is supposed to happen (note: the .txt was added to allow upload to GitHub).  If you wish to have the FreeCAD files for troubleshooting, please contact me at eleonar@sandia.gov

 
[image]
[image]

[desired_array_behavior-worldVOL.gdml.txt](https://github.com/KeithSloan/GDML/files/15155595/desired_array_behavior-worldVOL.gdml.txt)

KeithSloan, 2024-04-30:
The Array facility uses the Draft workbench - Array command and it places the Array at the base of the document.

The **wiki has the following info**

Arrays of Objects

Arrays of objects, constructed under the Draft Array command, or the GDML Array command can be exported as GDML objects. The arrays must be of currently supported objects (GDML, or Part objects, such as extrusions, or revolutions). Arrays of arrays can also be exported. Currently the following types of arrays are supported:

**"The Array object must be added to a Part under the WorldVolume tree**. Note that the array object is exported as a single multiUnion gdml solid. Thus the objects in the array are considered as part of one solid, and hence have one material. It is best practice to assign the material to the Array object. If a material is not assigned to the Array object itself, its (sub)objects are scanned recursivley and the material of the first object that does have a material property is selected."

So the second screen shot should be correct.

I will look at fronting the commands for extrude | revolve | mirror | array.

But it would also have to deal with the situation where the user wishes to create an Array of an Array etc.

KeithSloan, 2024-05-08:
Please checkout Branch **Elliott**

eleonar, 2024-05-08:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Got it! I'll probably do this at the end of the day for me in about 3 hr.

Best,

Elliott Leonard

KeithSloan, 2024-05-09:
Thanks again for raising the problem and checking the fixes.

closed 2024-05-09 by KeithSloan