KeithSloan/GDML #131: Regarding the issue of installing GDML workbench based on FreeCAD0.21.2 on Windows

Gadajilehu, 2024-05-30:
I installed gmsh and lXML in the site packages folder in the root directory of freeCAD according to the tutorial and confirmed that they can be used. However, when I wanted to install GDML workbench in freeCAD through the Addon manager, it prompted me to install a library called veto. I'm not sure if this library must be installed in the freeCAD folder or on my computer. This may be a beginner's question. But I am indeed confused because I successfully installed vedo on my computer using pip, but there were many errors when I tried to install vedo in the site packages folder.
[image]

KeithSloan, 2024-05-30:
Thanks for reporting

FreeCAD needs python libraries installed in a location different to standard Python.

The installation of vedo library is for something I am working on, it is not currently used for the Main branch.
I will push an update to remove it as a required library until this gets resolved

KeithSloan, 2024-06-12:
There are problems with FreeCAD 0.21.2 Please use a different version of FreeCAD