KeithSloan/GDML #132: Unable to create new geometry in GDML workbench

Gadajilehu, 2024-05-31:
Thank you for your response to my previous question，and I have a new problem.I feel like I have successfully installed the GDML workbench. I can import gmsh and lXML from the freecad command line, but when I create any geometry in the workbench, this error will be reported.
``` shell
11:58:20  Running the Python command 'EllipsoidCommand' failed:
Traceback (most recent call last):
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\GDMLCommands.py", line 1201, in Activated
    GDMLEllipsoid(obj, 10, 20, 30, 0, 0, "mm", material)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\GDMLObjects.py", line 1050, in __init__
    setMaterial(obj, material)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\GDMLObjects.py", line 160, in setMaterial
    rebuildMaterialsList()
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\GDMLObjects.py", line 108, in rebuildMaterialsList
    addMaterialsFromGroup(doc, MaterialsList, "Materials")
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\GDMLObjects.py", line 87, in addMaterialsFromGroup
    buildDefaultGDMLDoc(doc)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\GDMLObjects.py", line 94, in buildDefaultGDMLDoc
    processGDML(
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\importGDML.py", line 3365, in processGDML
    preProcessLoops.preprocessLoops(root)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\preProcessLoops.py", line 97, in preprocessLoops
    for loop in root.getroot().iterdescendants(tag="loop"):

'xml.etree.ElementTree.Element' object has no attribute 'getroot'
'''

KeithSloan, 2024-05-31:
Are you following https://github.com/KeithSloan/GDML/wiki/Model_Creation ?

That is the workflow of

Activate GDML Workbench
New file

Should give you a structure that looks like
[image]

From the error messages it looks like you are using the Ellipse Command and it is not finding the Materials Group that is setup by the above workflow and it is trying to correct for this i.e. buildDefaultDoc function Objects line 87 and there is a problem with this recovery action.

I will look into this but would be good if yiu could confirm.

A file save after the initial setup should look like
[initialFile.FCStd.txt](https://github.com/user-attachments/files/15508982/initialFile.FCStd.txt)

KeithSloan, 2024-05-31:
I tried to recreate your problem by deleting the Materials Group and then using the Ellipsoid Command but it did not fail. Please could you send me a copy of the Report View from your FreeCAD

`
06:55:04  Activated
06:55:09  running with lxml.etree
06:55:09  processGDML type 1
06:55:09  Print Verbose : False
06:55:09  Import GDML file : /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/./freecad/gdml/Resources/Default.gdml
06:55:09  ImportGDML Version 1.9b
06:55:09  pathName /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/freecad/gdml/Resources
06:55:09  Parse : /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/./freecad/gdml/Resources/Default.gdml
06:55:09  running with lxml.etree 
06:55:09  Process Positions
06:55:09  Process Materials DocSet
06:55:09  define xml <Element define at 0x1160d7580>
06:55:09  materialsl <Element materials at 0x11be66b00>
06:55:09  Process Materials : Materials
06:55:09  Process - Opticals: matrix_spreadsheet
06:55:09  define xml <Element define at 0x1160d7580>
06:55:09  Find all Matrix
06:55:09  process GEANT4 Materials : /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/./freecad/gdml/Resources/Geant4Materials.xml
06:55:09  Parse : /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/./freecad/gdml/Resources/Geant4Materials.xml
06:55:09  running with lxml.etree 
06:55:09  process new G4
06:55:09  Process Materials : G4Materials
06:55:09  New getGroupedMaterials len GroupMaterials 0
06:55:09  doc.G4Materials <group object>
06:55:09  doc.Materials <group object>
06:55:09  Geant4
06:55:09  Parse Volume : worldVOL Phylvl -1 parent worldVOL
06:55:09  expandVolume : worldVOL importFlag 1
06:55:09  Parse Volume : worldVOL Phylvl -1
06:55:09  Process Volume : worldVOL importFlag1
06:55:09  solidref : WorldBox
06:55:09  Material G4_AIR
06:55:09  Set transparency
06:55:09  Process PhysVols importFlag 1
06:55:09  name: worldVOL parentpart = <Part object>
06:55:09  paramvol = None
06:55:09  ProcessVol returning <Part::PartFeature> GDMLBox_WorldBox
06:55:09  Process Surfaces
06:55:09  skinsurface
06:55:09  bordersurface
06:55:10  End processing GDML file
06:55:10  time : 0.8026 seconds
06:55:52  getstate : Type GDMLBox
07:04:38  processGDML type 1
07:04:38  Print Verbose : False
07:04:38  Import GDML file : /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/./freecad/gdml/Resources/Default.gdml
07:04:38  ImportGDML Version 1.9b
07:04:38  pathName /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/freecad/gdml/Resources
07:04:38  Parse : /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/./freecad/gdml/Resources/Default.gdml
07:04:38  running with lxml.etree 
07:04:38  Process Positions
07:04:38  Process Materials DocSet
07:04:38  define xml <Element define at 0x155f4a2c0>
07:04:38  materialsl <Element materials at 0x155f49e40>
07:04:38  Process Materials : Materials
07:04:38  Process - Opticals: matrix_spreadsheet
07:04:38  define xml <Element define at 0x155f4a2c0>
07:04:38  Find all Matrix
07:04:38  process GEANT4 Materials : /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/./freecad/gdml/Resources/Geant4Materials.xml
07:04:38  Parse : /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/./freecad/gdml/Resources/Geant4Materials.xml
07:04:38  running with lxml.etree 
07:04:38  process new G4
07:04:38  Process Materials : G4Materials
07:04:38  New getGroupedMaterials len GroupMaterials 5
07:04:38  doc.G4Materials <group object>
07:04:38  doc.Materials <group object>
07:04:38  Geant4
07:04:38  Parse Volume : worldVOL Phylvl -1 parent worldVOL
07:04:38  expandVolume : worldVOL importFlag 1
07:04:38  Parse Volume : worldVOL Phylvl -1
07:04:38  Process Volume : worldVOL importFlag1
07:04:38  solidref : WorldBox
07:04:38  Material G4_AIR
07:04:38  Set transparency
07:04:38  Process PhysVols importFlag 1
07:04:38  name: worldVOL parentpart = <Part object>
07:04:38  paramvol = None
07:04:38  ProcessVol returning <Part::PartFeature> GDMLBox_WorldBox
07:04:38  Process Surfaces
07:04:38  skinsurface
07:04:38  bordersurface
07:04:38  End processing GDML file
07:04:38  time : 0.7388 seconds
07:05:10  running with lxml.etree
07:05:10  Hide Material
07:05:46  Add SetMaterial
07:05:46  initUI
07:05:46  New getGroupedMaterials len GroupMaterials 5
07:05:46  process GEANT4 Materials : /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/./freecad/gdml/Resources/Geant4Materials.xml
07:05:46  Parse : /Users/keithsloan/Library/Application Support/FreeCAD/Mod/GDML/./freecad/gdml/Resources/Geant4Materials.xml
07:05:46  running with lxml.etree 
07:05:46  process new G4
07:05:46  Process Materials : G4Materials
07:05:46  doc.G4Materials <group object>
07:05:46  doc.Materials <group object>
07:05:46  Geant4
07:05:46  NIST
07:05:46  Element
07:05:46  HEP
07:05:46  Space
07:05:46  BioChemical
`

KeithSloan, 2024-05-31:
Again looking at the last error in your report

`
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\preProcessLoops.py", line 97, in preprocessLoops
    for loop in root.getroot().iterdescendants(tag="loop"):

'xml.etree.ElementTree.Element' object has no attribute 'getroot'
`

I am wondering if there is a bug in the Windows version of lxml

Would be good if you could confirm if your system correctly sets up the initial environment after
activate Workbench, new file, does it process the Default.gdml file in resources and create the various groups.

Please could you report your level of lxml

From FC in the Python console
`
>>> import lxml
>>> print(lxml.__version__)
4.9.3`

KeithSloan, 2024-05-31:
"I can import gmsh and lXML from the freecad command line"

Is lXML a typo? library should be lxml ( In lower case), maybe you are defaulting to xml rather than lxml.

Even if yes it would be a bug in that should not fail but if you are defaulting to xml as a result, it would help me track things down - Thanks

Gadajilehu, 2024-05-31:
> "I can import gmsh and lXML from the freecad command line"
>
> Is lXML a typo? library should be lxml ( In lower case), maybe you are defaulting to xml rather than lxml.
>
> Even if yes it would be a bug [...]

```shell
>>> import lxml
>>> print(lxml.__version__)
5.2.2
```
this is my version of lxml.

KeithSloan, 2024-05-31:
""Would be good if you could confirm if your system correctly sets up the initial environment after
activate Workbench, new file, does it process the Default.gdml file in resources and create the various groups."

Do you get errors if you just activate the workbench and start a new file.

Gadajilehu, 2024-05-31:
> 你在关注 https://github.com/KeithSloan/GDML/wiki/Model_Creation 吗？
> 
> 这就是
> 
> 激活 GDML Workbench 新文件
> 
> 应该给你一个看起来像 ![Image 31-05-2024 at 06 52](https://private-user-images.githubusercontent.com/2291247/335460734-5b99dda9-98db-46b8-bbb2-e0c2ac6d7ca3.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcxNDIyOTMsIm5iZiI6MTcxNzE0MTk5MywicGF0aCI6Ii8yMjkxMjQ3LzMzNTQ2MDczNC01Yjk5ZGRhOS05OGRiLTQ2YjgtYmJiMi1lMGMyYWM2ZDdjYTMuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MzFUMDc1MzEzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGI5MTA4NTZmMzcwOTZjYmU4YzBkMWI0MzZiOWYzYzA2MWY2YWIxOWZjNGI5YjI3OGNkZDRiYmNiMDM2OGUzZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.3WzMPiSqvLDu_5zXwqpGNBd68K2u3lD_ffYNGQVXTQU)
> 
> 从错误消息来看，您似乎正在使用椭圆命令，并且找不到由上述工作流程设置的材料组，并且它正在尝试纠正此问题，即 buildDefaultDoc 函数对象第 87 行，并且此恢复操作存在问题。
> 
> 我会对此进行调查，但如果你能确认一下，那就太好了。
> 
> 初始设置后的文件保存应如下所示[initialFile.FCStd.txt](https://github.com/user-attachments/files/15508982/initialFile.FCStd.txt)

I'm sure I have an interface that looks the same as yours. Because I added the gdml workbench directly from the add-on manager instead of gdml (another plugin).
[image]

Gadajilehu, 2024-05-31:
> ""Would be good if you could confirm if your system correctly sets up the initial environment after activate Workbench, new file, does it process the Default.gdml file in resources and create the various groups."
> 
> Do you get errors if you just activate the workbench and start a new file.

> ""Would be good if you could confirm if your system correctly sets up the initial environment after activate Workbench, new file, does it process the Default.gdml file in resources and create the various groups."
> 
> Do you get errors if you just activate the workbench and start a new file.

```
16:21:55  Activated
16:22:05  running with xml.etree.ElementTree
16:22:05  processGDML type 1
16:22:05  Print Verbose : False
16:22:05  Import GDML file : C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\Resources/Default.gdml
16:22:05  ImportGDML Version 1.9b
16:22:05  pathName C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\freecad\gdml\Resources
16:22:05  Parse : C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\Resources/Default.gdml
16:22:05  running with etree.ElementTree (import limitations)
16:22:05   for full import add lxml library 
16:22:05  <xml.etree.ElementTree.ElementTree object at 0x0000014AC9C828E0>
16:22:05  Process Positions
16:22:05  Traceback (most recent call last):
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\init_gui.py", line 73, in slotCreatedDocument
    processGDML(
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\importGDML.py", line 3365, in processGDML
    preProcessLoops.preprocessLoops(root)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\preProcessLoops.py", line 97, in preprocessLoops
    for loop in root.getroot().iterdescendants(tag="loop"):
<class 'AttributeError'>: 'xml.etree.ElementTree.Element' object has no attribute 'getroot'
```

KeithSloan, 2024-05-31:
Not sure why my Mac version of lxml 4.9.2 is so far behind, latest is 5.2.2 as you have.

I will try and upgrade to 5.2.2 to see if I can recreate
But maybe you could try an earlier version as a work around

Get current path to lxml in FC python console

**import lxml
print(lxml.__path__)**

from outside FC

pip3 install lxml="version" -t <path as noted from above>

I would try 4.9.3

KeithSloan, 2024-05-31:
I just tried 5.2.2 on my system and did not have a problem

`
keithsloan@Keiths-iMac-2 gdml % pip3 install lxml -t /Applications/FreeCAD_0.22.app/Contents/Resources/lib/python3.10/site-packages
Collecting lxml
  Downloading lxml-5.2.2-cp310-cp310-macosx_10_9_x86_64.whl.metadata (3.4 kB)
Downloading lxml-5.2.2-cp310-cp310-macosx_10_9_x86_64.whl (4.4 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.4/4.4 MB 5.0 MB/s eta 0:00:00
Installing collected packages: lxml
Successfully installed lxml-5.2.2
WARNING: Target directory /Applications/FreeCAD_0.22.app/Contents/Resources/lib/python3.10/site-packages/lxml already exists. Specify --upgrade to force replacement.
keithsloan@Keiths-iMac-2 gdml % pip3 install lxml -t /Applications/FreeCAD_0.22.app/Contents/Resources/lib/python3.10/site-packages --upgrade
Collecting lxml
  Using cached lxml-5.2.2-cp310-cp310-macosx_10_9_x86_64.whl.metadata (3.4 kB)
Using cached lxml-5.2.2-cp310-cp310-macosx_10_9_x86_64.whl (4.4 MB)
Installing collected packages: lxml
Successfully installed lxml-5.2.2
`

[image]

[Ellipsoid.FCStd.txt](https://github.com/user-attachments/files/15510781/Ellipsoid.FCStd.txt)

Gadajilehu, 2024-05-31:
```shell
>>> import lxml
>>> print(lxml.__version__)
4.9.2
```
Yes, I rolled back to 4.9.2 and the error did not change. I also checked the getroot method in the XML and found that it has not changed since Python 3.8.

Gadajilehu, 2024-05-31:
> I just tried 5.2.2 on my system and did not have a problem
> 
> `keithsloan@Keiths-iMac-2 gdml % pip3 install lxml -t /Applications/FreeCAD_0.22.app/Contents/Resources/lib/python3.10/site-packages Collecting lxml Downloading lxml-5.2.2-cp310-cp310-macosx_10_9_x86_64.whl.metadata (3.4 kB) Downloading lxml-5.2.2-cp310-cp310-macosx_10_9_x86_64.whl (4.4 MB) ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.4/4.4 MB 5.0 MB/s eta 0:00:00 Installing collected packages: lxml Successfully installed lxml-5.2.2 WARNING: Target directory /Applications/FreeCAD_0.22.app/Contents/Resources/lib/python3.10/site-packages/lxml already exists. Specify --upgrade to force replacement. keithsloan@Keiths-iMac-2 gdml % pip3 install lxml -t /Applications/FreeCAD_0.22.app/Contents/Resources/lib/python3.10/site-packages --upgrade Collecting lxml Using cached lxml-5.2.2-cp310-cp310-macosx_10_9_x86_64.whl.metadata (3.4 kB) Using cached lxml-5.2.2-cp310-cp310-macosx_10_9_x86_64.whl (4.4 MB) Installing collected packages: lxml Successfully installed lxml-5.2.2`
> 
> ![Image 31-05-2024 at 09 38](https://private-user-images.githubusercontent.com/2291247/335503895-effc3d59-45cc-45fc-ae0f-76ffd69ff587.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcxNDUwOTEsIm5iZiI6MTcxNzE0NDc5MSwicGF0aCI6Ii8yMjkxMjQ3LzMzNTUwMzg5NS1lZmZjM2Q1OS00NWNjLTQ1ZmMtYWUwZi03NmZmZDY5ZmY1ODcuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MzFUMDgzOTUxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDJlZTBhNTU2N2UxYWM3YTQxZDdiMjdjYTYwZDc2ZWRmNjRmZjZmMGQyZGQzMzEzNDhiY2FmNjkwMWE2NDJlYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.D3J1k1dYhOj17VAYrq1NCBHpYQQTtKN30zrBDCI_aKA)
> 
> [Ellipsoid.FCStd.txt](https://github.com/user-attachments/files/15510781/Ellipsoid.FCStd.txt)

The projects in my combo view are somewhat different from yours. My projects only have Constants, Variables, and Quantities.I'm trying to figure out if I made any foolish operational mistakes.

Gadajilehu, 2024-05-31:
> Thanks for report. view
> 
> 16:21:55 Activated 16:22:05 running with xml.etree.ElementTree 16:22:05 processGDML type 1 16:22:05 Print Verbose : False 16:22:05 Import GDML file : C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\Resources/Default.gdml 16:22:05 ImportGDML Version 1.9b 16:22:05 pathName C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\freecad\gdml\Resources 16:22:05 Parse : C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\Resources/Default.gdml 16:22:05 running with etree.ElementTree (import limitations) 16:22:05 for full import add lxml library 16:22:05 <xml.etree.ElementTree.ElementTree object at 0x0000014AC9C828E0> 16:22:05 Process Positions 16:22:05 Traceback (most recent call last): File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\init_gui.py", line 73, in slotCreatedDocument processGDML( File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\importGDML.py", line 3365, in processGDML preProcessLoops.preprocessLoops(root) File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\preProcessLoops.py", line 97, in preprocessLoops for loop in root.getroot().iterdescendants(tag="loop"): <class 'AttributeError'>: 'xml.etree.ElementTree.Element' object has no attribute 'getroot' This is the error reported by the report view when I only open the workbench and create a new file
> 
> The messages
> 
> 16:22:05 running with etree.ElementTree (import limitations) 16:22:05 for full import add lxml library
> 
> Mean it is not finding lxml and defaulting to xml and it is xml that is barfing <class 'AttributeError'>: 'xml.etree.ElementTree.Element' object has no attribute 'getroot'
> 
> Yet it appears you can use lxml from the FC console

Is it possible that there are some lXML dependency libraries that I did not install successfully? I will check this.Although I installed it using pip。

KeithSloan, 2024-05-31:
The messages and dropping to xml come from

`print("Parse : " + filename)
    try:
        from lxml import etree

        FreeCAD.Console.PrintMessage("running with lxml.etree \n")
        parser = etree.XMLParser(resolve_entities=True)
        root = etree.parse(filename, parser=parser)
        # print('error log')
        # print(parser.error_log)

    except ImportError:
        try:
            import xml.etree.ElementTree as etree

            FreeCAD.Console.PrintMessage(
                "running with etree.ElementTree (import limitations)\n"
            )
            FreeCAD.Console.PrintMessage(
                " for full import add lxml library \n"
            )
            
   so if you have lxml but it is failing to find xtree, that would explain a lot.

Gadajilehu, 2024-05-31:
> The messages and dropping to xml come from
> 
> ```
>     try:
>         from lxml import etree
> 
>         FreeCAD.Console.PrintMessage("running with lxml.etree \n")
>         parser = etree.XMLParser(resolve_entities=True)
>         root = etree.parse(filename, parser=parser)
>         # print('error log')
>         # print(parser.error_log)
> 
>     except ImportError:
>         try:
>             import xml.etree.ElementTree as etree
> 
>             FreeCAD.Console.PrintMessage(
>                 "running with etree.ElementTree (import limitations)\n"
>             )
>             FreeCAD.Console.PrintMessage(
>                 " for full import add lxml library \n"
>             )
> ```
> 
> so if you have lxml but it is failing to find xtree, that would explain a lot.

yes I think this is problem，I will all of dependency of lxml and sure I can import etree

Gadajilehu, 2024-06-04:
I have completed my installation and would like to share my installation steps. Firstly, regarding the installation process you mentioned in README.txt.
At first,when you use shell
```shell
D:\FreeCAD 0.20\bin\python -m pip install --target="D:\FreeCAD 0.20\bin\Lib\site-packages" --upgrade gmsh
D:\FreeCAD 0.20\bin\python -m pip install -i https://gmsh.info/python-packages --force-reinstall --no-cache-dir --upgrade --target="D:\FreeCAD 0.20\bin\Lib\site-packages" gmsh-dev
```

This is no problem, but when you go https://gmsh.info/bin/Windows/gmsh-4.10.5-Windows64-sdk.zip Download the package, the version of your pip download may not be consistent with the version you downloaded directly from this website, which will result in an import gmsh error.This is no problem, but when you go https://gmsh.info/bin/Windows/gmsh-4.10.5-Windows64-sdk.zip Download the package. The version of your pip download may not be consistent with the version you downloaded directly from this website, which may result in an import gmsh error. Therefore, you need to confirm your gmsh version and provide it in the https://gmsh.info/bin/Windows/ download.
For:
```shell
D:\FreeCAD 0.20\bin\python -m pip install --target="D:\FreeCAD 0.20\bin\Lib\site-packages" --upgrade lxml
```
After the successful installation of PIP, you need to check whether the import lXML from Etree and import lXML from HTML can succeed. If not, you can refer to my experience. I created a new virtual environment using Conda, and the Python version of this virtual environment is consistent with Freecad's Python. Then, I used PIP to install lXML in this environment and confirmed that there are no issues with the use of lXML in this environment. Then, I copied all the files in the site-package folder of this environment to Freecad's site-package.
This is all my experience of successfully installing on Windows。

Gadajilehu, 2024-06-04:
> The messages and dropping to xml come from
>
> ```
>     try:
>         from lxml import etree
>
>         FreeCAD.Console.PrintMessage("running with lxml.etree \n")
>         parser = [...]

I have completed my installation and would like to share my installation steps. Firstly, regarding the installation process you mentioned in README.txt.
At first,when you use shell
```shell
D:\FreeCAD 0.20\bin\python -m pip install --target="D:\FreeCAD 0.20\bin\Lib\site-packages" --upgrade gmsh
D:\FreeCAD 0.20\bin\python -m pip install -i https://gmsh.info/python-packages --force-reinstall --no-cache-dir --upgrade --target="D:\FreeCAD 0.20\bin\Lib\site-packages" gmsh-dev
```

This is no problem, but when you go https://gmsh.info/bin/Windows/gmsh-4.10.5-Windows64-sdk.zip Download the package, the version of your pip download may not be consistent with the version you downloaded directly from this website, which will result in an import gmsh error.This is no problem, but when you go https://gmsh.info/bin/Windows/gmsh-4.10.5-Windows64-sdk.zip Download the package. The version of your pip download may not be consistent with the version you downloaded directly from this website, which may result in an import gmsh error. Therefore, you need to confirm your gmsh version and provide it in the https://gmsh.info/bin/Windows/ download.
For:
```shell
D:\FreeCAD 0.20\bin\python -m pip install --target="D:\FreeCAD 0.20\bin\Lib\site-packages" --upgrade lxml
```
After the successful installation of PIP, you need to check whether the import lXML from Etree and import lXML from HTML can succeed. If not, you can refer to my experience. I created a new virtual environment using Conda, and the Python version of this virtual environment is consistent with Freecad's Python. Then, I used PIP to install lXML in this environment and confirmed that there are no issues with the use of lXML in this environment. Then, I copied all the files in the site-package folder of this environment to Freecad's site-package.
This is all my experience of successfully installing on Windows。

KeithSloan, 2024-06-05:
Thanks for your feedback.

Can I ask if you ever tried to install just by using the Addon Manager?

i.e. Rather than having to run pip commands, you tried  FreeCAD | Tools | Addon Manager

Select the GDML Workbench and responded to the install python dependencies button.

I would be very interested to know if the errors you encounter also occurred using the Addon Manager

Gadajilehu, 2024-06-06:
> Thanks for your feedback.
>
> Can I ask if you ever tried to install just by using the Addon Manager?
>
> i.e. Rather than having to run pip commands, you tried FreeCAD | Tools | Addon Manager
>
> Select the [...]

I use new computer test the Addon Manager install dependencies.It can open workbench sucessfully,but i think it need to to add the missed "gmsh.exe", "gmsh-4.10.dll" and update "gmsh.py" .
And i have another problem that when I just open the workbench and create a geometry without doing anything, my report view will report this error.
```shell
10:29:22  time : 0.3697 seconds
10:44:10  PropertyPythonObject::toString(): failed for <class 'freecad.gdml.GDMLObjects.GDMLconstant'>
10:44:10  Traceback (most recent call last):
  File "E:\software\freecad\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "E:\software\freecad\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "E:\software\freecad\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "E:\software\freecad\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
10:44:10  PropertyPythonObject::toString(): failed for <class 'freecad.gdml.GDMLObjects.GDMLconstant'>
10:44:10  Traceback (most recent call last):
  File "E:\software\freecad\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "E:\software\freecad\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
```
you can see 10:29:22 is my last operation and i do nothing until 10：44 report these error。

KeithSloan, 2024-06-06:
Geant4 knows of all the the predefined (e.g. NIST) materials.

The VALIDATION ERROR is because of the Schema checking and can be ignored unless in your Geant4 installation
you did not take the option for the predefined NIST materials.

KeithSloan, 2024-06-07:
If you need the material definitions for software other than Geant4, then there is a configuration setting to have G4 Material definitions exported.
[image]

Test file exported with option set.
[Unnamed-worldVOLA.gdml.txt](https://github.com/user-attachments/files/15720351/Unnamed-worldVOLA.gdml.txt)

Gadajilehu, 2024-06-07:
> If you need the material definitions for software other than Geant4, then there is a configuration setting to have G4 Material definitions exported. ![Image 07-06-2024 at 08 02](https://private-user-images.githubusercontent.com/2291247/337559618-37fae5bb-96f4-4b49-91d1-8f17b2d16c56.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc3NDQzODcsIm5iZiI6MTcxNzc0NDA4NywicGF0aCI6Ii8yMjkxMjQ3LzMzNzU1OTYxOC0zN2ZhZTViYi05NmY0LTRiNDktOTFkMS04ZjE3YjJkMTZjNTYuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDdUMDcwODA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Mjg2MjdlYTdjMzFmOGQyZjRmZmU4MTFlZDMwNjBlOGNmODViZTZkMDY5Y2VlNGE4ODZhMjNhNDMzZjEwYjQ5MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.1CvVyP-qW4Vyl7lVyn8FhiTgUIj3P9jalKMkCfkynG4)
> 
> Test file exported with option set. [Unnamed-worldVOLA.gdml.txt](https://github.com/user-attachments/files/15720351/Unnamed-worldVOLA.gdml.txt)

Yes, thank you for your reply. I have successfully exported the first gdml file and read it out in geant4。I am currently trying to import STP format files into Freecad and convert them using Tessellate via Gmsh.
```shell
15:12:46  Running the Python command 'TessellateCommand' failed:
Traceback (most recent call last):
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\GDMLCommands.py", line 2341, in Activated
    vol = createPartVol(obj)

createPartVol() missing 1 required positional argument: 'doc'
```
Is this related to the complexity of my model? This is a file containing sixteen clover type HPGe detectors.
Alternatively, I should try the GMSH method, but operating it directly on the toolbar would result in an error. I have read that the error may be due to an issue with the Numpy version, and I am currently trying to resolve it.Thank you for your reply in the past few days. Your help in GDML conversion has even surpassed that of my supervisor.

KeithSloan, 2024-06-07:
> Yes, thank you for your reply. I have successfully exported the first gdml file and read it out in geant4。I am currently trying to import STP format files into Freecad and convert them using [...]

Sorry that is a bug that has crept in.

createPartVol should have had a default of FreeCAD.ActiveDoument i.e.
def createPartVol(obj, doc=FreeCAD.ActiveDocument, selection=False)

Please try the latest version.

def createPartVol(obj, doc=FreeCAD.ActiveDocument, selection=False)

KeithSloan, 2024-06-07:
>  This is a file containing sixteen clover type HPGe detectors. 

Have you considered using an Arrays with GDMLObjects? see [https://github.com/KeithSloan/GDML/wiki#arrays-of-objects](url)

Arrays are exported as MultiUnions and if you can avoid the need to Tessellate will be much faster to process.

Arrays

KeithSloan, 2024-06-07:
> > Thanks for your feedback.
> > Can I ask if you ever tried to install just by using the Addon Manager?
> > i.e. Rather than having to run pip commands, you tried FreeCAD | Tools | Addon Manager
> > Select the GDML Workbench and responded to the install python dependencies button.
> > I would be very interested to know if the errors you encounter also occurred using the Addon Manager
> 
> I use new computer test the Addon Manager install dependencies.It can open workbench sucessfully,It means lxml install is sucessfully, but i think it need to to add the missed "gmsh.exe", "gmsh-4.10.dll" and update "gmsh.py" .Because there will be a prompt in the report view that gmsh-4.10.dll cannot be found. And i have another problem that when I just open the workbench and create a geometry without doing anything, my report view will report this error.
> 
> ```shell
> 10:29:22  time : 0.3697 seconds
> 10:44:10  PropertyPythonObject::toString(): failed for <class 'freecad.gdml.GDMLObjects.GDMLconstant'>
> 10:44:10  Traceback (most recent call last):
>   File "E:\software\freecad\bin\lib\json\__init__.py", line 231, in dumps
>     return _default_encoder.encode(obj)
>   File "E:\software\freecad\bin\lib\json\encoder.py", line 199, in encode
>     chunks = self.iterencode(o, _one_shot=True)
>   File "E:\software\freecad\bin\lib\json\encoder.py", line 257, in iterencode
>     return _iterencode(o, 0)
>   File "E:\software\freecad\bin\lib\json\encoder.py", line 179, in default
>     raise TypeError(f'Object of type {o.__class__.__name__} '
> <class 'TypeError'>: Object of type FeaturePython is not JSON serializable
> 10:44:10  PropertyPythonObject::toString(): failed for <class 'freecad.gdml.GDMLObjects.GDMLconstant'>
> 10:44:10  Traceback (most recent call last):
>   File "E:\software\freecad\bin\lib\json\__init__.py", line 231, in dumps
>     return _default_encoder.encode(obj)
>   File "E:\software\freecad\bin\lib\json\encoder.py", line 199, in encode
>     chunks = self.iterencode(o, _one_shot=True)
> ```
> 
> you can see 10:29:22 is my last operation and i do nothing until 10：44 report these error。

The workbench uses Part::FeaturePython for GDML Objects and these get saved and loaded with JSON
There must be a bug but I would need a sample file to track down the bug.

Gadajilehu, 2024-06-12:
> The workbench uses Part::FeaturePython for GDML Objects and these get saved and loaded with JSON There must be a bug but I would need a sample file to track down the bug.

Sorry, there have been experiments these days and no response has been received.I am unable to upload my project because the maximum size is limited to 26MB.If you want to use it for testing, please give me an email.
And there will be such an error when I export my gdml file for this file.
```shell
09:09:36  Traceback (most recent call last):
  File "<string>", line 8, in <module>
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 3052, in export
    exportGDMLworld(first, filepath, fileExt)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2818, in exportGDMLworld
    exportGDML(first, filepath, fileExt)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2752, in exportGDML
    exportWorldVol(first, fileExt)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2679, in exportWorldVol
    processVolAssem(vol, xmlParent, WorldVOL)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2343, in processVolAssem
    processVolume(vol, xmlParent, psPlacement, isPhysVol, volName=None)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2202, in processVolume
    solidExporter = SolidExporter.getExporter(topObject)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 3193, in getExporter
    typeId = obj.Proxy.Type
<class 'AttributeError'>: 'GDMLTessellated' object has no attribute 'Type'
```

KeithSloan, 2024-06-12:
My email address is at the bottom of the README, just replace [at] with @ and [dot] with .

KeithSloan, 2024-06-12:
i have another problem that when I just open the workbench and create a geometry without doing anything, my report view will report this error.
> ```shell
> 10:29:22  time : 0.3697 seconds
> 10:44:10  PropertyPythonObject::toString(): failed for <class 'freecad.gdml.GDMLObjects.GDMLconstant'>
> 10:44:10  Traceback (most recent call last):
>   File [...]

Not sure why that is happening, somehow one of the constant values is being corrupted,

I you open a new file in the Workbench the first Group should be "Constants"

If you click on Constants you should see

[image]

And if you click on the contants you should see
[image]
[image]
[image]

KeithSloan, 2024-06-12:
09:09:36  Traceback (most recent call last):
  File "<string>", line 8, in <module>
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 3052, in export
    exportGDMLworld(first, filepath, fileExt)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2818, in exportGDMLworld
    exportGDML(first, filepath, fileExt)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2752, in exportGDML
    exportWorldVol(first, fileExt)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2679, in exportWorldVol
    processVolAssem(vol, xmlParent, WorldVOL)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2343, in processVolAssem
    processVolume(vol, xmlParent, psPlacement, isPhysVol, volName=None)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2202, in processVolume
    solidExporter = SolidExporter.getExporter(topObject)
  File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 3193, in getExporter
    typeId = obj.Proxy.Type
<class 'AttributeError'>: 'GDMLTessellated' object has no attribute 'Type'

Found a bug in that Type was not being set if the Gui was not up, could it have been created using the command line?
Now fixed in latest version

if you email me the file I can fix.

KeithSloan, 2024-06-12:
What version of FreeCAD are you using? There are problems with files created with FreeCAD 0.21.2

Gadajilehu, 2024-06-12:
> What version of FreeCAD are you using? There are problems with files created with FreeCAD 0.21.2

version 0.21.2

Gadajilehu, 2024-06-12:
> 09:09:36 Traceback (most recent call last): File "", line 8, in File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\exportGDML.py", line 3052, in export exportGDMLworld(first, filepath, fileExt) File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\exportGDML.py", line 2818, in exportGDMLworld exportGDML(first, filepath, fileExt) File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\exportGDML.py", line 2752, in exportGDML exportWorldVol(first, fileExt) File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\exportGDML.py", line 2679, in exportWorldVol processVolAssem(vol, xmlParent, WorldVOL) File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\exportGDML.py", line 2343, in processVolAssem processVolume(vol, xmlParent, psPlacement, isPhysVol, volName=None) File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\exportGDML.py", line 2202, in processVolume solidExporter = SolidExporter.getExporter(topObject) File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\GDML.\freecad\gdml\exportGDML.py", line 3193, in getExporter typeId = obj.Proxy.Type <class 'AttributeError'>: 'GDMLTessellated' object has no attribute 'Type'
> 
> Found a bug in that Type was not being set if the Gui was not up, could it have been created using the command line? Now fixed in latest version
> 
> if you email me the file I can fix.

ok,i will try later and feedback, now I have other work to do.