KeithSloan/GDML #146: Error for GDML in FreeCad ver 0.21.1

jjonathan2020, 2024-10-02:
Hi. I am an undergraduate student using this tool to build detectors for export to Geant4. I am using Windows and followed the instruction guides outlined in the github but I ran into an issue. 

in <module>
    def edgelistArea(edgelist: list[Part.Edge]) -> float:

'type' object is not subscriptable

I cannot create an object using the gdml workbench nor can I export step files to gdml. 

I looked up the error and there seems that the version of python is not up to date (needs 3.9). Is there something I am missing here, or if I missed something in the installation process. Kindly let me know as soon as possible and if you need more information or I did not explain it clearly please let me know.

KeithSloan, 2024-10-03:
Sorry about this, please could you edit the code to not use the type reference

I.e no reference to the variable type

def edgelistArea(

You are correct Variable types needs 3.9

KeithSloan, 2024-10-03:
Closed in error

KeithSloan, 2024-10-03:
You could try one of the FC 1.0 Release Candidates, I have yet to check RC 2.0 but RC 1.0 seems fine on my Mac.

a-sunkari, 2024-10-05:
Can confirm same issue; switching to FC 1.0 RC 1.0 solved this problem!

KeithSloan, 2024-10-05:
A-sunkari - Thanks

a-sunkari, 2024-10-07:
Wanted to also add to change this line:

`def alreadyExported(shape) -> str | None:`

Looks like Python 3.8 in the 0.21 interpreter in FreeCAD doesn't like the `|` syntax.

I fixed the issue by adding

`from typing import Union`

and changing the alreadyExported function to

`def alreadyExported(shape) -> Union[str, None]:`

in addition to the changes Mr. Sloan has suggested above

jjonathan2020, 2024-10-07:
Thank you guys so much! I switched to the FC RC 1.0 and I did not have any issues.

closed 2024-10-07 by jjonathan2020