KeithSloan/GDML #147: Error for GDML export in FreeCAD 1.0 (RC 1.0 and 2.0)

a-sunkari, 2024-10-06:
OS: Windows 11
FreeCAD Version: 1.0 (In both RC 1.0 and 2.0)

I recently started using the GDML workbench in FreeCAD 1.0 due to an [issue in version 0.21 where the tesselate function did not work properly](https://github.com/KeithSloan/GDML/issues/146).

The GDML workbench worked fine until today, where I am having the following error after I attempt to export the worldVOL to GDML:

```
12:38:27  pyException: Traceback (most recent call last):
  File "<string>", line 8, in <module>
  File "C:\Users\appin\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 3101, in export
    exportGDMLworld(first, filepath, fileExt)
  File "C:\Users\appin\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2869, in exportGDMLworld
    exportGDML(first, filepath, fileExt)
  File "C:\Users\appin\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2792, in exportGDML
    exportWorldVol(first, fileExt)
  File "C:\Users\appin\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2720, in exportWorldVol
    processVolAssem(vol, xmlParent, WorldVOL)
  File "C:\Users\appin\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2328, in processVolAssem
    if isContainer(vol):
       ^^^^^^^^^^^^^^^^
  File "C:\Users\appin\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2489, in isContainer
    if not isAssembly(obj):
           ^^^^^^^^^^^^^^^
  File "C:\Users\appin\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 2538, in isAssembly
    for ob in childObjects[obj]:
              ~~~~~~~~~~~~^^^^^
<class 'KeyError'>: (<Part object>,)
```

It looks like <Part object> is not found in childObjects.

To double check, I opened a FreeCAD document I was able to successfully export yesterday and tried exporting it today, and I got the same error. I also attempted using both RC 1.0 and 2.0, but the same issue presented itself in both versions.

Let me know if more information is needed, or if there is anything I can do to change the code to get the export function working!

KeithSloan, 2024-10-07:
Looking further

KeithSloan, 2024-10-07:
Obviously there is a bug of some sort but looks like something has gone wrong with the structure, shoul look like, check combo view as per

[image]

mhindi2, 2024-10-07:
Is it possible to email me the file, or at least the part that is causing the failure? I have fixed several issues recently, but I will have to wait for Keith to get back from vacation to merge the fixes.

a-sunkari, 2024-10-07:
Thank you all for your quick responses!

@KeithSloan yes, I made sure the file followed the right "structure" for the world volume. This was a .step geometry that I used the object-to-tesselate function on. I've attached a screenshot of it, as well as the FreeCAD file, to this comment.

[image]

Interestingly, I was able to export this file to GDML a few days ago with no problem (as commented in the other post) but couldn't do the next day after! I don't know why that would be, because as far as I know, there hadn't been any changes to the GDML workbench or my FreeCAD installation. I've attached that GDML file to this comment as well.

@mhindi2 yes, the FreeCAD file is attached to this comment. Attached is also the GDML I was able to successfully export a few days ago. No changes have been made to the FreeCAD file since then.

[FreeCAD Files.zip](https://github.com/user-attachments/files/17281370/FreeCAD.Files.zip)

Thank you all again for your help!

mhindi2, 2024-10-07:
@a-sunkari I responded by email a short time ago, but it seems the response did not make it to this page. So I will paste what I wrote:

Thanks for giving me a copy of the file. I am responding to this via email; not sure what will show up on the github page.

On my system (FreeCAD 0.21.1) and with my current version of the workbench I don't get the error you reported when I export the file to gdml. See attached screenshot. I can send you the version of the exportGDML.py file that I cam currently using, but this is probably out of step of what's on the github and still has some bugs that I already fixed in a new branch, but I am not ready yet to release the new branch.

A question for you, though. All the tessellated shapes (converted from step) seem to be tubes. Is there any reason why you don't build them as tubes instead of importing them as step files. Simulations would run much faster with tubes than with tessellations.

[image]

a-sunkari, 2024-10-07:
Thanks for the review!

So I went back to 0.21.2 and it seems to export fine, although there are some changes to be made to the repo. One, as mentioned in the [other thread](https://github.com/KeithSloan/GDML/issues/146) is to change the edgelistArea function to match the Python 3.8 interpreter in FreeCAD 0.21. The other error I got:

`"C:\Users\appin\AppData\Roaming\FreeCAD\Mod\GDML\.\freecad\gdml\exportGDML.py", line 6244, in AutoTessellateExporter
    def alreadyExported(shape) -> str | None:
<class 'TypeError'>: unsupported operand type(s) for |: 'type' and 'NoneType'`

was also a Python version mismatch; I solved it with adding

`from typing import Union`

and changing the alreadyExported function to

`def alreadyExported(shape) -> Union[str, None]:`

to get around the inability to use the `|` syntax in that version of Python. I will add this to the other thread as well.

I _think_ that the reason for the export error I have been having in FreeCAD 1.0 is due to the fact that it uses **Python 3.11** in the interpreter rather than 3.8; perhaps that causes the problem? I will investigate and update if I find anything.

@mhindi2 thanks for the feedback about using tubes versus tessellated shapes! I plan on using more complex geometry in the future, but I just wanted to test a basic geometry to make sure the .step -> tessellation -> GDML conversion works well.

mhindi2, 2024-10-07:
Thanks for working on this! As far as I know,  the type annotation is
something that the python interpreter itself should ignore. But it seems in
Windows the python interpreter complains about types it does not
understand. You could easily just remove the types altogether.

Keith has been looking into issues with FC 1.0; I have not tested that
version myself yet. When Keith returns we will address those issues. Thanks
for sticking with us and don't hesitate to bring up any issues you might
find!

KeithSloan, 2024-10-15:
Okay back from vacation. I just picked up the version posted above and had no problem on export or loading into geant4
[image]

robinsheehy, 2024-11-27:
Hello,

Just a quick note to say that I am seeing this exact error in FreeCAD 1.0.0. I also could export the day before, but cannot now. The export also fails on the simplest design I could make, a single box GDMLBox inside the worldVOL.

KeithSloan, 2024-11-27:
I just downloaded the problem file in this thread and tried with FC1.0.0 on my system and exported okay.
Please could you post your FreeCAD info.

I also attach a zip of the files concerned
[FreeCAD Files 2.zip](https://github.com/user-attachments/files/17930430/FreeCAD.Files.2.zip)

My FreeCAD info is

OS: macOS 14.4
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ImportNURBS 1.1.0
  * ImportExport_3DM 0.1.0
  * DynamicData 2.69.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.0
  * fasteners 0.5.30
  * Design456 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * SWPart
  * toSketch 1.0.1
  * Curves 0.6.51

KeithSloan, 2024-11-27:
Have had one user on Windows that reported that, they had to save and restart their system to fix an export problem, that suggests to me that there is a memory bug somewhere.

Given that you could export the day before maybe you are seeing a similar problem.

KeithSloan, 2024-11-27:
Okay I just updated the package.xml file in the repro to say 2.0.1 Beta
so if you go to tools | addon Manager 
Refesh cache

You should see the latest version is 2.0.1 Beta.

Try uninstall | restart FC | install

Ignore install of python gmsh ( There is an issue with FC 1.0.0 and the gmsh python lib ) All non Gmsh facilities should be fine.

If you then FreeCAD | About 
Copy info an paste you should see something like
OS: macOS 14.4
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ImportNURBS 1.1.0
  * ImportExport_3DM 0.1.0
  * DynamicData 2.70.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.1Beta
  * fasteners 0.5.32
  * Design456 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * SWPart
  * toSketch 1.0.1
  * Curves 0.6.51

If you see GDML 2.0.1 Beta
Then this will check you have the latest version

(For some reason the addon manager does not pick up on the change)
It says 2.0.0 is latest but after install says installed 2.0.1
[image]

`
<package format="1" xmlns="https://wiki.freecad.org/Package_Metadata">
  <name>GDML workbench</name>
  <description>An external workbench for creating GDML models for Geant4 and Root</description>
  <version>2.0.1 Beta</version>
  <date>2024-27-11</date>
  <maintainer email="keith@sloan-home.co.uk">Keith Sloan</maintainer>
  <license file="LICENSE">LGPL-2.1</license>
  <url type="repository" branch="Main">https://github.com/KeithSloan/GDML</url>
  <url type="documentation">https://github.com/KeithSloan/GDML/wiki</url>
  <url type="readme">https://github.com/KeithSloan/GDML/blob/Main/README.md</url>
`

KeithSloan, 2024-11-27:
For how to successfully install the gmsh python library see [https://github.com/KeithSloan/GDML/issues/154](url)

KeithSloan, 2024-11-27:
We hope to shortly update the workbench but really need to do more testing - before doing so.

branch MainMHexport is the current Main repro with the new version of exportGDML if you wish to give it a try
to see if it resolves the issue.

KeithSloan, 2024-12-02:
> Thank you all for your quick responses!
> 
> @KeithSloan yes, I made sure the file followed the right "structure" for the world volume. This was a .step geometry that I used the object-to-tesselate function on. I've attached a screenshot of it, as well as the FreeCAD file, to this comment.
> 
> ![image](https://private-user-images.githubusercontent.com/57467156/374243268-fa2fbf5f-c675-430d-91ef-49ec4236dfa4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxNzA0MjksIm5iZiI6MTczMzE3MDEyOSwicGF0aCI6Ii81NzQ2NzE1Ni8zNzQyNDMyNjgtZmEyZmJmNWYtYzY3NS00MzBkLTkxZWYtNDllYzQyMzZkZmE0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDIwMDg0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMyYmEwNjNlZWMzMWI1ZjFiZTc0NWI0NzdiYzc5ODU4NDBmY2NhZjMxN2M3YjAwY2I4YjM3Yjk4YzZjMzUzOWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.zc0Kozq5ShdkolCox_cpZwDjiKEiV6LfRW8mo-b5Jv0)
> 
> Interestingly, I was able to export this file to GDML a few days ago with no problem (as commented in the other post) but couldn't do so the day after! I don't know why that would be, because as far as I know, there hadn't been any changes to the GDML workbench or my FreeCAD installation. I've attached that GDML file to this comment as well.
> 
> @mhindi2 yes, the FreeCAD file is attached to this comment. Attached is also the GDML I was able to successfully export a few days ago. No changes have been made to the FreeCAD file since then.
> 
> [FreeCAD Files.zip](https://github.com/user-attachments/files/17281370/FreeCAD.Files.zip)
> 
> Thank you all again for your help!

May I ask why you rotated the WorldVol 90 degrees?

This make things very difficult to create valid GDML for its contents.

KeithSloan, 2024-12-04:
Thanks - If you can provide the original file un-rotated some time, we can look the issue.

a-sunkari, 2024-12-17:
Here's a test file I've been working on that doesn't export for me in 1.0 - but works fine on previous versions, with the small changes to the code I've made. The world volume is not rotated in this file. Let me know if any more info is needed!

File provided as link because GitHub doesn't support the uploading of FreeCAD files.

[](https://file.io/eAg5tzP2FQCM)

mhindi2, 2024-12-18:
When I try access the file I get a message that says it has been deleted. Could you  please repost, or else email me directly. Thanks.

zhangcaocao, 2025-01-18:
I seem to have solved this problem by modifying the function def buildDocTree():
You can refer to: https://github.com/KeithSloan/GDML/issues/158#issuecomment-2599593615

adastra-orion, 2025-02-19:
Im now getting familiar with FreeCAD and the GDML workbench, has this function mod been verified to resolve export issues in Win or Linux yet? I unfortunately was having the same issues with Part Obj , in both OS. If so, can this function modification be done in FreeCAD itself?

mhindi2, 2025-02-19:
Could you please send me the file you are having problems with. On my Linux
machine the export works fine and on Windows it works with the modification
suggested in one of the responses to #147.

Munther

adastra-orion, 2025-02-24:
I am using just the simple feature function present in the workbench, generating a cube, and attempting to export the worldvol.

adastra-orion, 2025-02-24:
[image]

mhindi2, 2025-02-24:
Could yopu please send me or show me the report view (under View->Panels->Report View). You can save that as a text file. Also, please send the information shown under Help->About FreeCAD.

Thanks,
Munther

adastra-orion, 2025-02-25:
[FreeCAD about.txt](https://github.com/user-attachments/files/18951720/FreeCAD.about.txt)
[FreeCAD report.txt](https://github.com/user-attachments/files/18951719/FreeCAD.report.txt)

mhindi2, 2025-02-25:
Got it, thank you! I will check out the branch you are using and and see if I get the same error. I know that I have fixed this a while ago, and I thought the fix should be in the Main branch now, but I will compare against what I have.

KeithSloan, 2025-02-25:
If you look at commits for the Main Branch it is basically MHPull155-Merge and fix for FC1.0.

Where MHPull155-Merge is Munther Pull requested, manually merged.

Of course there is the possibility that a fix was missed on the manual merge.

adastra-orion, 2025-02-26:
Let me know how I can try the fix

KeithSloan, 2025-02-26:
I am a bit stuck in that I have a problem with my main machine.

I do have a MacBook Pro 2015 but this will not update passed Monterey 12.7.6
I went to the Apple store planning to trade in but a new machine would mean Apple Silicon M3 and the latest Mac OS don't support OpenGL which is required by FreeCAD.

I have created a branch Zhang on the main repro which has the latest MH version of exportGDML.py with Zhang Fix but I currently have NO means of testing.

We have a test environment see https://github.com/KeithSloan/FreeCAD_GDML_Test_Collection

And it would be good to at least test with the FC macro GDMLExportDirectory.FCMacro

mhindi2, 2025-02-26:
My idea was to let @adastra-orion test my branch as is, without first merging with your main branch.

@adastra-orion: if you want to test my branch without having to wait for Keith to merge with the official branch here is what you can do: Remove the official GDML branch using the Addon manger tool in FreeCAD. Download my branch from
[https://github.com/mhindi2/GDML/tree/zhang_buildDocTree_fix](url) (you might have to add .git at the end of the url, I am not sure) then copy the downloaded files to a directory GDML under yourpathto/FreeCAD/Mod

KeithSloan, 2025-02-26:
Branch Zhang2 should be existing Main Branch with buildDocTree updated in exportGDML.py

adastra-orion, 2025-02-27:
Issue fixed, thank you @mhindi2 and @KeithSloan !!

adastra-orion, 2025-02-27:
Thank you @zhangcaocao !

mhindi2, 2025-02-27:
> Issue fixed, thank you [@mhindi2](https://github.com/mhindi2) and [@KeithSloan](https://github.com/KeithSloan) !!

Excellent! Thank you for your patience!

closed 2025-05-09 by KeithSloan