KeithSloan/GDML #148: Could not export old FreeCAD project...

porosev, 2024-10-31:
Hello all,

I decided to memorize how to create GDML file from FreeCAD project and tried to export the old trivial project, but it failed with message "GDMLBox" has no the attribute "Type". Something changed in the latest version of GDML Workbench???

In attachment I have put the screenshot of the message and my stupid project.

Sincerely Yours,
Vyacheslav

[image]
[TelescopeIdeal.zip](https://github.com/user-attachments/files/17589794/TelescopeIdeal.zip)

KeithSloan, 2024-10-31:
Was the project created with FC 0.20.2?

There is a regression in FC 0.20.2 where for PartFeaturePython Objects the Type is no longer stored in Proxy, i,e, Proxy.Type but in just Type.

If you load your docment into say FC 1.0 RC2 it will prompt thatthe doc needs a recompute and for me exports okay.
[TelescopeIdea.zip](https://github.com/user-attachments/files/17590133/TelescopeIdea.zip)

porosev, 2024-11-01:
WIth FC 1.0 RC2 I got another error :((
[image]

KeithSloan, 2024-11-01:
Please zip or add txt to FCstd file and post, we will try an resolve.

KeithSloan, 2024-11-01:
Is this the same file as TelescopeIdea?

The zip file I posted, I loaded your file into FC 1.0 RC and it prompted for recompute and the file in my zip is the recomputed one. What happens if you try it.

When you opended you file with FC 1.0 RC 2, did it suggest a recompute?

I am on Mac 

OS: macOS 14.4
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * Pyramids-and-Polyhedrons
  * GDML 2.0.0
  * fasteners 0.5.29
   * toSketch 1.0.1

I assume you are on Windows

KeithSloan, 2024-11-01:
Not that it helps your particular situation but here is the log of loadng the export gdml file into geant4 and screen shot.

Maybe check you have the latest version of the workbench

[geantOut.txt](https://github.com/user-attachments/files/17599229/geantOut.txt)
[image]

porosev, 2024-11-01:
Yes, it is the same project as I uploaded at the beginning... 
I have tried to export my project under Ubuntu with FC 1.0 RC2 as under Windows but it reports the same problem.
and, Yes, it prompted for recomputation....

So, You are lucky under MacOS!

[image]

KeithSloan, 2024-11-01:
Check you version of exportGDML.py with this one
[exportGDML.py.txt](https://github.com/user-attachments/files/17602154/exportGDML.py.txt)

porosev, 2024-11-02:
I am sorry, but a result is exactly the same.

[image]
[image]

KeithSloan, 2024-11-02:
What is the difference between screenshot_1.jpg and screenshot_2.jpg?

screenshot_2.jpg - looks correct, have you checked the file created D:/1/TelescopeIdea-world.gdml

Your screenshots are of the python console  and the second one shows no errors and an export to the file.

porosev, 2024-11-02:
I just switched on the GDML option : "Print debug information in the Console" to check if something new will
be show in the Python console... but nothing interesting happened

and I see the same errors as with my old project as with your converted copy in TelescopoIdea.zip

KeithSloan, 2024-11-02:
Okay stab in the dark - Do you have the Export utilized Geant4 Matrials set ?

If yes, try without.

KeithSloan, 2024-11-02:
Are you able to try a branch? 

If yes try branch symmetric - we know it still has bugs which we are in the process of addressing but you may do better with it.

porosev, 2024-11-02:
I have checked few versions and I have found that the old version "cutTube" can export my stupid project without errors... but the all newer ones failed.

[image]

Is it OK that gmsh is shown in dependencies?
[image]

KeithSloan, 2024-11-02:
If you wish to use the Gmsh facilities in the workbench - Gmsh and Gmsh Min then the workbench needs the gmsh python library, you also have to make the gmsh shared library available to FreeCAD see details in the README

The Gmsh facilities in workbench use facilities available via the python library which is different from the API parts of Gmsh that Mesh and FEM use.

KeithSloan, 2024-11-05:
Another user reported a very similar problem and they resolved it by restarting their PC.
You might try restarting FreeCAD or thw workbench.

porosev, 2024-11-29:
Just few comments about this problem...

It is funny, but sometimes the export works! 
For example, I open a project and try to export... but I see the error. After that I make some clicks on geometry, object tree and try to make export again... and it work! After that I can open another project and I could export it without problems too.
But it works until I exit FC... 

Unfortunately, I could not find the magic sequence of action to get 100% reproduceable result yet.

At present moment I installed official release of FC1.0 and the latest GDML Workbench.
On my private opinion..it looks like some variables are uninitialized, and for different compilers we see different results. For MacOS its OK, but for Windows we had troubles.

porosev, 2024-11-29:
One way to make export working... 
1. Open the project.
2. Right mouse click on object and select Material in drop-down menu.
[image]
just close it pressing "Close" button
[image]
3. Try to make export... it should work

KeithSloan, 2024-11-29:
Munther has been doing a lot of work on exportGDML.py One of the problems fixed is duplicate SensDet.

A few days ago I created a branch MainMHexport which has the fix,

If. you are unable to git fetch origin
and git checkout MainMHexport

You can download exportGDML.py and GDMLShared.py from https://github.com/KeithSloan/GDML/tree/MainMHexport
and replace your versions.

Munther has been taking a well earned break to enjoy Thanksgiving with his family, we hope to do a lot more testing and merge a number of updates shortly.

KeithSloan, 2024-11-30:
Not sure I understand why Gui.runCommand('Std_SetMaterial',0)
would make a difference.

Std_SetMaterial is FreeCAD's setting of the Material.

The GDML workbench was started back at the time when FreeCAD was version 0.13.
At the time the only support for Materials was for the FEM and Arch Workbenches.

They introduced a new Materials system a short while back. At the time they were doing so we put in the requirements that we would need but unfortunately they were not addressed. 

Materials for FreeCAD in general are very different from GDML Materials which have a lot of Nuclear requirements rather than engineering.

It is unfortunate that there are FreeCAD native way of setting the Material and with GDML you have to either use the Material in GDMLObject properties or the GDLML workbench toolbar - SetMaterial.

Just wondering if you have tried to use the FC way to set the Material and this has caused the need for 'Std_SetMaterial',0)