KeithSloan/GDML #158: Problems with export using Windows 11

KeithSloan, 2024-12-24:
There is a bug under Qt and Windows

The BuildTree function uses
```
# Get world volume from document tree widget
worldObj = FreeCADGui.Selection.getSelection()[0]
tree = FreeCADGui.getMainWindow().findChildren(QtGui.QTreeWidget)[0]
it = QtGui.QTreeWidgetItemIterator(tree)
```

Which returns correct values under Linux and MacOS  but under Windows Returns None

zhangcaocao, 2025-01-18:
Hi @KeithSloan 

### I encountered the same problem: https://github.com/KeithSloan/GDML/issues/147#issue-2568794335

### The version I use is：

```
OS: macOS 14.3
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * GDML.zip
  * GDML 2.0.1Beta
```

### I made a modification to the "def buildDocTree():" function, here is my code:

```
def buildDocTree():
    from PySide import QtWidgets

    global childObjects
    childObjects = {}  # dictionary of list of child objects for each object
    # TypeIds that should not go in to the tree
    skippedTypes = ["App::Origin", "Sketcher::SketchObject", "Part::Compound"]

    def addDaughters(item: QtWidgets.QTreeWidgetItem):
        print (f"--------addDaughters {item.text(0)}")
        objectLabel = item.text(0)
        object = App.ActiveDocument.getObjectsByLabel(objectLabel)[0]
        if object not in childObjects:
            childObjects[object] = []
        for i in range(item.childCount()):
            childItem = item.child(i)
            treeLabel = childItem.text(0)
            try:
                childObject = App.ActiveDocument.getObjectsByLabel(treeLabel)[0]
                objType = childObject.TypeId
                if objType not in skippedTypes:
                    childObjects[object].append(childObject)
                    addDaughters(childItem)
            except Exception as e:
                print(e)
        return

    # Get world volume from document tree widget
    worldObj = FreeCADGui.Selection.getSelection()[0]
    # tree = FreeCADGui.getMainWindow().findChildren(QtGui.QTreeWidget)[0]
    # it = QtGui.QTreeWidgetItemIterator(tree)

    mw1 = FreeCADGui.getMainWindow()
    print (f"---------Number of trees {len(mw1.findChildren(QtGui.QTreeWidget))}")
    treesSel = mw1.findChildren(QtGui.QTreeWidget)
    print (f"---------Number of trees {len(treesSel)}")

    doc = FreeCAD.ActiveDocument
    found = False

    for tree in treesSel:
        print(f"--------Tree {tree.objectName()}")
        items = tree.selectedItems()
        for item in items:
            treeLabel = item.text(0)
            print(f"--------Item {treeLabel}")
            print(f"--------Doc.Label {doc.Label}")
            # if not found:
            #     if treeLabel != doc.Label:
            #         continue
            # found = True
            try:
                objs = doc.getObjectsByLabel(treeLabel)
                print(f"--------Objects {objs}")
                if len(objs) == 0:
                    continue

                obj = objs[0]
                if obj == worldObj:
                    print(f"--------World Object {obj.Label}")
                    # we presume first app part is world volume
                    addDaughters(item)
                    break
            except Exception as e:
                print(e)
                FreeCADobject = None
```

[image]

[image]

### I'm not familiar with FreeCad's API, but it seems to work fine in my environment.

### Hope it can provide some reference.

KeithSloan, 2025-05-09:
Please reopen if not resolved

closed 2025-05-09 by KeithSloan