KeithSloan/GDML #161: GDML position units doesn't work with cm

ty-borg, 2025-04-29:
Hello,

I've been putting together a gdml file by hand and found that when importing it into FreeCAD that it ignores the units I put for the position. 

For example:

	<position name="detector_position" unit="cm" x="1" y="1" z="1"/>

Always places the detector at a position of (1, 1, 1) mm.  The numbers and the units look fine in the define spreadsheet, but when applied to objects it seems to ignore the units.

ty-borg, 2025-04-30:
Here is the file.  I had to change the .gdml extension to .txt in order to upload the file.  I appreciate the help! [test.txt](https://github.com/user-attachments/files/19981461/test.txt)

KeithSloan, 2025-05-03:
Should now be in Main Branch

```
diff --git a/freecad/gdml/GDMLShared.py b/freecad/gdml/GDMLShared.py
index 1e906f18..8e1f523d 100644
--- a/freecad/gdml/GDMLShared.py
+++ b/freecad/gdml/GDMLShared.py
@@ -1078,21 +1078,37 @@ def setPlacement(obj, xml, invertRotation=True):
     obj.Placement.Base = FreeCAD.Vector(0, 0, 0)
     posName = getRef(xml, "positionref")
 
-    # TODO deal with unit
     if posName is not None:
         row = getPositionRow(posName)
         if row is not None:
+            unit_alias = defineSpreadsheet.getAlias(definesColumn['pos_unit'] + str(row))
+            default_unit = True
+            if unit_alias is not None:
+                unit = defineSpreadsheet.get(unit_alias)
+                default_unit = False
             xAlias = defineSpreadsheet.getAlias(definesColumn['pos_x'] + str(row))
             if xAlias is not None:
-                obj.setExpression('.Placement.Base.x', f"<<defines>>.{xAlias}")
+                if default_unit:
+                    expr = f"<<defines>>.{xAlias}"
+                else:
+                    expr = f"<<defines>>.{xAlias}*1{unit}"
+                obj.setExpression('.Placement.Base.x', expr)
 
             yAlias = defineSpreadsheet.getAlias(definesColumn['pos_y'] + str(row))
             if yAlias is not None:
-                obj.setExpression('.Placement.Base.y', f"<<defines>>.{yAlias}")
+                if default_unit:
+                    expr = f"<<defines>>.{yAlias}"
+                else:
+                    expr = f"<<defines>>.{yAlias}*1{unit}"
+                obj.setExpression('.Placement.Base.y', expr)
 
             zAlias = defineSpreadsheet.getAlias(definesColumn['pos_z'] + str(row))
             if zAlias is not None:
-                obj.setExpression('.Placement.Base.z', f"<<defines>>.{zAlias}")
+                if default_unit:
+                    expr = f"<<defines>>.{zAlias}"
+                else:
+                    expr = f"<<defines>>.{zAlias}*1{unit}"
+                obj.setExpression('.Placement.Base.z', expr)
 
     else:
         pos = xml.find("position")
```

KeithSloan, 2025-05-09:
Please re open if not resolved

closed 2025-05-09 by KeithSloan